FreeCAD/FreeCAD #8130: [Issue] Update all FreeCAD source code headers to 2023

luzpaz, 2023-01-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

n/a

### Version

0.21 (Development)

### Full version info

```shell
n/a
```

### Subproject(s) affected?

None

### Issue description

self-explanatory

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-05:
Would we want to wait until certain large pending PRs are merged ?

luzpaz, 2023-01-09:
Should I close this issue since it's obvious ?

luzpaz, 2023-01-09:
We should do this in conjunction with #6166

donovaly, 2023-01-09:
What exactly has to be updated? For the *-cpp/*.h files we usually use the year the file was created of heavily modified?

Can you therefore please add a list of dates that have to be updated?

luzpaz, 2023-01-09:
@donovaly you're absolutely right, I was mistaken. There are only 3 files that need to be updated. I've created a PR. Please squash before merging. TIA!

PR #8156 (2023-01-09): Increment year in copyright notice (2023)
Closes #8130  
Note to merge dev, please squash commits
 
Thank you for creating a pull request to contribute to FreeCAD! Place an "X" in between the brackets below to "check off" to confirm that you have satisfied the requirement, or ask for help in the [FreeCAD forum](https://forum.freecadweb.org/viewforum.php?f=10) if there is something you don't understand.

- [X]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR

Please remember to upd [... cut]
closed 2023-01-09 by donovaly