FreeCAD/FreeCAD #8134: [Bug] [Crash] Segfault while adding an edge to the chamfer

arcivanov, 2023-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
FreeCAD 0.20.1, Libs: 0.20.1RUnknown
```

### Subproject(s) affected?

PartDesign

### Issue description

Adding an edge to a chamfer results in a segfault. Fedora 37, x86-64. 

```
Jan 05 19:18:57 hostname plasmashell[2306502]: Program received signal SIGSEGV, Segmentation fault.
Jan 05 19:18:57 hostname plasmashell[2306502]: #0  /lib64/libc.so.6(+0x3cb50) [0x7fab1105fb50]
Jan 05 19:18:57 hostname plasmashell[2306502]: #1  0x7faa340c9aaa in BRep_Tool::Curve(TopoDS_Edge const&, TopLoc_Location&, double&, double&) from /lib64/libTKBRep.so.7+0x2a
Jan 05 19:18:57 hostname plasmashell[2306502]: #2  0x7faa340c9c0a in BRep_Tool::Curve(TopoDS_Edge const&, double&, double&) from /lib64/libTKBRep.so.7+0x4a
Jan 05 19:18:57 hostname plasmashell[2306502]: #3  0x7faa2b5183ff in ChFi3d_Builder::PerformIntersectionAtEnd(int) from /lib64/libTKFillet.so.7+0x34af
Jan 05 19:18:57 hostname plasmashell[2306502]: #4  0x7faa2b50c2c7 in ChFi3d_Builder::PerformOneCorner(int, bool) from /lib64/libTKFillet.so.7+0x947
Jan 05 19:18:57 hostname plasmashell[2306502]: #5  0x7faa2b4c8f31 in ChFi3d_Builder::PerformFilletOnVertex(int) from /lib64/libTKFillet.so.7+0x4a1
Jan 05 19:18:57 hostname plasmashell[2306502]: #6  0x7faa2b4cbcd6 in ChFi3d_Builder::Compute() from /lib64/libTKFillet.so.7+0xae6
Jan 05 19:18:57 hostname plasmashell[2306502]: #7  0x7faa2b619ebb in BRepFilletAPI_MakeChamfer::Build() from /lib64/libTKFillet.so.7+0x2b
Jan 05 19:18:57 hostname plasmashell[2306502]: #8  0x7faa2bfa267a in PartDesign::Chamfer::execute() from /usr/lib64/freecad/lib64/_PartDesign.so+0x9ba
Jan 05 19:18:57 hostname plasmashell[2306502]: #9  0x7fab135d2eb2 in App::DocumentObject::recompute() from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x462
Jan 05 19:18:57 hostname plasmashell[2306502]: #10  0x7faa2ba0f41f in Part::Feature::recompute() from /usr/lib64/freecad/lib64/Part.so+0xf
Jan 05 19:18:57 hostname plasmashell[2306502]: #11  0x7fab1357ed4c in App::Document::_recomputeFeature(App::DocumentObject*) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x22c
Jan 05 19:18:57 hostname plasmashell[2306502]: #12  0x7fab13583073 in App::Document::recomputeFeature(App::DocumentObject*, bool) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x113
Jan 05 19:18:57 hostname plasmashell[2306502]: #13  0x7faa27d2f808 in PartDesignGui::TaskDressUpParameters::referenceSelected(Gui::SelectionChanges const&) from /usr/lib64/freecad/lib64/PartDesignGui.so+0x1a8
Jan 05 19:18:57 hostname plasmashell[2306502]: #14  0x7faa27d28960 in PartDesignGui::TaskChamferParameters::onSelectionChanged(Gui::SelectionChanges const&) from /usr/lib64/freecad/lib64/PartDesignGui.so+0x60
Jan 05 19:18:57 hostname plasmashell[2306502]: #15  0x7fab1424191e in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x2e
Jan 05 19:18:57 hostname plasmashell[2306502]: #16  /usr/lib64/freecad/lib64/libFreeCADGui.so(+0x623809) [0x7fab14023809]
Jan 05 19:18:57 hostname plasmashell[2306502]: #17  0x7fab14249036 in Gui::SelectionSingleton::slotSelectionChanged(Gui::SelectionChanges const&) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x216
Jan 05 19:18:57 hostname plasmashell[2306502]: #18  /usr/lib64/freecad/lib64/libFreeCADGui.so(+0x623809) [0x7fab14023809]
Jan 05 19:18:57 hostname plasmashell[2306502]: #19  0x7fab1424cad5 in Gui::SelectionSingleton::notify(Gui::SelectionChanges&&) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x135
Jan 05 19:18:57 hostname plasmashell[2306502]: #20  0x7fab1425292a in Gui::SelectionSingleton::addSelection(char const*, char const*, char const*, float, float, float, std::vector<Gui::SelectionSingleton::SelObj, std::allocator<Gui::SelectionSingleton::SelObj> > const*, bool) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0xc4a
Jan 05 19:18:57 hostname plasmashell[2306502]: #21  0x7fab14086d3a in Gui::SoFCUnifiedSelection::setSelection(std::vector<Gui::SoFCUnifiedSelection::PickedInfo, std::allocator<Gui::SoFCUnifiedSelection::PickedInfo> > const&, bool) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x61a
Jan 05 19:18:57 hostname plasmashell[2306502]: #22  0x7fab14087761 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x141
Jan 05 19:18:57 hostname plasmashell[2306502]: #23  0x7fab108ef766 in SoAction::traverse(SoNode*) from /lib64/libCoin.so.80+0x66
Jan 05 19:18:57 hostname plasmashell[2306502]: #24  0x7fab10a7ef08 in SoChildList::traverse(SoAction*, int, int) from /lib64/libCoin.so.80+0x158
Jan 05 19:18:57 hostname plasmashell[2306502]: #25  0x7fab10ac500f in SoGroup::doAction(SoAction*) from /lib64/libCoin.so.80+0x4f
Jan 05 19:18:57 hostname plasmashell[2306502]: #26  0x7fab10ad9c5b in SoSeparator::doAction(SoAction*) from /lib64/libCoin.so.80+0x2b
Jan 05 19:18:57 hostname plasmashell[2306502]: #27  0x7fab108ef766 in SoAction::traverse(SoNode*) from /lib64/libCoin.so.80+0x66
Jan 05 19:18:57 hostname plasmashell[2306502]: #28  0x7fab108f8ebb in SoHandleEventAction::beginTraversal(SoNode*) from /lib64/libCoin.so.80+0xab
Jan 05 19:18:57 hostname plasmashell[2306502]: #29  0x7fab108efdb1 in SoAction::apply(SoNode*) from /lib64/libCoin.so.80+0xc1
Jan 05 19:18:57 hostname plasmashell[2306502]: #30  0x7fab10a93dc9 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /lib64/libCoin.so.80+0x79
Jan 05 19:18:57 hostname plasmashell[2306502]: #31  0x7fab14107192 in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x22
Jan 05 19:18:57 hostname plasmashell[2306502]: #32  0x7fab1410c2c8 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x58
Jan 05 19:18:57 hostname plasmashell[2306502]: #33  0x7fab1413e88d in Gui::View3DInventorViewer::processSoEventBase(SoEvent const*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0xd
Jan 05 19:18:57 hostname plasmashell[2306502]: #34  0x7fab1411aced in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x2d
Jan 05 19:18:57 hostname plasmashell[2306502]: #35  0x7fab1411b3c8 in Gui::CADNavigationStyle::processSoEvent(SoEvent const*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x218
Jan 05 19:18:57 hostname plasmashell[2306502]: #36  0x7fab1411adae in Gui::NavigationStyle::processEvent(SoEvent const*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x7e
Jan 05 19:18:57 hostname plasmashell[2306502]: #37  0x7fab14141dff in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x6f
Jan 05 19:18:57 hostname plasmashell[2306502]: #38  0x7fab14104565 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x1b5
Jan 05 19:18:57 hostname plasmashell[2306502]: #39  0x7fab118a8041 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /lib64/libQt5Core.so.5+0x91
Jan 05 19:18:57 hostname plasmashell[2306502]: #40  0x7fab125aed02 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib64/libQt5Widgets.so.5+0x72
Jan 05 19:18:57 hostname plasmashell[2306502]: #41  0x7fab125b6a2a in QApplication::notify(QObject*, QEvent*) from /lib64/libQt5Widgets.so.5+0xa7a
Jan 05 19:18:57 hostname plasmashell[2306502]: #42  0x7fab13e7c530 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x90
Jan 05 19:18:57 hostname plasmashell[2306502]: #43  0x7fab118a8278 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib64/libQt5Core.so.5+0x118
Jan 05 19:18:57 hostname plasmashell[2306502]: #44  0x7fab125b5482 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib64/libQt5Widgets.so.5+0x1c2
Jan 05 19:18:57 hostname plasmashell[2306502]: #45  /lib64/libQt5Widgets.so.5(+0x20ab7c) [0x7fab1260ab7c]
[... 10 lines cut ...]
Jan 05 19:18:57 hostname plasmashell[2306502]: #56  0x7fab118f938a in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib64/libQt5Core.so.5+0x6a
Jan 05 19:18:57 hostname plasmashell[2306502]: #57  0x7fab118a6cca in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib64/libQt5Core.so.5+0x13a
[... cut ...]
Jan 05 19:18:57 hostname plasmashell[2306502]:   #     ##  #### ####  #     #   #  #   #
Jan 05 19:18:57 hostname plasmashell[2306502]:   ####  # # #  # #  #  #     #####  #   #
Jan 05 19:18:57 hostname plasmashell[2306502]:   #     #   #### ####  #    #     # #   #
Jan 05 19:18:57 hostname plasmashell[2306502]:   #     #   #    #     #    #     # #   #  ##  ##  ##
Jan 05 19:18:57 hostname plasmashell[2306502]:   #     #   #### ####   ### #     # ####   ##  ##  ##

```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

arcivanov, 2023-01-06:
This is always reproducible. 
Adding `Body002.Pocket007.Edge86` to chamfer to the already present Edge39 always results in a crash.
Attaching the project as well. The code is proprietary, but OSS copyrighted to Arcadiy Ivanov under Apache License 2.0.

arcivanov, 2023-01-06:
[vone_carrier_jensen.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10356486/vone_carrier_jensen.FCStd.zip)

arcivanov, 2023-01-06:
Same with trying to add `Body002.Pocket007.Edge137` - also segfault.

arcivanov, 2023-01-06:
Same crash with today's Fedora nightly (0.21.0Rpre_31513.fc37):

```
Jan 05 19:45:31 hostname plasmashell[2314967]: Program received signal SIGSEGV, Segmentation fault.
Jan 05 19:45:31 hostname plasmashell[2314967]: #0  /lib64/libc.so.6(+0x3cb50) [0x7f62e7a5fb50]
Jan 05 19:45:31 hostname plasmashell[2314967]: #1  0x7f61d0d8aaaa in BRep_Tool::Curve(TopoDS_Edge const&, TopLoc_Location&, double&, double&) from /lib64/libTKBRep.so.7+0x2a
Jan 05 19:45:31 hostname plasmashell[2314967]: #2  0x7f61d0d8ac0a in BRep_Tool::Curve(TopoDS_Edge const&, double&, double&) from /lib64/libTKBRep.so.7+0x4a
Jan 05 19:45:31 hostname plasmashell[2314967]: #3  0x7f61d0b183ff in ChFi3d_Builder::PerformIntersectionAtEnd(int) from /lib64/libTKFillet.so.7+0x34af
Jan 05 19:45:31 hostname plasmashell[2314967]: #4  0x7f61d0b0c2c7 in ChFi3d_Builder::PerformOneCorner(int, bool) from /lib64/libTKFillet.so.7+0x947
Jan 05 19:45:31 hostname plasmashell[2314967]: #5  0x7f61d0ac8f31 in ChFi3d_Builder::PerformFilletOnVertex(int) from /lib64/libTKFillet.so.7+0x4a1
Jan 05 19:45:31 hostname plasmashell[2314967]: #6  0x7f61d0acbcd6 in ChFi3d_Builder::Compute() from /lib64/libTKFillet.so.7+0xae6
Jan 05 19:45:31 hostname plasmashell[2314967]: #7  0x7f61d0c19ebb in BRepFilletAPI_MakeChamfer::Build() from /lib64/libTKFillet.so.7+0x2b
Jan 05 19:45:31 hostname plasmashell[2314967]: #8  0x7f61d849e8bc in PartDesign::Chamfer::execute() from /usr/lib64/freecad/lib64/_PartDesign.so+0x78c
Jan 05 19:45:31 hostname plasmashell[2314967]: #9  0x7f62eada4fb9 in App::DocumentObject::recompute() from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x75
Jan 05 19:45:31 hostname plasmashell[2314967]: #10  0x7f61d127be7f in Part::Feature::recompute() from /usr/lib64/freecad/lib64/Part.so+0x1d
Jan 05 19:45:31 hostname plasmashell[2314967]: #11  0x7f62eac7af64 in App::Document::_recomputeFeature(App::DocumentObject*) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x1be
Jan 05 19:45:31 hostname plasmashell[2314967]: #12  0x7f62eac7bdb4 in App::Document::recomputeFeature(App::DocumentObject*, bool) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0xf8
Jan 05 19:45:31 hostname plasmashell[2314967]: #13  0x7f61cc79b21e in PartDesignGui::TaskDressUpParameters::referenceSelected(Gui::SelectionChanges const&) from /usr/lib64/freecad/lib64/PartDesignGui.so+0x2f2
Jan 05 19:45:31 hostname plasmashell[2314967]: #14  0x7f61cc78a706 in PartDesignGui::TaskChamferParameters::onSelectionChanged(Gui::SelectionChanges const&) from /usr/lib64/freecad/lib64/PartDesignGui.so+0x44
Jan 05 19:45:31 hostname plasmashell[2314967]: #15  0x7f62ecd220a0 in Gui::SelectionObserver::_onSelectionChanged(Gui::SelectionChanges const&) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x54
Jan 05 19:45:31 hostname plasmashell[2314967]: #16  0x7f62ecd3febc in boost::_mfi::mf1<void, Gui::SelectionObserver, Gui::SelectionChanges const&>::operator()(Gui::SelectionObserver*, Gui::SelectionChanges const&) const from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x70
Jan 05 19:45:31 hostname plasmashell[2314967]: #17  0x7f62ecd3f63d in void boost::_bi::list2<boost::_bi::value<Gui::SelectionObserver*>, boost::arg<1> >::operator()<boost::_mfi::mf1<void, Gui::SelectionObserver, Gui::SelectionChanges const&>, boost::_bi::rrlist1<Gui::SelectionChanges const&> >(boost::_bi::type<void>, boost::_mfi::mf1<void, Gui::SelectionObserver, Gui::SelectionChanges const&>&, boost::_bi::rrlist1<Gui::SelectionChanges const&>&, int) from /usr/lib>
Jan 05 19:45:31 hostname plasmashell[2314967]: #18  0x7f62ecd3ec8a in void boost::_bi::bind_t<void, boost::_mfi::mf1<void, Gui::SelectionObserver, Gui::SelectionChanges const&>, boost::_bi::list2<boost::_bi::value<Gui::SelectionObserver*>, boost::arg<1> > >::operator()<Gui::SelectionChanges const&>(Gui::SelectionChanges const&) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x40
Jan 05 19:45:31 hostname plasmashell[2314967]: #19  0x7f62ecd3df51 in boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf1<void, Gui::SelectionObserver, Gui::SelectionChanges const&>, boost::_bi::list2<boost::_bi::value<Gui::SelectionObserver*>, boost::arg<1> > >, void, Gui::SelectionChanges const&>::invoke(boost::detail::function::function_buffer&, Gui::SelectionChanges const&) from /usr/lib64/freecad/lib64/libFreeCADGui.s>
Jan 05 19:45:31 hostname plasmashell[2314967]: #20  0x7f62ec999b43 in boost::function1<void, Gui::SelectionChanges const&>::operator()(Gui::SelectionChanges const&) const from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x5d
Jan 05 19:45:31 hostname plasmashell[2314967]: #21  0x7f62ec997cd4 in boost::signals2::detail::void_type boost::signals2::detail::call_with_tuple_args<boost::signals2::detail::void_type>::m_invoke<boost::function<void (Gui::SelectionChanges const&)>, 0u, Gui::SelectionChanges const&>(boost::function<void (Gui::SelectionChanges const&)>&, boost::signals2::detail::unsigned_meta_array<0u>, std::tuple<Gui::SelectionChanges const&> const&, boost::enable_if<boost::is_vo>
Jan 05 19:45:31 hostname plasmashell[2314967]: #22  0x7f62ec994ee9 in boost::signals2::detail::void_type boost::signals2::detail::call_with_tuple_args<boost::signals2::detail::void_type>::operator()<boost::function<void (Gui::SelectionChanges const&)>, Gui::SelectionChanges const&, 1ul>(boost::function<void (Gui::SelectionChanges const&)>&, std::tuple<Gui::SelectionChanges const&> const&, mpl_::size_t<1ul>) const from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x2d
[... cut ...]
Jan 05 19:45:31 hostname plasmashell[2314967]:   #                    #      # #   #   #
Jan 05 19:45:31 hostname plasmashell[2314967]:   #     ##  #### ####  #     #   #  #   #
Jan 05 19:45:31 hostname plasmashell[2314967]:   ####  # # #  # #  #  #     #####  #   #
Jan 05 19:45:31 hostname plasmashell[2314967]:   #     #   #### ####  #    #     # #   #
Jan 05 19:45:31 hostname plasmashell[2314967]:   #     #   #    #     #    #     # #   #  ##  ##  ##
Jan 05 19:45:31 hostname plasmashell[2314967]:   #     #   #### ####   ### #     # ####   ##  ##  ##
```

arcivanov, 2023-01-06:
[code]
OS: Fedora Linux 37 (Workstation Edition) (KDE/default)
Word size of FreeCAD: 64-bit
Version: 0.21.0.pre_31513.fc37 (Git)
Build type: Unknown
Python 3.11.1, Qt 5.15.7, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.0
Locale: English/United States (en_US)
[/code]

arcivanov, 2023-01-06:
Same error with same file changing constraint `Sketch017.Constraint17` from `-4.8` to `5.0` will result in the same segfault.

```
Jan 06 14:32:21 hostname plasmashell[210974]: Program received signal SIGSEGV, Segmentation fault.
Jan 06 14:32:21 hostname plasmashell[210974]: #0  /lib64/libc.so.6(+0x3cb50) [0x7f47bb65fb50]
Jan 06 14:32:21 hostname plasmashell[210974]: #1  0x7f46a498aaaa in BRep_Tool::Curve(TopoDS_Edge const&, TopLoc_Location&, double&, double&) from /lib64/libTKBRep.so.7+0x2a
Jan 06 14:32:21 hostname plasmashell[210974]: #2  0x7f46a498ac0a in BRep_Tool::Curve(TopoDS_Edge const&, double&, double&) from /lib64/libTKBRep.so.7+0x4a
Jan 06 14:32:21 hostname plasmashell[210974]: #3  0x7f46a47183ff in ChFi3d_Builder::PerformIntersectionAtEnd(int) from /lib64/libTKFillet.so.7+0x34af
Jan 06 14:32:21 hostname plasmashell[210974]: #4  0x7f46a470c2c7 in ChFi3d_Builder::PerformOneCorner(int, bool) from /lib64/libTKFillet.so.7+0x947
Jan 06 14:32:21 hostname plasmashell[210974]: #5  0x7f46a46c8f31 in ChFi3d_Builder::PerformFilletOnVertex(int) from /lib64/libTKFillet.so.7+0x4a1
Jan 06 14:32:21 hostname plasmashell[210974]: #6  0x7f46a46cbcd6 in ChFi3d_Builder::Compute() from /lib64/libTKFillet.so.7+0xae6
Jan 06 14:32:21 hostname plasmashell[210974]: #7  0x7f46a4819fcb in BRepFilletAPI_MakeFillet::Build() from /lib64/libTKFillet.so.7+0x2b
Jan 06 14:32:21 hostname plasmashell[210974]: #8  0x7f46a569cb5a in PartDesign::Fillet::execute() from /usr/lib64/freecad/lib64/_PartDesign.so+0x458
Jan 06 14:32:21 hostname plasmashell[210974]: #9  0x7f47be9a4fb9 in App::DocumentObject::recompute() from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x75
Jan 06 14:32:21 hostname plasmashell[210974]: #10  0x7f46a4e7be7f in Part::Feature::recompute() from /usr/lib64/freecad/lib64/Part.so+0x1d
Jan 06 14:32:21 hostname plasmashell[210974]: #11  0x7f47be87af64 in App::Document::_recomputeFeature(App::DocumentObject*) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x1be
Jan 06 14:32:21 hostname plasmashell[210974]: #12  0x7f47be878d9b in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x8c1
Jan 06 14:32:21 hostname plasmashell[210974]: #13  0x7f47bea237b9 in App::DocumentPy::recompute(_object*) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x2c7
Jan 06 14:32:21 hostname plasmashell[210974]: #14  0x7f47bea1b6cf in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0xf3
Jan 06 14:32:21 hostname plasmashell[210974]: #15  /lib64/libpython3.11.so.1.0(+0x1d8768) [0x7f47bd3d8768]
Jan 06 14:32:21 hostname plasmashell[210974]: #16  /lib64/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x73) [0x7f47bd3ba393]
Jan 06 14:32:21 hostname plasmashell[210974]: #17  /lib64/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x912) [0x7f47bd3c2ff2]
Jan 06 14:32:21 hostname plasmashell[210974]: #18  /lib64/libpython3.11.so.1.0(+0x1bedaa) [0x7f47bd3bedaa]
Jan 06 14:32:21 hostname plasmashell[210974]: #19  /lib64/libpython3.11.so.1.0(PyEval_EvalCode+0xac) [0x7f47bd4493dc]
Jan 06 14:32:21 hostname plasmashell[210974]: #20  /lib64/libpython3.11.so.1.0(+0x268203) [0x7f47bd468203]
Jan 06 14:32:21 hostname plasmashell[210974]: #21  /lib64/libpython3.11.so.1.0(+0x26473a) [0x7f47bd46473a]
Jan 06 14:32:21 hostname plasmashell[210974]: #22  /lib64/libpython3.11.so.1.0(PyRun_StringFlags+0x6d) [0x7f47bd45910d]
Jan 06 14:32:21 hostname plasmashell[210974]: #23  0x7f47bdaa89bf in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/lib64/freecad/lib64/libFreeCADBase.so+0xd5
Jan 06 14:32:21 hostname plasmashell[210974]: #24  0x7f47c0304e54 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0xca
Jan 06 14:32:21 hostname plasmashell[210974]: #25  0x7f47c0304b03 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0xfd
Jan 06 14:32:21 hostname plasmashell[210974]: #26  0x7f47c0305c81 in Gui::Command::updateActive() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x3d
Jan 06 14:32:21 hostname plasmashell[210974]: #27  0x7f469f0d3e3e in SketcherGui::tryAutoRecompute(Sketcher::SketchObject*, bool&) from /usr/lib64/freecad/lib64/SketcherGui.so+0xc0
Jan 06 14:32:21 hostname plasmashell[210974]: #28  0x7f469f0d3e9b in SketcherGui::tryAutoRecompute(Sketcher::SketchObject*) from /usr/lib64/freecad/lib64/SketcherGui.so+0x1f
Jan 06 14:32:21 hostname plasmashell[210974]: #29  0x7f469f27ab0c in SketcherGui::EditDatumDialog::accepted() from /usr/lib64/freecad/lib64/SketcherGui.so+0x468
Jan 06 14:32:21 hostname plasmashell[210974]: #30  /usr/lib64/freecad/lib64/SketcherGui.so(+0x47af7b) [0x7f469f27af7b]
Jan 06 14:32:21 hostname plasmashell[210974]: #31  /lib64/libQt5Core.so.5(+0x2dbdaf) [0x7f47bbedbdaf]
Jan 06 14:32:21 hostname plasmashell[210974]: #32  0x7f47bcdbe428 in QDialogPrivate::finalize(int, int) from /lib64/libQt5Widgets.so.5+0x58
Jan 06 14:32:21 hostname plasmashell[210974]: #33  /lib64/libQt5Core.so.5(+0x2dbdaf) [0x7f47bbedbdaf]
Jan 06 14:32:21 hostname plasmashell[210974]: #34  /lib64/libQt5Widgets.so.5(+0x350020) [0x7f47bcd50020]
Jan 06 14:32:21 hostname plasmashell[210974]: #35  /lib64/libQt5Core.so.5(+0x2dbdaf) [0x7f47bbedbdaf]
[... cut ...]
Jan 06 14:32:22 hostname plasmashell[210974]:   #                    #      # #   #   #
Jan 06 14:32:22 hostname plasmashell[210974]:   #     ##  #### ####  #     #   #  #   #
Jan 06 14:32:22 hostname plasmashell[210974]:   ####  # # #  # #  #  #     #####  #   #
Jan 06 14:32:22 hostname plasmashell[210974]:   #     #   #### ####  #    #     # #   #
Jan 06 14:32:22 hostname plasmashell[210974]:   #     #   #    #     #    #     # #   #  ##  ##  ##
Jan 06 14:32:22 hostname plasmashell[210974]:   #     #   #### ####   ### #     # ####   ##  ##  ##
```

arcivanov, 2023-01-07:
@luzpaz the info has been updated

luzpaz, 2023-01-11:
* Current stable OCC v7.7.0  
* OCC bugs in the tracker:  https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aissue+is%3Aopen+label%3A%223rd+Party%3A+OCC%22
* OCC bugs tracked in the forum: https://forum.freecadweb.org/viewtopic.php?style=1&f=10&t=20264

arcivanov, 2023-01-11:
> * Current stable OCC v7.7.0

I understand that, but Fedora 37 packages 7.5.0, which is 2 years old.

luzpaz, 2023-01-11:
This is a downstream Fedora packaging issue. Can you reproduce using AppImage or Snap 'edge'  ?

arcivanov, 2023-01-13:
The crash went away with:

```
OS: Fedora Linux 37 (Workstation Edition) (KDE/default)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * ExtremeProDark 2.6.4
  * ProDarkThemePreferencePack 1.0.0
```

closed 2023-01-13 by arcivanov