FreeCAD/FreeCAD #8137: [Issue] Opening OpenSCAD file fails

michalfikera, 2023-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Czech/Czech Republic (cs_CZ)
[/code]
```

### Subproject(s) affected?

OpenSCAD

### Issue description

Opening the attached openScad file fails. The fulllog is attached as well. The version 0.19 opened the file without any problem.

15:27:22  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Program Files\FreeCAD 0.20\Mod\OpenSCAD\importCSG.py", line 130, in open
    processcsg(tmpfile)
  File "C:\Program Files\FreeCAD 0.20\Mod\OpenSCAD\importCSG.py", line 192, in processcsg
    result = parser.parse(f.read())
  File "C:\Program Files\FreeCAD 0.20\bin\lib\site-packages\ply\yacc.py", line 333, in parse
    return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
  File "C:\Program Files\FreeCAD 0.20\bin\lib\site-packages\ply\yacc.py", line 1120, in parseopt_notrack
    p.callable(pslice)
  File "C:\Program Files\FreeCAD 0.20\Mod\OpenSCAD\importCSG.py", line 693, in p_intersection_action
    mycommon.Shape = mycommon.Base.Shape.common(mycommon.Tool.Shape)
<class 'AttributeError'>: 'Part.Feature' object has no attribute 'Base'

### Anything else?

[Freecad_errorlog.zip](https://github.com/FreeCAD/FreeCAD/files/10360807/Freecad_errorlog.zip)
[Prijon_spunt.zip](https://github.com/FreeCAD/FreeCAD/files/10360809/Prijon_spunt.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-14:
@keepitsimplejim would you have a moment to look in to this?

KeithSloan, 2023-02-14:
Okay a number of things.

There was a bug in intersection of more than two objects, the file in question was trying to intersect three.
I have fixed this in the Alternate Importer [https://github.com/KeithSloan/OpenSCAD_Alt_Import](url)

The changes are in branch FixCommonBase.

The code for handling two objects, more than two and less than two should be very similar between
Fuse, Common and Intersection, certainly in the Alternative Import things are not as tidy as they should be.
Maybe Chris could tidy in the main repro.

Would be good if there was test cases for various possibilities Fuse, Common, Intersection
i.e two objects, greater than two, one and none.

Also had problems when I downloaded Prijon_spunt.zip on my Mac. I could not load the file into
OpenSCAD or FreeCAD, it was given a file name of Prijon_spunt.scad but a Mac File Kind or "OpenSCAD Design".

Had to resort to opening a terminal window and copying. Would like advice about this.

KeithSloan, 2023-02-14:
As the file opened in FC 0.19, I assume there was code regression for the OpenSCAD importer in FC 0.20

KeithSloan, 2023-02-14:
Question maybe @chennes can answer. Boolean operations should either be between two objects or more than two.
The importer code also seems to try and deal with one and none. I not that sometimes a placeholder is mentioned,
wonder if this dates back to the days when Sebasian shoogen was involved. To my mind booleans with one or no objects is not valid and OpenSCAD should fail anyway. Do we have test cases with one and none? What does OpenSCAD do with them?

chennes, 2023-02-14:
OpenSCAD accepts single-argument booleans, and simply does nothing with them. I've added code to handle it in some places, but possibly not everywhere. And I haven't looked at all at intersecting more than two objects.

KeithSloan, 2023-02-15:
The bug was because for Booleans with two objects it was necessary to add a check to see if the Base and Tool had valid Shapes and if not do a recompute for them so they do. For Multi booleans i.e. more than two objects there is no Base or Tool property and for an intersection the code was trying to perform a check. The checks were introduced after 0.19 version so I think that would explain the regression seen.

KeithSloan, 2023-11-16:
The problem occurs on an intersection of three objects

Matrix_Deformation
cylinder004
union

the intersection code creates a MultiCommon with the three shapes, i.e. myCommon.Shapes = p[5]
where p[5] is the list of the three shapes but the code then tries to perform a common on the Base and the Tool which do not exist.

I tried creating the multiCommon using myCommon.Shape = first.multiCommon(rest)
where first = p[0] and rest = p[1:]

but the Matrix_Deformation Shape does not have a multiCommon function

Debugging version of importCSG.py attached

[importCSG.py.txt](https://github.com/FreeCAD/FreeCAD/files/13373761/importCSG.py.txt)

Maybe the bug is as a result of bug in Part::MultiCommon, I don't have a older version of FreeCAD to test.
@chennes What should a multiCommon of the three objects as above look like.

KeithSloan, 2023-11-16:
Latest attempt
[importCSG.py.txt](https://github.com/Fre
eCAD/FreeCAD/files/13374246/importCSG.py.txt)

Still failing

08:32:31  intersection
08:32:31  Single Common
08:32:31  End Intersection
08:32:31  intersection
08:32:31  Multi Common
08:32:31  subobj Matrix_Deformation
08:32:31  subobj cylinder004
08:32:31  subobj union
08:32:31  End Intersection
08:32:31  <Exception> Input shape is null
08:32:31  Recompute failed : intersection001
08:32:31  End processing CSG file

Need a break and relook afresh.

KeithSloan, 2023-11-16:
Interesting, Okay I get the error messages

08:32:31 Input shape is null
08:32:31 Recompute failed : intersection001

And if I toggle the visibility on cylinder003 I get
[image]

Which looks like the native OpenSCAD version
[image]

Like I said I need a break and to look at this afresh.

luzpaz, 2024-02-11:
@KeithSloan any possibility for you to revisit this ?

KeithSloan, 2024-02-11:
Okay think I have fixed this in the Alternate importer

Relevant changes were and need to be made to code in Main repro.

```
def p_intersection_action(p):
    'intersection_action : intersection LPAREN RPAREN OBRACE block_list EBRACE'

    if printverbose: print("intersection")
    # Is this Multi Common
    if (len(p[5]) > 2):
       if printverbose: print("Multi Common")
       mycommon = doc.addObject('Part::MultiCommon',p[1])
       mycommon.Shapes = p[5]
       for s in mycommon.Shapes:
           checkObjShape(s)
       if gui:
           for subobj in mycommon.Shapes:
               subobj.ViewObject.hide()
    elif (len(p[5]) == 2):
       if printverbose: print("Single Common")
       mycommon = doc.addObject('Part::Common',p[1])
       mycommon.Base = p[5][0]
       mycommon.Tool = p[5][1]
       checkObjShape(mycommon.Base)
       checkObjShape(mycommon.Tool)
       mycommon.Shape = mycommon.Base.Shape.common(mycommon.Tool.Shape)
       if gui:
           mycommon.Base.ViewObject.hide()
           mycommon.Tool.ViewObject.hide()
    elif (len(p[5]) == 1):
        mycommon = p[5][0]
    else : # 1 child
        mycommon = placeholder('group',[],'{}')
    p[0] = [mycommon]
    if printverbose: print("End Intersection")
```

And 

```
def fuse(lst,name):
    global doc
    if printverbose: 
       print("Fuse")
       print(lst)
       for obj in lst :
           print(obj.Label)
    if len(lst) == 0:
        myfuse = placeholder('group',[],'{}')
    elif len(lst) == 1:
       return lst[0]
    # Is this Multi Fuse
    elif len(lst) > 2:
       if printverbose: print("Multi Fuse")
       myfuse = doc.addObject('Part::MultiFuse',name)
       myfuse.Shapes = lst
       for s in myfuse.Shapes:
           checkObjShape(s)
       if gui:
           for subobj in myfuse.Shapes:
               subobj.ViewObject.hide()
    else:
       if printverbose: print("Single Fuse")
       myfuse = doc.addObject('Part::Fuse',name)
       myfuse.Base = lst[0]
       myfuse.Tool = lst[1]
       checkObjShape(myfuse.Base)
       checkObjShape(myfuse.Tool)
       myfuse.Shape = myfuse.Base.Shape.fuse(myfuse.Tool.Shape)
       myfuse.Placement = FreeCAD.Placement()
       if gui:
           myfuse.Base.ViewObject.hide()
           myfuse.Tool.ViewObject.hide()
    return(myfuse)
```

That is if the operation involves more than two objects, loop through objects and call checkObjShape to perform recompute if it is null
```
for s in xxxx.Shapes:
      checkObjShape(s)
```

luzpaz, 2024-08-23:
@KeithSloan how can users use the alternate importer ?

KeithSloan, 2024-08-23:
You install via FreeCAD Addon Manager

Tools | Addon Manager

Scroll down to
[image]

Install

Then when you open a scad or csg file with

File | Open

For scad filetype file, you should get prompted with for scad

[image]

For csg filetype
[image]

New Importer : SCAD (importAltCSG) is the Alternate importer
OpenSCAD Format  : ImportCSG is the standard FC importer

Other options are for SCAD objects see the README at https://github.com/KeithSloan/OpenSCAD_Alt_Import
Just scroll past file details.

closed 2025-01-12 by github-actions[bot]