FreeCAD/FreeCAD #8138: [Bug] [Crash] Segfault opening in file in nightly 0.21.0Rpre_31513.fc37

arcivanov, 2023-01-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Fedora Linux 37 (Workstation Edition) (KDE/default)
Word size of FreeCAD: 64-bit
Version: 0.21.0.pre_31513.fc37 (Git)
Build type: Unknown
Python 3.11.1, Qt 5.15.7, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.0
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Issue description

```
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: Program received signal SIGSEGV, Segmentation fault.
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #0  /lib64/libc.so.6(+0x3cb50) [0x7f1cf2c5fb50]
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #1  0x7f1cf5eaccf2 in std::_Hashtable<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > con>
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #2  0x7f1cf5eac6d7 in std::_Hashtable<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > con>
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #3  0x7f1cf5e97807 in std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, App::DocumentObject*, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, s>
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #4  0x7f1cf5e7fe40 in App::Document::getObject(char const*) const from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x52
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #5  0x7f1cf60ecbf8 in App::ObjectIdentifier::getDocumentObject(App::Document const*, App::ObjectIdentifier::String const&, std::bitset<32ul>&) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x68
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #6  0x7f1cf60ed318 in App::ObjectIdentifier::resolve(App::ObjectIdentifier::ResolveResults&) const from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x3a0
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #7  0x7f1cf60f3ce2 in App::ObjectIdentifier::ResolveResults::ResolveResults(App::ObjectIdentifier const&) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x1ec
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #8  0x7f1cf60ef789 in App::ObjectIdentifier::getDocumentName() const from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x33
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #9  0x7f1cf60edd3a in App::ObjectIdentifier::getDocument(App::ObjectIdentifier::String, bool*) const from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x52
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #10  0x7f1cf60edfdc in App::ObjectIdentifier::getDocumentObject() const from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x6e
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #11  0x7f1cf7898338 in Gui::ExpressionBinding::isBound() const from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x1c
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #12  0x7f1cf7898aa8 in Gui::ExpressionBinding::hasExpression() const from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x20
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #13  0x7f1cf7ea62a9 in Gui::ExpressionSpinBox::drawControl(QStyleOptionSpinBox&) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x1d
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #14  0x7f1cf7ea021f in Gui::QuantitySpinBox::paintEvent(QPaintEvent*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x47
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #15  0x7f1cf41f1818 in QWidget::event(QEvent*) from /lib64/libQt5Widgets.so.5+0x1b8
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #16  0x7f1cf7ea1dc5 in Gui::QuantitySpinBox::event(QEvent*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x23
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #17  0x7f1cf41aed12 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib64/libQt5Widgets.so.5+0x82
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #18  0x7f1cf78add36 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x104
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #19  0x7f1cf34a8278 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib64/libQt5Core.so.5+0x118
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #20  0x7f1cf41e9ab9 in QWidgetPrivate::sendPaintEvent(QRegion const&) from /lib64/libQt5Widgets.so.5+0x39
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #21  0x7f1cf41ea801 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /lib64/libQt5Widgets.so.5+0xcf1
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #22  0x7f1cf41eb666 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) >
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #23  0x7f1cf41eb4c9 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) >
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #24  0x7f1cf41eb4c9 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) >
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #25  0x7f1cf41eb4c9 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) >
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #26  0x7f1cf41eb4c9 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) >
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #27  0x7f1cf41eb4c9 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) >
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #28  0x7f1cf41eb4c9 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) >
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #29  0x7f1cf41eb4c9 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) >
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #30  0x7f1cf41eb4c9 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) >
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #31  0x7f1cf41eb4c9 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) >
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #32  0x7f1cf41eb4c9 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) >
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #33  0x7f1cf41e9d82 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /lib64/libQt5Widgets.so.5+0x272
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #34  0x7f1cf41eb666 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) >
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: #35  0x7f1cf41eb4c9 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidget [... cut ...]
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]:   #     #   #    #     #    #     # #   #  ##  ##  ##
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]:   #     #   #### ####   ### #     # ####   ##  ##  ##
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: No point picked.
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: No point picked.
Jan 07 00:55:50 ai-karellen-lap plasmashell[212260]: No point picked.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

arcivanov, 2023-01-07:
Same file opened fine on 0.20.1 stable.

arcivanov, 2023-01-07:
[kp3s_heat_shroud.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10365604/kp3s_heat_shroud.FCStd.zip)

luzpaz, 2023-03-15:
@arcivanov can you still reproduce on latest HEAD ? (Can you also use a more up-to-date version of OCC? Current stable is 7.7.0)

arcivanov, 2023-03-16:
Opened the attached example with no issues.

```
OS: Fedora Linux 37 (Workstation Edition) (KDE/default)
Word size of FreeCAD: 64-bit
Version: 0.21.0.pre_31942.fc37 (Git)
Build type: Unknown
Python 3.11.1, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * ExtremeProDark 2.6.4
  * ProDarkThemePreferencePack 1.0.0
```

closed 2023-03-16 by arcivanov