FreeCAD/FreeCAD #8140: [Issue] Path Helix Produces dangerous gcode

sliptonic, 2023-01-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=74977

### Version

0.21 (Development)

### Full version info

```shell
OS: Fedora Linux 37 (KDE Plasma) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31508 (Git)
Build type: Release
Branch: master
Hash: ed8e45ac9b9196ec42572b6746b27cdd6183c903
Python 3.11.1, Qt 5.15.7, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * lattice2 1.0.0
  * kicadStepUpMod 10.16.5
  * fasteners 0.4.51
  * freecad.gears 1.0.0
```

### Subproject(s) affected?

Path

### Issue description

The helix operation can produce G0 moves below the safeheight that are potentially dangerous.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-06:
Still an issue in 
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.1
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.31
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.5.6
  * Silk 0.1.5

```

sslupsky, 2024-12-06:
@maxwxyz  Not sure why #18318 was closed considering that there was a PR #8147 to address the issue and @sliptonic closed it.  For unknown reasons did not fix this serious problem.  Please consider reviving the PR and merging it ... since this generates machine code the consequence of this issue could cause serious safety issues for machine operators and should be addressed.

sliptonic, 2025-06-03:
There have been some recent changes to helix code.  Please retest and comment here.  If we don't hear back by 1 July, @maxwxyz please close the issue as resolved.

closed 2025-06-18 by github-actions[bot]