FreeCAD/FreeCAD #8142: [Issue] 0.20.2 on win10 is unable to open files from networked drives

gczuczy, 2023-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Hungarian/Hungary (hu_HU)
[/code]
```

### Subproject(s) affected?

None

### Issue description

I store my FreeCAD files on a network drive on win10, and since upgrading from 0.18 to 0.20 I'm unable to open any of them, and the network drives are also completely missing from the open dialog.

Please revert the functionality so we can store our files non-locally.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

gczuczy, 2023-01-08:
Freecad error message, from previous version's leftover "open recent":

[image]

And the file is actually there:
[image]

gczuczy, 2023-01-08:
Open dialog completely missing network drives:
[image]

gczuczy, 2023-01-08:
The save dialog is also completely missing the network drives:
[image]

(The "network" section is expanded here as well)

gczuczy, 2023-01-08:
Uhm, this might just be half an issue. FreeCAD was running by starting it right from the installer. After closing it, and normally opening it, everything got back to normal. I guess when FreeCAD is started from the installer, it might be running as a different user or with different privileges. Still very confusing, I think this capability should either be fixed, or not possible, disabled.

closed 2023-01-08 by gczuczy

luzpaz, 2023-01-08:
@donovaly is this a known issue for windows installer ?

donovaly, 2023-01-09:
> FreeCAD was running by starting it right from the installer.

Do you install for the current user only (this is the default) or for all users? When starting FC from the installer, FC is of course executed from the user that executed the installer.

However, after an upgrade you cannot assure that the recent documents will be found. Since FC 0.18 there were many changes and fixes.

Note that there was a bug in the Windows installer, please assure that you use the installer which is online since the last 24 hours. If you installed FC earlier, please install FC again 8over the existing installation using the new installer that is online.

gczuczy, 2023-01-09:
I've installed it for all users, not just for the current user. I've just downloaded the installer a couple of minutes before opening this issue, so I guess it was the fixed one you are referring to.

And my guess is also that due to being installed for all users requiring elevated privileges through UAC, the process inherited the administrator privs, and that caused the issue.