FreeCAD/FreeCAD #8143: Periodic BSpline

Bernard-19, 2023-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31432 (Git)
Build type: Release
Branch: master
Hash: 63a2855751d638bf27ed969dc60654bda2b54baf
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

Hello,
I defining the degree of a Periodic BSpline with the shortcut by the D key, I enter the value 8 for example with 6 points and FreeCAD enters an infinite calculation.

I tested with a BSpline by control points and the result is correct. The degree displayed is 8.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-08:
@AjinkyaDahale assigning this to you initially just to get the ball rolling.

AjinkyaDahale, 2023-01-08:
@luzpaz I'm aware of this issue. I had asked the poster to make this issue.

AjinkyaDahale, 2023-01-10:
As of the following version, the infinite loop no longer happens. However, instead the entire system is marked underconstrained and cannot be moved.
```
OS: Ubuntu Core 20 (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31529 (Git) Snap 540
Build type: Release
Branch: master
Hash: a1e295e92ddbf10d0c42bf5b6e6f8ceb5f282e77
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
```

```
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31535 (Git)
Build type: Release
Branch: sketcher-fix-8143
Hash: 0439b4dca94b308b22f0f42ae8f0317d12888a9f
Python 3.10.8, Qt 5.15.7, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.5.12
  * Curves.backup1662220878.4365516 0.4.4
  * ExtremeProDark 2.5.3
  * ProDarkThemePreferencePack 1.0.0
  * ProDarkThemePreferencePack.backup1662220881.5889003 1.0.0
```

closed 2023-01-16 by abdullahtahiriyo