FreeCAD/FreeCAD #8144: [Issue]

CNC7125, 2023-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=13&t=75087

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31513 (Git) AppImage
Build type: Release
Branch: master
Hash: b2ab8edba4bfd71681e639f8c3f1105066bed4c7
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly4 0.12.4
  * TabBar
  * Assembly4.backup1662904298.038576
  * A2plus 0.4.55b
[/code]
```

### Subproject(s) affected?

Techdraw

### Issue description

I drew a cabinet, then passed it to TechDraw as a view. Now I wanted to adjust the scale so that the drawing fits on a Din A3 sheet. To do this, I went into the formula editor under "View" and entered the scale 1:10, confirmed with OK and finished with the Update button. And now FreeCAD crashes. The file and a movie is already stored in the FreeCAD forum.

### Anything else?

[FreeCAD_stürzt_ab.webm](https://user-images.githubusercontent.com/122218527/211204948-19e226f2-16ee-40c7-a28c-c213b03aed8e.webm)
[Einbauschrank_01.zip](https://github.com/FreeCAD/FreeCAD/files/10368768/Einbauschrank_01.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-01-08:
No crash here on 3 different versions.  

[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31527 +2 (Git)
Build type: debug
Branch: fixViewDirectionPrecision
Hash: 4c868e05121d35e157171df6e9acc1999d47ca92
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk , OCC 7.6.3
Locale: German/Germany (de_DE) [ OS: English/Canada (en_CA) ]
Installed mods: 
  * sheetmetal 0.2.59
  * FreeCAD_Electric
  * Assembly4 0.12.4
  * A2plus 0.4.56a
[/code]

[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31413 (Git)
Build type: Unknown
Branch: master
Hash: aff217b2141c28306b0dcf1e1f5399447d9c4a3f
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: German/Germany (de_DE) [ OS: English/Canada (en_CA) ]
Installed mods: 
  * sheetmetal 0.2.59
  * FreeCAD_Electric
  * Assembly4 0.12.4
  * A2plus 0.4.56a
[/code]

[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31527 (Git) AppImage
Build type: Release
Branch: master
Hash: 664b2560adcd9c9ab012e4f8c0fb92143c202fe9
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE) [ OS: English/Canada (en_CA) ]
Installed mods: 
  * sheetmetal 0.2.59
  * FreeCAD_Electric
  * Assembly4 0.12.4
  * A2plus 0.4.56a
[/code]

luzpaz, 2023-01-09:
@CNC7125 do you mind starting FreeCAD from the CLI, reproducing the crash, and pasting the crash output back here?

closed 2023-01-09 by wwmayer

WandererFan, 2023-01-09:
@wwmayer: any idea why this would not crash for me?

wwmayer, 2023-02-15:
@WandererFan in case it's still relevant then write a test function:
```
TopoDS_Shape shape;
BRep_Tool::IsClosed(shape);
```
If it crashes for you then it can be excluded that a different OCC version shows different behaviour. So, then you must be very lucky that none of the wires were null. But to figure out why this happens wouldn't be easy.

CNC7125, 2023-02-15:
Hello all,
I installed the latest version and tried to reconstruct the error,
everything fits, runs as it should.

commit a963ca5f (2023-01-09): TD: fixes #8144: TD crash