FreeCAD/FreeCAD #8149: [Feature Request] Path helix op - add option to enforce single helix

itain, 2023-01-08:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=74977

### Subproject(s) affected?

Path

### Idea description

For large **through** circular holes (unlike pockets) we need to process only the outer-most helix.
This option will avoid clearing the center of the hole.

Small holes (with hole diameter <= 2*tool diameter) should not be affected.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sslupsky, 2024-11-07:
This would be a useful feature.  I attempted to set the Starting Radius property to the diameter of a hole minus the radius of the tool.  This generates a single tool path within the hole but unfortunately it also generates a second tool path at the face of the hole.

sslupsky, 2024-11-07:
Here is an illustration of what I mean by the second tool path at the face of the hole:

[image]

tarman3, 2025-07-04:
#21971