FreeCAD/FreeCAD #8150: [Issue] Path Dressup LeadInOut should  start at clearance height

itain, 2023-01-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Fedora Linux 37 (KDE Plasma) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31527 (Git)
Build type: Release
Branch: master
Hash: 664b2560adcd9c9ab012e4f8c0fb92143c202fe9
Python 3.11.1, Qt 5.15.7, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * lattice2 1.0.0
  * kicadStepUpMod 10.16.5
  * fasteners 0.4.51
  * freecad.gears 1.0.0
[/code]
```

### Subproject(s) affected?

None

### Issue description

Path profile starts ON with a G0 Z<clearance height>, then moves to starting XY location.
When I apply dressup LeadInOut this changes to a single G0 to XYZ.
This could be dangerous depending on the initial tool location.

See attached images.

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

silopolis, 2023-01-09:
>
> Issue description
>
> Path profile starts ON with a G0 Z, then moves to starting XY location.
> When I apply dressup LeadInOut this changes to a single G0 to XYZ.
> This could be dangerous depending on the initial tool location.
>

I believe this is the right behaviour, but some kind of "force retract"
option could be useful.

itain, 2023-01-09:
I think all operations should start with a force retract to safety height.
In this case there was a a retract to safety height, but then the lead-in dressup added a new move before it.

silopolis, 2023-01-09:
Le lun. 9 janv. 2023 à 15:42, Itai Nahshon ***@***.***> a
écrit :

> I think all operations should start with a force retract to safety height.
> In this case there was a a retract to safety height, but then the lead-in
> dressup added a new move before it.
>

You're absolutely right!
A "keep tool down" option would be much more secure 👍👌

rmu75, 2023-01-23:
Maybe a related issue https://forum.freecadweb.org/viewtopic.php?f=15&p=655340&t=75465&sid=8559335ea424cac776e4922b05fa05b3

Profiling inside, LeadInOut and keep tool down option on results in linking moves at depth, I'm pretty sure that makes no sense and is indeed dangerous.

[image]

sliptonic, 2023-02-22:
I'm seeing quite a few issues related to the LeadInOut dressup.  This dressup is some old code and the developer who wrote it is deceased.  I would suggest that this dressup be refactored.  

As a first step, the GUI and dressup logic could be (and should be) separated.   This will make the dressup conform to the same pattern as the boundary, dogbone, and tag dressups.  It'll simplify the code, reduce redundancy, and make it easier to work on.

The actual code that generates the leadIn and leadout commands could be further abstracted to generators. These would be pure functions that take a small number of arguments and return a set of commands.  It would then be the job of the dressup to call the generator and add the resulting commands at the right place.

luzpaz, 2025-06-24:
> PS UI panel totally broken in my PR and I will not fix this  

What do you mean here ? Your PR is breaking the UI panel or it was already broken ?

sliptonic, 2025-11-29:
Closing this per #22669.   Will reopen if any complaints remain.

closed 2025-11-29 by sliptonic