FreeCAD/FreeCAD #8158: [Issue] Using [Part Design]->[Create a Subtractive Feature]->[Groove] on a circular face causes a freeze

Kribylet, 2023-01-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

I created a simple revolved part from a sketch following a youtube tutorial, then tried to apply the subtractive groove feature to a circular face of the generated body. This softkilled my OS as FreeCAD ate up all computation resources.

Reproduction:
1. Open the included project.
2. Using the Part Design workbench, click the circular face of the wider diameter ("top") section of the model.
3. Make sure you saved everything else you've been up to today.
4. Navigate to [Part Design]->[Create a Subtractive Feature]->[Groove] in the context menu, and hit OK.
5. Be logic bombed.

### Anything else?

[subtractive_groove_crash_issue.zip](https://github.com/FreeCAD/FreeCAD/files/10380318/subtractive_groove_crash_issue.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-11-17:
@xtemp09 can you reproduce on latest stable and dev ?

xtemp09, 2023-11-18:
```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34935 (Git)
Build type: Debug
Branch: main
Hash: e235a1b795567a99d67e1887a1c7151e60ebe395
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: Russian/Russia (ru_RU)
```

This version reports:

```
12:41:20  Groove: Cut out of base feature failed
```

----

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

This version reports:

```
12:43:21  Recompute failed!
```

----

No crash occurs in the both versions.

xtemp09, 2023-11-18:
This version crashlessly reports:

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35046 (Git)
Build type: Release
Branch: main
Hash: 6a66034ce27290e7b385a933d77375f55492d194
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

```
12:52:10  Groove: Cut out of base feature failed
```

----

This version crashlessly reports:

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)

```

```
12:54:18  Recompute failed!
```

xtemp09, 2023-11-18:
No crash takes place. No freeze occurs.
FreeCAD remains responsive.

> This softkilled my OS as FreeCAD ate up all computation resources.

> Be logic bombed.

What's "softkill?" Who "logic bombs?"
OOMs? Segfaults? Stack overflows?
Your words are quite evasive.

Kribylet, 2023-11-18:
The entire OS froze up because all CPU cores were dedicated to some FreeCAD
task related to the groove operation. It was reproducible on my system,
though it obviously involves more than FreeCAD when the entire system is
allowed to lock up. That said, nothing similar has ever happened to me in
other applications.

xtemp09, 2023-11-18:
@Kribylet, can you reproduce the problem with >0.20.2?

Kribylet, 2023-11-18:
I will try to give it a go by tomorrow (I'm unable to today as I am moving
into a new residence).

Kribylet, 2023-11-28:
I have tested this using the 0.21.1 x86 appimage and the total OS freeze did not occur, rather the operation failed with a "recompute failed!" message. Apologies for the delayed response, it was due to factors outside my control. There may or may not be an issue with whether this operation should be computable, but with regards to the crash this issue seems resolved.

luzpaz, 2023-11-28:
@xtemp09 should we rename ticket then since the issue has changed in latest stable ?

xtemp09, 2023-11-29:
It is for the topic-starter to decide. I don't see any issue. 🤷‍♂️

luzpaz, 2023-11-29:
Why 

> This version crashlessly reports:
> 
> ```ruby
> Version: 0.22.0dev.35046 (Git)
> ```
> 
> ```
> 12:52:10  Groove: Cut out of base feature failed
> ```
> 
> This version crashlessly reports:
> 
> ```ruby
> Version: 0.21.1.33668 +26 (Git)
> ```
> 
> ```
> 12:54:18  Recompute failed!
> ```

@xtemp09 any idea why we're seeing these failures ?

luzpaz, 2023-11-29:
@FlachyJoe https://github.com/FreeCAD/FreeCAD/issues/8158#issuecomment-1831803153 

Any idea ?

FlachyJoe, 2023-11-29:
This red shape (top face revolution) has no intersection with the cylinder below. The same occurs in the test file.
[image]

closed 2025-01-12 by github-actions[bot]