FreeCAD/FreeCAD #8160: [Issue]

berkakinci, 2023-01-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=75158

### Version

0.21 (Development)

### Full version info

```shell
Reproduced on:
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.0.31513 (Git)
Build type: Release
Branch: master
Hash: b2ab8edba4bfd71681e639f8c3f1105066bed4c7
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)

Observed in "the wild" at:
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
```

### Subproject(s) affected?

PartDesign

### Issue description

[Note this redundant info; mostly a copy of original post in forum]

I use PartDesign Pads and Pockets with taper angles for injection moldability. So taper angles are small (~5deg). I've had annoying failures when using taper angles. I think it's specific to circles/arcs. The error is:
```
Extrusion: end face of tapered against extrusion is empty
This means most probably that the against taper angle is too large or small.

```
However this error is not correct. No faces are disappearing. I will show this in the attached model. It also seems very very specific to a 2mm sketch dimension; that makes me think some "magic" non-zero value may be in a check.

I finally got around to reproducing in a simple file and reproducing on the latest weekly.

To reproduce:
- Open attached file.
- Navigate in Model view to Part->Body->Pad->Sketch
- In Data pane, change Constraints->Constraint2 from 2.01mm to 1.99mm

Expected:
Tapered cylinder updates

Observed:
The error message in report view:
```
08:33:12  Extrusion: end face of tapered against extrusion is empty
This means most probably that the against taper angle is too large or small.
08:33:12  Recompute failed! Please check report view.

```
Changing back to 2.01mm (or even 2.00001mm for me) works; 1.99mm or 2.00mm throws error. You can clearly see the face is nowhere close to disappearing.

I observed this originally on 0.20.1, then on 0.20.2. I reproduced it also on the weekly dev builds. See full version info for details.

### Anything else?

Screenshot showing taper with 2.01mm circle sketch
[image]

To reproduce, use:
[Taper_for_2mm.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10387219/Taper_for_2mm.FCStd.zip)

This is a workaround demo:
[Taper_for_2mm_2arc_workaround.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10387217/Taper_for_2mm_2arc_workaround.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

berkakinci, 2023-01-10:
I should add the same happens for tapered pockets as well.  To reproduce:
- Open attached file (Taper_for_2mm_Pocket.FCStd)
- Navigate in Model view to Part->Body->_Pocket_->Sketch
- In Data pane, change Constraints->Constraint2 from 2.01mm to 1.99mm

[Taper_for_2mm_Pocket.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10387378/Taper_for_2mm_Pocket.FCStd.zip)

GeneGH, 2023-01-11:
As pointed out in the forum discussion the problem appears to be related to the use of a circle. Other shapes appear to work correctly.

tbMaker1, 2023-03-18:
Hello, in the case of injection molding, maybe you should use PartDesign_draft where it fits, instead of pad/taper.

berkakinci, 2023-03-18:
> Hello, in the case of injection molding, maybe you should use PartDesign_draft where it fits, instead of pad/taper.

Thanks for the recommendation. I do use drafts too. However, I prefer to use the taper at the feature generation when possible. I find it a lot more robust.  With drafts, as any other dress up feature, changes after-the-fact often break the design.  That requires a lot of error prone cleanup operations late in the project.

I have had to work around issues like these by alternate approaches.  Unfortunately, in some of those cases I find the alternate approaches have related failures. I can’t remember if I tried draft here.

tbMaker1, 2023-03-18:
I am looking into this issue as part also of #8587, #8949 and #8950.

Stokestack, 2023-05-07:
I encountered this on a pretty simple shape from an imported SVG that has no circle, so not sure it's circle-dependent.

berkakinci, 2023-05-08:
> I encountered this on a pretty simple shape from an imported SVG that has no circle, so not sure it's circle-dependent. I'm on Mac OS 13.3.1 on Apple Silicon.

I imagine a lot of different possibilities for errors with geometry not created in FreeCAD. Still, you should upload an example file.

Stokestack, 2023-05-08:
Here you go. I've attached the FreeCAD file right before the extrude step that will fail, and the original SVG I imported.

One of the shapes in the SVG works, but the other doesn't. I don't know why they'd be different. [Here's a video](https://youtu.be/MBFRi7smlPY) of how to reproduce it.

[extrudeFail.zip](https://github.com/FreeCAD/FreeCAD/files/11425629/extrudeFail.zip)

grubertechnic, 2023-07-04:
When i enter <= 2mm, it crashes.
```
Err: Program received signal SIGSEGV, Segmentation fault.
Err: #0  /lib/x86_64-linux-gnu/libc.so.6(+0x38d60) [0x7fc5133fcd60]
Err: #1  0x7fc3da6ca4c9 in MAT_ListOfBisector::Current() const from /usr/local/lib/libTKTopAlgo.so.7.7+0x9
Err: #2  0x7fc3da6e03f7 in MAT2d_Mat2d::CreateMat(MAT2d_Tool2d&) from /usr/local/lib/libTKTopAlgo.so.7.7+0x36e7
Err: #3  0x7fc3da708050 in BRepMAT2d_BisectingLocus::Compute(BRepMAT2d_Explorer&, int, MAT_Side, GeomAbs_JoinType, bool) from /usr/local/lib/libTKTopAlgo.so.7.7+0xa80
Err: #4  0x7fc3db229c6a in BRepFill_OffsetWire::Init(TopoDS_Face const&, GeomAbs_JoinType, bool) from /usr/local/lib/libTKBool.so.7.7+0x19a
Err: #5  0x7fc3db22a1a0 in BRepFill_OffsetWire::BRepFill_OffsetWire(TopoDS_Face const&, GeomAbs_JoinType, bool) from /usr/local/lib/libTKBool.so.7.7+0x1a0
Err: #6  /usr/local/lib/libTKOffset.so.7.7(+0x97a8e) [0x7fc3dc5f2a8e]
Err: #7  0x7fc3dc5f3d93 in BRepOffsetAPI_MakeOffset::Perform(double, double) from /usr/local/lib/libTKOffset.so.7.7+0xbc3
Err: #8  0x7fc3dcf5fa7d in Part::ExtrusionHelper::createTaperedPrismOffset(TopoDS_Wire, gp_Vec const&, double, bool, TopoDS_Wire&) from /home/user/Programs/FreeCADmaster/build/Mod/Part/Part.so+0x21d
Err: #9  0x7fc3dcf61257 in Part::ExtrusionHelper::makeDraft(TopoDS_Shape const&, gp_Dir const&, double, double, double, double, bool, std::__cxx11::list<TopoDS_Shape, std::allocator<TopoDS_Shape> >&, bool) from /home/user/Programs/FreeCADmaster/build/Mod/Part/Part.so+0x1407
Err: #10  0x7fc3d86e47b4 in PartDesign::FeatureExtrude::generateTaperedPrism(TopoDS_Shape&, TopoDS_Shape const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, gp_Dir const&, double, double, double, double, bool) from /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/_PartDesign.so+0x144
Err: #11  0x7fc3d86f22ed in PartDesign::Pad::execute() from /home/user/Programs/FreeCADmaster/build/Mod/PartDesign/_PartDesign.so+0x165d
Err: #12  0x7fc5154ff17a in App::DocumentObject::recompute() from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0xba
Err: #13  0x7fc3dcde3369 in Part::Feature::recompute() from /home/user/Programs/FreeCADmaster/build/Mod/Part/Part.so+0x9
Err: #14  0x7fc5154ac3ed in App::Document::_recomputeFeature(App::DocumentObject*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0x29d
Err: #15  0x7fc5154b38ae in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADApp.so+0x56e
Err: #16  0x7fc516099328 in Gui::PropertyEditor::PropertyEditor::recomputeDocument(App::Document*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0x68
Err: #17  0x7fc5160993d0 in Gui::PropertyEditor::PropertyEditor::closeTransaction() from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0x70
Err: #18  0x7fc51609c9e9 in Gui::PropertyEditor::PropertyEditor::closeEditor(QWidget*, QAbstractItemDelegate::EndEditHint) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0x39
Err: #19  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2e45e0) [0x7fc513a635e0]
Err: #20  0x7fc5147658c0 in QAbstractItemDelegate::closeEditor(QWidget*, QAbstractItemDelegate::EndEditHint) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x50
Err: #21  0x7fc513a58ff1 in QObject::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x291
Err: #22  0x7fc5144ef15f in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7f
Err: #23  0x7fc515e1f618 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0x68
Err: #24  0x7fc513a2cfca in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12a
Err: #25  0x7fc513a2fa01 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x171
Err: #26  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x305e93) [0x7fc513a84e93]
Err: #27  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x25b) [0x7fc5117c9e6b]
Err: #28  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x52118) [0x7fc5117ca118]
Err: #29  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2f) [0x7fc5117ca1cf]
Err: #30  0x7fc513a8451f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x5f
Err: #31  0x7fc513a2b98b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
Err: #32  0x7fc513a33c00 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x90
Err: #33  0x7fc515d88148 in Gui::Application::runApplication() from /home/user/Programs/FreeCADmaster/build/lib/libFreeCADGui.so+0x2498
Err: #34  /home/user/Programs/FreeCADmaster/build/bin/FreeCAD(+0x98ac) [0x5564fbf788ac]
Err: #35  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xea) [0x7fc5133e7d0a]
Err: #36  /home/user/Programs/FreeCADmaster/build/bin/FreeCAD(+0x9b5a) [0x5564fbf78b5a]
```

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33487 (Git)
Build type: Release
Branch: master
Hash: 3cb5bdcbe4266b933b3c7a53055cc4212cd3cfa1
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * Assembly4 0.12.6
  * CurvedShapes 1.0.4
  * Curves 0.6.8
```

berkakinci, 2023-07-04:
That’s interesting because it could point straight at the bug.  I didn’t experience a crash IIRC. is the crash reproducible or randomly happening?One other unfortunate thing: the devs have moved on quite bit since this version. It’ll be tough to get their attention without reproducing on the new version.

view it on GitHub, or 
unsubscribe.

grubertechnic, 2023-07-04:
There are some weird cone and draft issues atm, example #5649, unexplainable.

luzpaz, 2023-11-01:
@FlachyJoe have you encountered this issue ?

berkakinci, 2024-12-23:
This bug is still reproducible on 1.0.0
I've also reproduced by starting from a new document, so it's not some artifact of what's left in the example files above.

```
OS: macOS 15.1.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

berkakinci, 2024-12-23:
The error messages in the current release version:
```
16:15:07  Extrusion: end face of tapered along extrusion is empty
This means most probably that the along taper angle is too large or small.
16:15:07  Pad: Extrusion: end face of tapered extrusion is empty
```

FEA-eng, 2025-06-08:
Still relevant

berkakinci, 2025-08-08:
Still Reproducible in:
```
OS: macOS 15.6
Architecture: arm64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * kicadStepUpMod.backup1753993021.644542 (Disabled)
  * kicadStepUpMod.backup1753993062.0308852 (Disabled)
  * kicadStepUpMod
```

luzpaz, 2025-08-09:
@FEA-eng did you test on 1.1dev ? If so please add full About info. TIA

luzpaz, 2026-09-06:
This is OCC related, right ?

ElementW, 2026-09-13:
Oh boi there's a lot to unpack here. For `Taper_for_2mm.FCStd` and `Taper_for_2mm_Pocket.FCStd`:

* On `FreeCAD_weekly-2026.09.09-Linux-x86_64.AppImage`:
  * Shape is incorrect (though geometry is valid)
    [image]
    [image]
  * Flipping the taper angles and setting constraint to `1.99 mm` emits an error (both for pad and pocket):
    ```
    19:39:45  Extrusion: end face of tapered along extrusion is empty
    This means most probably that the along taper angle is too large or small.
    ```
* On main (13b07d70e6e27ee61f8894531c22aeeb46aa93ed) with OCC 7.9.3:
  * Geo is correct and valid
  * Setting constraint to `1.99 mm` crashes:
    <details>

    ```
    #0  /usr/lib/libc.so.6(+0x40110) [0x7f6345056110]
    #1  0x7f62ec121af7 in MAT_ListOfBisector::Current() const from /usr/lib/libTKTopAlgo.so.7.9+0x7
    #2  0x7f62ec138fae in MAT2d_Mat2d::CreateMat(MAT2d_Tool2d&) from /usr/lib/libTKTopAlgo.so.7.9+0x33de
    #3  0x7f62ec16026c in BRepMAT2d_BisectingLocus::Compute(BRepMAT2d_Explorer&, int, MAT_Side, GeomAbs_JoinType, bool) from /usr/lib/libTKTopAlgo.so.7.9+0xabc
    #4  0x7f62ed3653af in BRepFill_OffsetWire::Init(TopoDS_Face const&, GeomAbs_JoinType, bool) from /usr/lib/libTKBool.so.7.9+0x1af
    #5  0x7f62ed36579e in BRepFill_OffsetWire::BRepFill_OffsetWire(TopoDS_Face const&, GeomAbs_JoinType, bool) from /usr/lib/libTKBool.so.7.9+0x14e
    #6  /usr/lib/libTKOffset.so.7.9(+0x99214) [0x7f62eeedb214]
    #7  0x7f62eeedc50e in BRepOffsetAPI_MakeOffset::Perform(double, double) from /usr/lib/libTKOffset.so.7.9+0xc3e
    #8  0x7f62ef9d41e8 in Part::BRepOffsetAPI_MakeOffsetFix::Perform(double, double) from /media/source/FreeCAD/build/Desktop-Debug/Mod/Part/Part.so+0x38
    #9  0x7f62ef9e0889 in Part::ExtrusionHelper::createTaperedPrismOffset(TopoDS_Wire, gp_Vec const&, double, bool, TopoDS_Wire&) from /media/source/FreeCAD/build/Desktop-Debug/Mod/Part/Part.so+0x1a7
    #10  0x7f62ef9e196e in Part::ExtrusionHelper::makeElementDraft(Part::ExtrusionParameters const&, Part::TopoShape const&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> >&, Base::Reference<App::StringHasher>) from /media/source/FreeCAD/build/Desktop-Debug/Mod/Part/Part.so+0xc46
    #11  0x7f62e4cc3aac in PartDesign::FeatureExtrude::generateSingleExtrusionSide(Part::TopoShape const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, double, double, App::PropertyLinkSub&, App::PropertyLinkSubList&, gp_Dir, double, bool, Part::TopoShape const&, TopLoc_Location&) from /media/source/FreeCAD/build/Desktop-Debug/Mod/PartDesign/_PartDesign.so+0x854
    #12  0x7f62e4cc18e6 in PartDesign::FeatureExtrude::buildExtrusion(Base::Flags<PartDesign::FeatureExtrude::ExtrudeOption>) from /media/source/FreeCAD/build/Desktop-Debug/Mod/PartDesign/_PartDesign.so+0x1f38
    #13  0x7f62e4ce1c94 in PartDesign::Pad::execute() from /media/source/FreeCAD/build/Desktop-Debug/Mod/PartDesign/_PartDesign.so+0x4c
    #14  0x7f634a87649c in App::DocumentObject::recompute() from /media/source/FreeCAD/build/Desktop-Debug/lib/libFreeCADApp.so+0x84
    #15  0x7f62ef74823f in Part::Feature::recompute() from /media/source/FreeCAD/build/Desktop-Debug/Mod/Part/Part.so+0x1d
    #16  0x7f62e4c41aa1 in PartDesign::Feature::recompute() from /media/source/FreeCAD/build/Desktop-Debug/Mod/PartDesign/_PartDesign.so+0x151
    #17  0x7f634a7b8e40 in App::Document::_recomputeFeature(App::DocumentObject*) from /media/source/FreeCAD/build/Desktop-Debug/lib/libFreeCADApp.so+0x1fe
    #18  0x7f634a7b6670 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /media/source/FreeCAD/build/Desktop-Debug/lib/libFreeCADApp.so+0xa92
    #19  0x7f634cb3b212 in Gui::PropertyEditor::PropertyEditor::recomputeDocument(App::Document*) from /media/source/FreeCAD/build/Desktop-Debug/lib/libFreeCADGui.so+0xa0
    #20  0x7f634cb3b3ed in Gui::PropertyEditor::PropertyEditor::closeTransaction() from /media/source/FreeCAD/build/Desktop-Debug/lib/libFreeCADGui.so+0x67
    #21  0x7f634cb3b497 in Gui::PropertyEditor::PropertyEditor::closeEditor(QWidget*, QAbstractItemDelegate::EndEditHint) from /media/source/FreeCAD/build/Desktop-Debug/lib/libFreeCADGui.so+0x8b
    #22  /usr/lib/libQt6Core.so.6(+0x1f2638) [0x7f63457f2638]
    #23  0x7f6346c9d084 in QAbstractItemDelegate::closeEditor(QWidget*, QAbstractItemDelegate::EndEditHint) from /usr/lib/libQt6Widgets.so.6+0x54
    #24  0x7f63457e88df in QObject::event(QEvent*) from /usr/lib/libQt6Core.so.6+0x13f
    #25  0x7f634698b14a in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt6Widgets.so.6+0x8a
    #26  0x7f634c3a62a9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /media/source/FreeCAD/build/Desktop-Debug/lib/libFreeCADGui.so+0x145
    #27  0x7f6345792288 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt6Core.so.6+0x168
    #28  0x7f634579617b in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /usr/lib/libQt6Core.so.6+0x22b
    #29  /usr/lib/libQt6Core.so.6(+0x44d5e7) [0x7f6345a4d5e7]
    #30  /usr/lib/libglib-2.0.so.0(+0x5aaf6) [0x7f6344304af6]
    #31  /usr/lib/libglib-2.0.so.0(+0x5e077) [0x7f6344308077]
    #32  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7f63443087cc]
    #33  0x7f6345a4cb23 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt6Core.so.6+0x73
[... cut ...]
* On main (13b07d70e6e27ee61f8894531c22aeeb46aa93ed) with OCC 8.1.0.dev1-3d097a0328:
  * Geo is correct and valid
  * Setting constraint to `1.99 mm` crashes works fine, and does so down to around `0.9 mm` where the tips would become points

@luzpaz Yes it is. This issue is miscategorised, it shouldn't be on the UI/UX Improvements board, but should be on the OCCT Liaison board instead. My guess is this is fixed by OCCT 8.0, but since I only verified for 8.1, you can set "Issue only in OCC version" to "< 8.1.0".

pieterhijma, 2026-09-30:
Marked this as Done in the OCCT Liaison board as it seems to be fixed by at least 8.1.

stefano2734, 2026-09-30:
Is marked with label 8.1.0. And this will come in January in occt plan.
So perhaps here milestone 27.1 is the better stone.

maxwxyz, 2026-11-07:
#33427 repeats this with a 2 mm circle and negative-taper Pad. MamaTierra reports 1.1.3/OCC 7.9.3 on EndeavourOS; VishalRamKN confirms on 27.1.0dev.49209/OCC 7.9.3 and corrects 2 m to 2 mm. drwho495 reproduces it on the October 6 weekly/OCC 8.0.1, so 8.0 isn't confirmed fixed. PR #33550 is still open, with review concerns about legacy files and larger circles.

maxwxyz, 2026-11-07:
Here is a fix from AstoCAD if someone wants to cherry-pick https://github.com/AstoCAD/FreeCAD/commit/64764763039029c80dc00278e1fb156bfd7e90c4

adrianinsaval, 2026-11-15:
fixed upstream: https://github.com/Open-Cascade-SAS/OCCT/pull/1453, looks like it will be included in 8.0.2, for the record this crashes on stock occt, with [blobfish patches](https://gitlab.com/blobfish/occt) (which we used up to freecad 1.1) this doesn't crash and just gives the error reported here

berkakinci, 2026-11-15:
>  this doesn't crash and just gives the error reported here
I really hope it doesn't give the error I reported in 2023, because that's a totally bogus error.
Please check closely.  The original reported problem was a real bug, reporting an error when there was no reason to fail on the taper as specified.