FreeCAD/FreeCAD #8161: [Bug] [PartDesign] Clones do not copy colour

PeterSHollander, 2023-01-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 1809
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

The problem of clones not copying colour as described in #5765 and fixed in #7885 for the Draft workbench appears to also be a problem in the Part Design workbench.  I have not made the time to review the FreeCAD source code, but am hopeful the fix may be as simple as it was for the Draft workbench.

## To reproduce:
- Colourise a part
- Ensure you're in the Part Design workbench
- Clone the colourised part
- Notice the lack of colour in the clone

It is worth noting this affects both part colour as well as face colour.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-05-04:
I believe Part Design clones intentionally do not copy color as they are meant to copy only geometry to make a new body from it. Certainly face color information is not of much use in Part Design where they will get overridden by the next feature. We may change it to copy the overall color property though, but I think this is very low priority and not worth marking as an objective for 0.21

recursivenomad, 2023-05-04:
To provide an example where Part Design Clones copying colour is preferred, I recently made a project which depends on parametric colouring by combining various monocoloured parts through boolean operations, thus parametrically persisting the face colours and allowing me to make edits to the model history without needing to manually recolour faces of the final model on every release.

Draft Clones create a non-parametric copy of geometry and colour.  Part Design Clones create a parametric copy of geometry with the added abillity to perform Part Design operations on the resulting Clone, but remove colour.  Links create a parametric link of geometry and colour and do not feature the ability to perform Part Design Operations.

Part Design Clones which create a parametric copy of geometry *and* colour would have been ideal for making iterative parts which preserve parametric colours.  Instead, as a workaround I followed a separate (and I believe unnecessarily convoluted) workflow creating multiple duplicate Links of the same source monocoloured parts to perform the same boolean operations on, but with one added part unioned here or one less part cut there.  If memory serves, it would have been significantly more straightforward to create one parametrically coloured part from boolean operations, and then clone that part to build upon it for the next iterative part as opposed to repeating the same actions in the feature tree multiple times.

So with that explanation out of the way, I would propose that parametrically perserving colour in Part Design Clones is just as important as parametrically perserving geometry, as parametric colours are (delicately) possible in FreeCAD; and in a case like the project described above, can be a defining factor of a part which needs to be carried through to its successors.

Feel free to explore the project [here](https://github.com/recursivenomad/ki-lime-pi-pico) (model source [here](https://github.com/recursivenomad/ki-lime-pi-pico/blob/main/models/RaspberryPi_Pico_Variations.FCStd)), and you can read through [BUILDING.md](https://github.com/recursivenomad/ki-lime-pi-pico/blob/main/BUILDING.md) for more thorough descriptions of parametric colouring and limitations therein.  My understanding is that this parametrically coloured workflow is fairly unconventional, so please don't hesitate to ask if you seek clarification on anything related to this workflow or project.

commit 18bf753e (2023-05-11): PD: fixes #8161: Clones do not copy colour