FreeCAD/FreeCAD #8163: [Bug] [Crash] Crash editing constraint

arcivanov, 2023-01-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Fedora Linux 37 (Workstation Edition) (KDE/default)
Word size of FreeCAD: 64-bit
Version: 0.20.1.Unknown
Build type: Unknown
Python 3.11.0rc1, Qt 5.15.5, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.0
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Sketcher

### Issue description

This for me reproduces every time.

```
Jan 11 01:07:52 hostname plasmashell[759068]: Program received signal SIGSEGV, Segmentation fault.
Jan 11 01:07:52 hostname plasmashell[759068]: #0  /lib64/libc.so.6(+0x3cb50) [0x7fdc7685fb50]
Jan 11 01:07:52 hostname plasmashell[759068]: #1  0x7fdc20afa51c in Geom2d_Curve::Value(double) const from /lib64/libTKG2d.so.7+0x1c
Jan 11 01:07:52 hostname plasmashell[759068]: #2  0x7fdc20d35889 in ChFi3d_Builder::PerformMoreThreeCorner(int, int) from /lib64/libTKFillet.so.7+0x9bf9
Jan 11 01:07:52 hostname plasmashell[759068]: #3  0x7fdc20d1c24d in ChFi3d_Builder::PerformIntersectionAtEnd(int) from /lib64/libTKFillet.so.7+0x72fd
Jan 11 01:07:52 hostname plasmashell[759068]: #4  0x7fdc20d0c2c7 in ChFi3d_Builder::PerformOneCorner(int, bool) from /lib64/libTKFillet.so.7+0x947
Jan 11 01:07:52 hostname plasmashell[759068]: #5  0x7fdc20cc8f31 in ChFi3d_Builder::PerformFilletOnVertex(int) from /lib64/libTKFillet.so.7+0x4a1
Jan 11 01:07:52 hostname plasmashell[759068]: #6  0x7fdc20ccbcd6 in ChFi3d_Builder::Compute() from /lib64/libTKFillet.so.7+0xae6
Jan 11 01:07:52 hostname plasmashell[759068]: #7  0x7fdc20e19fcb in BRepFilletAPI_MakeFillet::Build() from /lib64/libTKFillet.so.7+0x2b
Jan 11 01:07:52 hostname plasmashell[759068]: #8  0x7fdc212227ba in PartDesign::Fillet::execute() from /usr/lib64/freecad/lib64/_PartDesign.so+0x67a
Jan 11 01:07:52 hostname plasmashell[759068]: #9  0x7fdc78dd2eb2 in App::DocumentObject::recompute() from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x462
Jan 11 01:07:52 hostname plasmashell[759068]: #10  0x7fdc1bc0f41f in Part::Feature::recompute() from /usr/lib64/freecad/lib64/Part.so+0xf
Jan 11 01:07:52 hostname plasmashell[759068]: #11  0x7fdc78d7ed4c in App::Document::_recomputeFeature(App::DocumentObject*) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x22c
Jan 11 01:07:52 hostname plasmashell[759068]: #12  0x7fdc78d82478 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x558
Jan 11 01:07:52 hostname plasmashell[759068]: #13  0x7fdc78e0792f in App::DocumentPy::recompute(_object*) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x2bf
Jan 11 01:07:52 hostname plasmashell[759068]: #14  0x7fdc78e07a27 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /usr/lib64/freecad/lib64/libFreeCADApp.so+0x27
Jan 11 01:07:52 hostname plasmashell[759068]: #15  /lib64/libpython3.11.so.1.0(+0x1d8768) [0x7fdc785d8768]
Jan 11 01:07:52 hostname plasmashell[759068]: #16  /lib64/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x73) [0x7fdc785ba393]
Jan 11 01:07:52 hostname plasmashell[759068]: #17  /lib64/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x912) [0x7fdc785c2ff2]
Jan 11 01:07:52 hostname plasmashell[759068]: #18  /lib64/libpython3.11.so.1.0(+0x1bedaa) [0x7fdc785bedaa]
Jan 11 01:07:52 hostname plasmashell[759068]: #19  /lib64/libpython3.11.so.1.0(PyEval_EvalCode+0xac) [0x7fdc786493dc]
Jan 11 01:07:52 hostname plasmashell[759068]: #20  /lib64/libpython3.11.so.1.0(+0x268203) [0x7fdc78668203]
Jan 11 01:07:52 hostname plasmashell[759068]: #21  /lib64/libpython3.11.so.1.0(+0x26473a) [0x7fdc7866473a]
Jan 11 01:07:52 hostname plasmashell[759068]: #22  /lib64/libpython3.11.so.1.0(PyRun_StringFlags+0x6d) [0x7fdc7865910d]
Jan 11 01:07:52 hostname plasmashell[759068]: #23  0x7fdc78b0558c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/lib64/freecad/lib64/libFreeCADBase.so+0x6c
Jan 11 01:07:52 hostname plasmashell[759068]: #24  0x7fdc796a767a in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x9a
Jan 11 01:07:52 hostname plasmashell[759068]: #25  0x7fdc796a7814 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /usr/lib64/freecad/lib64/libFreeCADGui.so+0xe4
Jan 11 01:07:52 hostname plasmashell[759068]: #26  0x7fdc796af366 in Gui::Command::updateActive() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x46
Jan 11 01:07:52 hostname plasmashell[759068]: #27  0x7fdbf90e9f03 in SketcherGui::tryAutoRecompute(Sketcher::SketchObject*, bool&) from /usr/lib64/freecad/lib64/SketcherGui.so+0x73
Jan 11 01:07:52 hostname plasmashell[759068]: #28  0x7fdbf90e9fa2 in SketcherGui::tryAutoRecompute(Sketcher::SketchObject*) from /usr/lib64/freecad/lib64/SketcherGui.so+0x22
Jan 11 01:07:52 hostname plasmashell[759068]: #29  0x7fdbf91ee4c2 in SketcherGui::EditDatumDialog::accepted() from /usr/lib64/freecad/lib64/SketcherGui.so+0x382
Jan 11 01:07:52 hostname plasmashell[759068]: #30  /lib64/libQt5Core.so.5(+0x2dbdaf) [0x7fdc770dbdaf]
Jan 11 01:07:52 hostname plasmashell[759068]: #31  0x7fdc77fbe428 in QDialogPrivate::finalize(int, int) from /lib64/libQt5Widgets.so.5+0x58
Jan 11 01:07:52 hostname plasmashell[759068]: #32  /lib64/libQt5Core.so.5(+0x2dbdaf) [0x7fdc770dbdaf]
Jan 11 01:07:52 hostname plasmashell[759068]: #33  /lib64/libQt5Widgets.so.5(+0x350020) [0x7fdc77f50020]
Jan 11 01:07:52 hostname plasmashell[759068]: #34  /lib64/libQt5Core.so.5(+0x2dbdaf) [0x7fdc770dbdaf]
Jan 11 01:07:52 hostname plasmashell[759068]: #35  0x7fdc77ea31d6 in QAbstractButton::clicked(bool) from /lib64/libQt5Widgets.so.5+0x46
Jan 11 01:07:52 hostname plasmashell[759068]: #36  /lib64/libQt5Widgets.so.5(+0x2a344e) [0x7fdc77ea344e]
Jan 11 01:07:52 hostname plasmashell[759068]: #37  0x7fdc77ea5409 in QAbstractButton::click() from /lib64/libQt5Widgets.so.5+0x89
Jan 11 01:07:52 hostname plasmashell[759068]: #38  0x7fdc77fbf33c in QDialog::keyPressEvent(QKeyEvent*) from /lib64/libQt5Widgets.so.5+0x20c
[... 64 lines cut ...]
Jan 11 01:07:52 hostname plasmashell[759068]: #103  0x7fdc795f44b0 in Gui::Application::runApplication() from /usr/lib64/freecad/lib64/libFreeCADGui.so+0x1f10
Jan 11 01:07:52 hostname plasmashell[759068]: #104  /usr/bin/FreeCAD(+0x4f51) [0x55dc0d2aaf51]
Jan 11 01:07:52 hostname plasmashell[759068]: #105  /lib64/libc.so.6(+0x27510) [0x7fdc7684a510]
Jan 11 01:07:52 hostname plasmashell[759068]: #106  /lib64/libc.so.6(__libc_start_main+0x89) [0x7fdc7684a5c9]
Jan 11 01:07:52 hostname plasmashell[759068]: #107  /usr/bin/FreeCAD(+0x5f65) [0x55dc0d2abf65]
Jan 11 01:07:52 hostname plasmashell[759068]: FreeCAD 0.20.1, Libs: 0.20.1RUnknown
Jan 11 01:07:52 hostname plasmashell[759068]: © Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
Jan 11 01:07:52 hostname plasmashell[759068]: FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
Jan 11 01:07:52 hostname plasmashell[759068]: FreeCAD wouldn't be possible without FreeCAD community.
Jan 11 01:07:52 hostname plasmashell[759068]:   #####                 ####  ###   ####
Jan 11 01:07:52 hostname plasmashell[759068]:   #                    #      # #   #   #
Jan 11 01:07:52 hostname plasmashell[759068]:   #     ##  #### ####  #     #   #  #   #
Jan 11 01:07:52 hostname plasmashell[759068]:   ####  # # #  # #  #  #     #####  #   #
Jan 11 01:07:52 hostname plasmashell[759068]:   #     #   #### ####  #    #     # #   #
Jan 11 01:07:52 hostname plasmashell[759068]:   #     #   #    #     #    #     # #   #  ##  ##  ##
Jan 11 01:07:52 hostname plasmashell[759068]:   #     #   #### ####   ### #     # ####   ##  ##  ##
```

### Anything else?

This is possibly related to #6219, which was closed as unable to reproduce.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

arcivanov, 2023-01-11:
To reproduce edit constraint Sketch017.Constraint36 from `3` to `2` and press enter:

[image]

[vone_carrier_jensen.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10389440/vone_carrier_jensen.FCStd.zip)

arcivanov, 2023-01-11:
GDB bt. Looks like the issue is in OpenCascade.

```
Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
Geom2d_Curve::Value (this=this@entry=0x0, U=1.1831007915757421) at /usr/src/debug/opencascade-7.5.0-7.fc37.x86_64/src/gp/gp_XY.lxx:22
22      inline gp_XY::gp_XY () : x(0.), y(0.) { } 
(gdb) bt
#0  Geom2d_Curve::Value (this=this@entry=0x0, U=1.1831007915757421) at /usr/src/debug/opencascade-7.5.0-7.fc37.x86_64/src/gp/gp_XY.lxx:22
#1  0x00007fff6a335889 in ChFi3d_Builder::PerformMoreThreeCorner (this=0x7fffffff8080, Jndex=<optimized out>, nconges=1) at /usr/src/debug/opencascade-7.5.0-7.fc37.x86_64/src/ChFi3d/ChFi3d_Builder_CnCrn.cxx:2230
#2  0x00007fff6a31c24d in ChFi3d_Builder::PerformIntersectionAtEnd (this=0x7fffffff8080, Index=5) at /usr/src/debug/opencascade-7.5.0-7.fc37.x86_64/src/ChFi3d/ChFi3d_Builder_C1.cxx:2326
#3  0x00007fff6a30c2c7 in ChFi3d_Builder::PerformOneCorner (this=0x7fffffff8080, Index=5, thePrepareOnSame=<optimized out>) at /usr/src/debug/opencascade-7.5.0-7.fc37.x86_64/src/ChFi3d/ChFi3d_Builder_C1.cxx:788
#4  0x00007fff6a2c8f31 in ChFi3d_Builder::PerformFilletOnVertex (this=0x7fffffff8080, Index=5) at /usr/src/debug/opencascade-7.5.0-7.fc37.x86_64/src/ChFi3d/ChFi3d_Builder.cxx:731
#5  0x00007fff6a2cbcd6 in ChFi3d_Builder::Compute (this=0x7fffffff8080) at /usr/src/debug/opencascade-7.5.0-7.fc37.x86_64/src/ChFi3d/ChFi3d_Builder.cxx:294
#6  0x00007fff6a419fcb in BRepFilletAPI_MakeFillet::Build (this=this@entry=0x7fffffff8030) at /usr/src/debug/opencascade-7.5.0-7.fc37.x86_64/src/BRepFilletAPI/BRepFilletAPI_MakeFillet.cxx:533
#7  0x00007fff6a7a17ba in PartDesign::Fillet::execute (this=0x5555596a5c10) at /usr/src/debug/freecad-0.20.1-1.fc37.1.x86_64/src/Mod/PartDesign/App/FeatureFillet.cpp:110
#8  0x00007ffff69d2eb2 in App::DocumentObject::recompute (this=0x5555596a5c10) at /usr/src/debug/freecad-0.20.1-1.fc37.1.x86_64/src/App/DocumentObject.cpp:147
#9  0x00007fff6940f41f in Part::Feature::recompute (this=<optimized out>) at /usr/src/debug/freecad-0.20.1-1.fc37.1.x86_64/src/Mod/Part/App/PartFeature.cpp:101
#10 0x00007ffff697ed4c in App::Document::_recomputeFeature (this=0x555558924460, Feat=0x5555596a5c10) at /usr/src/debug/freecad-0.20.1-1.fc37.1.x86_64/src/App/Document.cpp:3781
#11 0x00007ffff6982478 in App::Document::recompute (this=0x555558924460, objs=..., force=<optimized out>, hasError=0x0, options=<optimized out>) at /usr/src/debug/freecad-0.20.1-1.fc37.1.x86_64/src/App/Document.cpp:3522
#12 0x00007ffff6a0792f in App::DocumentPy::recompute (this=0x555557f210f0, args=<optimized out>) at /usr/src/debug/freecad-0.20.1-1.fc37.1.x86_64/src/App/DocumentPyImp.cpp:531
#13 0x00007ffff6a07a27 in App::DocumentPy::staticCallback_recompute (self=<App.Document at remote 0x555557f210f8>, args=<optimized out>) at /usr/src/debug/freecad-0.20.1-1.fc37.1.x86_64/redhat-linux-build/src/App/DocumentPy.cpp:1781
#14 0x00007ffff61d8768 in cfunction_call (func=<built-in method recompute of App.Document object at remote 0x555557f210f8>, args=<optimized out>, kwargs=<optimized out>)
    at /usr/src/debug/python3.11-3.11.1-1.fc37.x86_64/Objects/methodobject.c:553
#15 0x00007ffff61ba393 in _PyObject_MakeTpCall (tstate=0x7ffff6520d38 <_PyRuntime+166328>, callable=<built-in method recompute of App.Document object at remote 0x555557f210f8>, args=<optimized out>, nargs=0, keywords=<optimized out>)
    at /usr/src/debug/python3.11-3.11.1-1.fc37.x86_64/Objects/call.c:214
#16 0x00007ffff61c2ff2 in _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=<optimized out>, throwflag=<optimized out>) at /usr/src/debug/python3.11-3.11.1-1.fc37.x86_64/Python/ceval.c:4772
#17 0x00007ffff61bedaa in _PyEval_EvalFrame (throwflag=0, frame=0x7ffff7f94020, tstate=0x7ffff6520d38 <_PyRuntime+166328>) at /usr/src/debug/python3.11-3.11.1-1.fc37.x86_64/Include/internal/pycore_ceval.h:73
#18 _PyEval_Vector (tstate=0x7ffff6520d38 <_PyRuntime+166328>, func=<optimized out>, locals=<optimized out>, args=<optimized out>, argcount=<optimized out>, kwnames=<optimized out>)
    at /usr/src/debug/python3.11-3.11.1-1.fc37.x86_64/Python/ceval.c:6435
#19 0x00007ffff62493dc in PyEval_EvalCode (co=<code at remote 0x7fff653457a0>, globals=<optimized out>, Python Exception <class 'gdb.error'>: There is no member named dk_size.
locals=) at /usr/src/debug/python3.11-3.11.1-1.fc37.x86_64/Python/ceval.c:1154
#20 0x00007ffff6268203 in run_eval_code_obj (tstate=0x7ffff6520d38 <_PyRuntime+166328>, co=0x7fff653457a0, Python Exception <class 'gdb.error'>: There is no member named dk_size.
globals=, Python Exception <class 'gdb.error'>: There is no member named dk_size.
locals=) at /usr/src/debug/python3.11-3.11.1-1.fc37.x86_64/Python/pythonrun.c:1714
#21 0x00007ffff626473a in run_mod (mod=<optimized out>, filename=<optimized out>, Python Exception <class 'gdb.error'>: There is no member named dk_size.
globals=, Python Exception <class 'gdb.error'>: There is no member named dk_size.
locals=, flags=<optimized out>, arena=<optimized out>) at /usr/src/debug/python3.11-3.11.1-1.fc37.x86_64/Python/pythonrun.c:1735
#22 0x00007ffff625910d in PyRun_StringFlags (str=<optimized out>, start=<optimized out>, Python Exception <class 'gdb.error'>: There is no member named dk_size.
globals=, Python Exception <class 'gdb.error'>: There is no member named dk_size.
locals=, flags=0x0) at /usr/src/debug/python3.11-3.11.1-1.fc37.x86_64/Python/pythonrun.c:1605
[... cut ...]
#114 0x00007ffff4caed92 in QCoreApplication::exec () at ../../include/QtCore/../../src/corelib/global/qflags.h:121
#115 0x00007ffff515fbe0 in QGuiApplication::exec () at kernel/qguiapplication.cpp:1863
#116 0x00007ffff59aec89 in QApplication::exec () at kernel/qapplication.cpp:2829
#117 0x00007ffff71f44b0 in Gui::Application::runApplication () at /usr/src/debug/freecad-0.20.1-1.fc37.1.x86_64/src/Gui/Application.cpp:2278
#118 0x0000555555558f51 in main (argc=<optimized out>, argv=<optimized out>) at /usr/src/debug/freecad-0.20.1-1.fc37.1.x86_64/src/Main/MainGui.cpp:290
```

luzpaz, 2023-01-11:
Note: You're running OCC 7.5.0  

I can't confirm this ticket at this time. drmacro on the forums runs OCC v7.8.0 (let me see if I can ping him here - need to find his github handle first)

arcivanov, 2023-01-11:
"I am" not, Fedora 37 does. 
Is the packager of FreeCAD in Fedora 37 a FreeCAD or Fedora team member?

luzpaz, 2023-01-11:
CC @hobbes1069 @PrzemoF

hobbes1069, 2023-01-11:
I'm working on getting things updated. We will be stuck on OCCT 7.6.3 until tbb is updated in Fedora but hopefully that fixes most if not all the current issues.

arcivanov, 2023-01-11:
I filed a corresponding Fedora bug

https://bugzilla.redhat.com/show_bug.cgi?id=2160161

luzpaz, 2023-01-14:
Is this related to #8104 by chance?

arcivanov, 2023-01-17:
> Is this related to #8104 by chance?

Nope, entirely different stacktrace.

luzpaz, 2023-01-17:
@AjinkyaDahale can you confirm?

AjinkyaDahale, 2023-01-17:
Cannot reproduce with:
```
OS: Manjaro Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31599 (Git)
Build type: Release
Branch: sketcher-knot-to-endpoint-tangent
Hash: 12ed13ddbea126b40546d5b3d67fdeb6abeeece9
Python 3.10.8, Qt 5.15.7, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods:
  * Curves 0.5.12
  * Curves.backup1662220878.4365516 0.4.4
  * ExtremeProDark 2.5.3
  * ProDarkThemePreferencePack 1.0.0
  * ProDarkThemePreferencePack.backup1662220881.5889003 1.0.0
```
There seems to be some toponaming issues happening as well:
[image]
[image]

@arcivanov can you try with RealThunder's branch?

arcivanov, 2023-01-17:
This might be the function of old OCC on Fedora 37 as well.

arcivanov, 2023-01-17:
Yours:
> Python 3.10.8, Qt 5.15.7, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3

Mine:
> Python 3.11.0rc1, Qt 5.15.5, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.0

AjinkyaDahale, 2023-01-17:
> Yours:
> 
> > Python 3.10.8, Qt 5.15.7, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
> 
> Mine:
> 
> > Python 3.11.0rc1, Qt 5.15.5, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.0

The solution would then be to somehow push for newer OCC in Fedora. Could you try getting the snap or appimage as well?

arcivanov, 2023-01-17:
Well, I've switched to RealThunder's Link Daily anyway due to other bugs as well, but yes, Fedora distro is on OCC 7.5.0, which hasn't been updated in 2 years and the bug filed for it in Fedora has not been responded yet by either the OCC or FreeCAD team.

luzpaz, 2023-01-17:
If this is an OCC issue (most likely) this is then a downstream issue. The Fedora OCC package maintainers are now needed to do the work necessary (no small task since OCC is a dependency for other packages, no doubt. There is little that FreeCAD folk can do.

arcivanov, 2023-01-17:
That's fine, the issue can be closed then.

closed 2023-01-17 by arcivanov

luzpaz, 2023-01-17:
@hobbes1069 / @PrzemoF sorry to ping y'all again, but would it be possible for you to look in to this?
(https://github.com/FreeCAD/FreeCAD/issues/8163#issuecomment-1384768645)