FreeCAD/FreeCAD #8168: [Problem] Text in Spreadsheet Is Black in Dark Mode

hmturnbull, 2023-01-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Problem description

The spreadsheet has black text on a dark background, and the only way I've been able to change the text colour is to set it to white <em>one cell at a time</em>, which is extremely time-consuming.  It goes without saying that dark text on a dark background is an accessibility nightmare.

A fix would be much appreciated.

NOTE:
- I did check for issues related to this, but I didn't find anything; apologies in advance if I missed an existing issue.

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-11:
What theme/preference-pack are you using?

hmturnbull, 2023-01-12:
> What theme/preference-pack are you using?

Behave-dark

luzpaz, 2023-01-12:
@hmturnbull your [About](https://wiki.freecad.org/About) info should show the version of behave-dark you have installed. Odd.

@Chrismettal care to look in to OP?

Chrismettal, 2023-01-12:
Briefly for now, might have time later to check the actual code:

All dark themes exhibit this issue currently. The text color is not defined via stylesheets or preference packs, and is instead chosen via the bar at the top. 

The DEFAULT color is black for all stylesheets, but can be changed individually. Additionally, only the default cell color is themed by the stylesheet, but can be overridden with the bar at the top as well.

The problem we would face, implementing these in PreferencePacks is, what to do when the stylesheet changes on a finished table? What do we do with all the colors set manually by the user?

Potential solutions include:
- Have the default font be "Automatic" and have it switch between a dark and light color depending on the actual cell color background (Like Microsoft Office) but not care about the chosen theme
- Have the default font be enforced by the PreferencePack. Overriding the color manually would make the chosen color stick forever, but without choosing anything, the PreferencePack would provide its color on pack change.
- Have the table be light in any theme (Not preferred, for obvious reasons)

I would like to see the second one implemented, but that would mean updating the code to read a new value from the preference pack, and have that value set for all preference packs. 
Solution one would be easier and faster to implement, but the color won't neccessarily match the theme fully.

--- 

Additionally, I think on most if not all dark themes there is no highlighting of the currently chosen cell. Not sure if that's neccessarily related.

luzpaz, 2023-01-13:
Thanks Chrismettal!  
Linked this ticket to #6805

MisterMakerNL, 2023-03-03:
@Chrismettal 
I don't know anymore how I fixed it....
Only the alias fix is in the cfg file of ExtremeProDark
[image]

luzpaz, 2023-03-15:
@MisterMakerNL can you link to the change you made in ExtremeProDark when you get a chance ?

MisterMakerNL, 2023-03-15:
@luzpaz 
It could be simple fix with below code:
```
QTableView::item {
    color: white;
}
```

luzpaz, 2023-04-02:
What's left here ?

cfunseth, 2024-01-31:
I just wanted to comment that I was struggling with this issue as well. I added the code from @MisterMakerNL to my Behave-dark.qss and it solved the issue upon rebooting FreeCAD:

[image]

If the acceptable fix is to simply add this code to each dark style sheet, I can certainly help and submit a PR. (Though my coding skills are basic at best)

luzpaz, 2024-02-10:
Any follow-up here ?

cfunseth, 2024-02-13:
> Any follow-up here ?

I'm not sure what MisterMakerNL had in mind, but figured I should note that by putting the "color: white" code in - the "Set cell(s) foreground color" option in the Spreadsheet workbench no longer functions properly. I can still set the background color, but no matter what I select for the foreground color it just stays white.

maxwxyz, 2024-03-09:
@obelisk79 could you weigh in, its styled in OpenDark too.
I guess this could be closed.

MisterMakerNL, 2024-03-09:
> I'm not sure what @MisterMakerNL had in mind, but figured I should note that by putting the "color: white" code in the "Set cell(s) foreground color" option in the Spreadsheet workbench no longer [...]

Seems indeed that the coloring options are a bit broken with a stylesheet on.
But I don't think the stylesheet does something weird and so this should be fixed in the spreadsheet wb.

luzpaz, 2024-04-15:
cc @obelisk79

obelisk79, 2024-04-16:
I concur with @MisterMakerNL, the best option would be to add a property either exposed in preferences or hidden for the property editor in order to define the default text color for spreadsheet outside of a stylesheet having to explicitly do so. 

The alternative would be to remove completely the ability to set color from the Spreadsheet toolbar.

luzpaz, 2024-07-27:
Is this not going to get attention before 1.0 ?

MisterMakerNL, 2024-07-27:
Text coloring doesn't work atm.
Stylesheets sets the color, and spreadsheet doesn't seem to overwrite this.
I don't see this as a blocker since coloring text in spreadsheets isn't something important for a CAD program

JDMils, 2024-07-29:
Just for more context:

With FreeCAD Classic theme selected and applied, in Spreadsheet, you can use the cell colour toolbar buttons to change the selected cell font and background colour- no problem here.

With FreeCAD Light theme selected and applied, in Spreadsheet, you can use the cell colour toolbar buttons to only change the selected cell background colour however the cell font colour toolbar button has no effect on the cell's font colour. Even trying to change the font colour using | Properties | Color, change White to any other colour, the cell's font colour does not change.

With FreeCAD Dark theme selected and applied, in Spreadsheet, you can use the cell colour toolbar buttons to only change the selected cell background colour however the cell font colour toolbar button has no effect on the cell's font colour. Even trying to change the font colour using | Properties | Color, change White to any other colour, the cell's font colour does not change.

While using the FreeCAD Dark theme, the current cell colours are:
Font: White
Background: Green

I then change the font colour to Yellow however nothing changes. I then change the theme to FreeCAD Light and the cell's font colour is now black. I then change the theme to FreeCAD Classic and the cell's font colour is now Yellow. I change the cell font to White and the cell shows a white font. I change the cell font to Magenta and change the theme to FreeCAD Dark however the cell font is now White.

My forum post:
https://forum.freecad.org/viewtopic.php?style=3&p=773564#p773564

`OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Installed mods:

Gridfinity 0.5.0
`

obelisk79, 2024-08-17:
@maxwxyz I have all of this working in OpenLight/OpenDark. I would consider this issue as written originally to be resolved and the issue should be closed. If there are residual issues with how FreeCAD Dark and Light have themed the spreadsheet, then new issues should be opened to have them resolved.

luzpaz, 2024-08-18:
Closing in that case. As obelisk79 writes above, if there are specific issues, please open a new ticket. (and reference this one)

closed 2024-08-18 by luzpaz

MisterMakerNL, 2024-08-18:
Ok this works in Opendark, now not yet in Freecad Dark.
Any tips Obelisk on how to fix this?

MisterMakerNL, 2024-08-18:
Ok this is awe-full how this works.
If I don't set the item color it will pick the color of the default text.
Then if you switch theme's it will not update the text color, but it will update the background color.
So any unmodified cell that gets used between dark and light theme will break.
But the text coloring option will work.
I kind a think it is more important to have interchangeability between theme's, so we have to find a better fix.
Probably something like we do with sketcher, where the color changes if it is not set by the user.
Or we make the table in dark and light, white.
Below screenshots of the same table in Freecad dark/light and Open dark/light. 
[image]
[image]
[image]
[image]

MisterMakerNL, 2024-09-19:
Could someone open this because it's not fixed.