FreeCAD/FreeCAD #8169: File, Edit & sometimes View in Menu Bar don't function with direct mouse click.

bstott, 2023-01-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2004
Word size of FreeCAD: 64-bit
Version: 0.21.0.31513 (Git)
Build type: Release
Branch: master
Hash: b2ab8edba4bfd71681e639f8c3f1105066bed4c7
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * A2plus 0.4.60g
  * Assembly4 0.12.5
  * boltsfc 2022.11.5
  * CfdOF 1.20.6
  * CurvedShapes 1.0.4
  * Curves 0.6.5
  * Defeaturing 1.2.0
  * Design456 0.0.1
  * DynamicData 2.46.0
  * EM
  * Estimate 0.1.2
  * ExplodedAssembly
  * fasteners 0.4.50
  * fcgear 1.0.0
  * FEMbyGEN 1.2.0
  * FEM_FrontISTR 0.1.0
  * freecad.gears 1.0.0
  * Help 1.0.3
  * InventorLoader 1.3.0
  * lattice2 1.0.0
  * Launcher
  * Manipulator 1.4.9
  * offline-documentation 1.0.0-alpha
  * OpticsWorkbench 1.0.9
  * parts_library
  * pcb
  * PieMenu 1.2.4
  * POV-Ray-Rendering
  * QuickMeasure 2022.10.28
  * Silk 1.0.0
  * slic3r-tools
  * SlopedPlanesMacro
  * symbols_library
  * ThreadProfile 1.84.0
  * toSketch 1.0.1
[/code]
```

### Subproject(s) affected?

None

### Problem description

Upon opening the application. The Menu Bar selections of 'File', 'Edit', and sometimes 'View' don't work. However, if you mouse over to 'Tools' and click for the drop down - you can move the mouse to the left and the drop downs for the 'View', 'Edit', and 'File' will then drop down and you can access the commands.  

Argh... Decade plus... And this application is sooo Buggy. Sigh. I've not ever been able to learn to use. Double, Sigh.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-12:
Is this a recent regression? Do you have any older window dev builds you can reproduce on?

bstott, 2023-01-12:
I upgraded from 0.20.2 to 0.21.0.
I experienced no issues with 0.19, 0.20, 0.21.
I just today updated to 0.21.0 and it doesn't work.
I live using the menu selections. Lol.

iplayfast, 2023-01-14:
Already reported. https://github.com/FreeCAD/FreeCAD/issues/7965

bstott, 2023-01-14:
Not the same. Similar. And different OS. This on Windows affects File, Edit, and 3/4 of the View. 
Problem appears to be a selection field for mouse over issue.  Possibly another program window feature is blocking those Menu screen locations.

bstott, 2023-01-14:
> Already reported. #7965

Not the same OS. Not completely same occurrence. And I don't use REM.

However, I did use the newer FreeCAD customization feature to add Sketcher WB Sketch button to at least one other WB toolbar (Curves, Part, ???). I don't remember which and will need to check desktop later, 1/14/23, to determine.

But, my problem is on Windows - Not UI on Linux.

jag45, 2023-01-25:
This problem is still appearing on Weekly Build 31641.
Both the File and the Edit menu cannot be accessed until you click on the Tools menu and move the mouse across to the File or Edit menus with the left button pressed.

I've found that the problem is caused by Assembly 3 being installed in the Add On Manager. If you uninstall Assembly 3, then restart Freecad the problem does not appear. Reinstalling Assembly 3, the problem comes back.

However this problem does not appear in Freecad 0.20.2 with or without Assembly 3 being installed.

Hope this helps you to find the issue.
Thanks
Jon

FREECAD 0.21.0.31641
CODE: [SELECT ALL](https://forum.freecadweb.org/viewtopic.php?f=10&t=74004&p=655556&hilit=weekly+build+menu#)

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31641 (Git)
Build type: Release
Branch: master
Hash: d28d63b87b60161419c6c0c532fbbfaed96926b8
Python 3.10.8, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * A2plus 0.4.60i
  * Assembly4 0.12.5
  * ExplodedAssembly
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * RemBench
  * sheetmetal 0.2.60
  * Woodworking 0.21.30486

FREECAD 0.20.2.29177 +426
CODE: [SELECT ALL](https://forum.freecadweb.org/viewtopic.php?f=10&t=74004&p=655556&hilit=weekly+build+menu#)

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * A2plus 0.4.60i
  * Assembly3 0.11.4
  * Assembly4 0.12.5
  * ExplodedAssembly
  * Help 1.0.3
  * QuickMeasure 2022.10.28
  * RemBench
  * sheetmetal 0.2.60
  * Woodworking 0.21.30486
[Top](https://forum.freecadweb.org/viewtopic.php?f=10&t=74004&p=655556&hilit=weekly+build+menu#top)

obelisk79, 2023-02-02:
My mod folder was causing this too, but a different mod
strange behavior

Syres916, 2023-02-03:
This should be closed as a duplicate, all updates are in #7965

closed 2023-02-05 by luzpaz