FreeCAD/FreeCAD #8176: [Issue] Broken geometry generated via tapered pocket

arcivanov, 2023-01-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Fedora Linux 37 (Workstation Edition) (KDE/default)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * ExtremeProDark 2.6.4
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

Given:

1. Complex shape
2. Tapered pocket of two rectangles `Mount.Pocket003.Sketch011` generating trapezoid groove
3. Several additional transformations.
4. Same tapered pocket of two rectangles `Mount.Pocket010.Sketch032` breaking geometry completely.

Green arrows point to correct behavior, red to broken geometry. 

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

arcivanov, 2023-01-12:
[vone_carrier_jensen_broken_geometry.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10405594/vone_carrier_jensen_broken_geometry.FCStd.zip)

arcivanov, 2023-01-12:
The parts are copyrighted Arcadiy Ivanov released under Apache License 2.0.

arcivanov, 2023-01-12:
Another paradoxical result, different plane, same approach:

[image]

[image]

[image]

[vone_carrier_jensen.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10406907/vone_carrier_jensen.FCStd.zip)

arcivanov, 2023-01-13:
Found a workaround: Sketch 21 plane was offset by `-1.5`. Removed offset to 0, reversed the associated pad (i.e. same operation in a different direction) and the pocket now worked.

arcivanov, 2023-01-13:
The original problem is still there

arcivanov, 2023-01-13:
Completely recreating the sketch does nothing, no workaround seems to be available.

[image]

arcivanov, 2023-01-13:
I'm completely stumped, nothing seems to help.

[image]

wsteffe, 2023-01-13:
You may use LinkDaily branch which is not affected by this bug:

 
[image]

arcivanov, 2023-01-13:
What you so much! Where do I find linkdaily? I can't find this branch. Do you mean for me to use a nightly?

wsteffe, 2023-01-14:
Hi arcivanov,
 LinkDaily Appimage can be downloaded at https://github.com/realthunder/FreeCAD/releases

But I would prefer the snap distribution which is updated more regularly.
Instruction for snap installation on your platform can be found at 
https://snapcraft.io/freecad-realthunder

arcivanov, 2023-01-14:
Same issue in 0.21.nightly

```
OS: Fedora Linux 37 (Workstation Edition) (KDE/default)
Word size of FreeCAD: 64-bit
Version: 0.21.0.pre_31513.fc37 (Git)
Build type: Unknown
Python 3.11.1, Qt 5.15.7, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.0
Locale: English/United States (en_US)
Installed mods: 
  * ExtremeProDark 2.6.4
  * ProDarkThemePreferencePack 1.0.0
  ```

wsteffe, 2023-01-14:
LinkDaily is not 0.21.nightly.  It is a fork developed by RealThunder (Zheng Lei) which has many improvements and bug fixes over FreeCAD master.

arcivanov, 2023-01-14:
> LinkDaily is not 0.21.nightly. It is a fork developed by RealThunder (Zheng Lei) which has many improvements and bug fixes over FreeCAD master.

I understood that, yes. I am however documenting the issue for FreeCAD proper to get this fixed.
@luzpaz

arcivanov, 2023-01-15:
@wsteffe 

```
18:39:18  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/snap/freecad-realthunder/102/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
<class 'ModuleNotFoundError'>: No module named 'Mesh'
```

The Realthunder is unable to export STL or 3MF.

arcivanov, 2023-01-15:
Installing "numpy-stl" was required.

wsteffe, 2023-01-16:
> Installing "numpy-stl" was required.

I think that this problem should be reported at https://github.com/realthunder/FreeCAD/issues
The missing python module has to be added to the snap build.

luzpaz, 2023-01-16:
How is this ticket realthunder's Link related ?

arcivanov, 2023-01-16:
> How is this ticket realthunder's Link related ?

This ticket is not related to realthunder, it's FreeCAD original. The realthunder seems like the only known workaround at present.

luzpaz, 2023-01-16:
Wait, is this [Toponaming](https://wiki.freecadweb.org/Topological_naming_problem) ?

arcivanov, 2023-01-16:
@luzpaz No. This ticket is due to broken geometry generated by FreeCAD 0.20.2 and 0.21-nightly with a pocket for no apparent reason.
The Realthunder LinkDaily generates correct geometry with the same exact project file and was proposed as a workaround by @wsteffe

luzpaz, 2023-01-16:
@realthunder if you have a moment, could you shed some light on how this is fixed in Link ?

arcivanov, 2023-01-16:
@luzpaz I'm not sure @realthunder is even aware it was a problem to be fixed. It might be incidental that his fork doesn't have this issue.

realthunder, 2023-01-16:
No, I didn't fix the problem. In fact, I have been chasing this same 'Ghostly' problem for quite some time. Now that I know it happens in upstream too. I am not sure about the root cause. What I know is that there may be some OCC operations that have some side effect on shapes of existing features, which somehow corrupt it. 

In you particular file, the corruption seems to start from `Sketch009`. After open your file, type the following command in Python console. It shows that the maximum tolerance of the sketch shape is 5.270833483333336, which is unacceptable,

```python
App.ActiveDocument.Sketch009.Shape.getTolerance(1)
```

Which results in the Pad generated being invalid. You can use `Check geometry` command in Part workbench to check `Pad002`, and it will report some error. Manually recompute `Pad002` won't fix the problem. Only after you manually recompute `Sketch009` (right click the tree item and select `Recompute object`), and then recompute `Pad002`, or just recompute the whole document can fix everything. You can verify that the sketch shape max tolerance is back to 1e-7.

So, the problem is to find out which operation that causes this side effect. I have a wild guess of a possible fix [here](https://github.com/realthunder/FreeCAD/commit/0e592432a33e62529ae0b5f6b78f325994227895). There maybe more than one place needs the fix though.

luzpaz, 2024-10-12:
Updated title to reflect @realthunder's feedback in https://github.com/FreeCAD/FreeCAD/issues/8176#issuecomment-1384681552

stefano2734, 2026-09-12:
There is an old state with occt 7.5.0.
I see here no state with actual occt  8.0.1 and 26.3 dev or occt 7.9.3 with 1.1.3.
Perhaps this is solved, changed in behavior or improved.

luzpaz, 2026-09-12:
>  I see here no state with actual occt 8.0.1 and 26.3 dev or occt 7.9.3 with 1.1.3.
> Perhaps this is solved, changed in behavior or improved.

@codeotaku any chance you can test and confirm this may be fixed in latest OCCT ?

@stefano2734 Thanks so much for testing with higher versions of OCCT. Please consider adding the full [About](https://wiki.freecad.org/About) info for builds you tested on. It saves confusion and time.