FreeCAD/FreeCAD #8177: Setas das cotas

ANV-2023, 2023-01-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Portuguese/Brazil (pt_BR)
Installed mods: 
  * A2plus 0.4.60f
  * ArchTextures
  * BIM 2021.12.0
  * ExplodedAssembly
  * ModernUI
[/code]
```

### Subproject(s) affected?

None

### Problem description

Estou usando a versão 0.20.2 e as setas das cotas não aparecem (nenhum formato). Já desinstalei e instalei a versão 0.19, mas o problema continua. Antes estava normal. Já pesquisei sobre isso e não encontrei nada.

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-13:
The bug tracker is english speaking. Do you mind uploading an example file (you can zip the .FCStd file)

WandererFan, 2023-01-14:
Arrow size in preferences?

[image]

Size = 0.0
[image]

Size = 5.0
[image]

ANV-2023, 2023-01-14:
> Tamanho da seta nas preferências?
> 
> [image]
> 
> Tamanho = 0,0 [image]
> 
> Tamanho = 5,0 [image]

> O rastreador de bugs fala inglês. Você se importa em enviar um arquivo de exemplo (você pode compactar o arquivo .FCStd)

Segue arquivo de exemplo:

[teste.zip](https://github.com/FreeCAD/FreeCAD/files/10417406/teste.zip)

ANV-2023, 2023-01-14:
It's not a problem of lack of value in preferences.

[image]

ANV-2023, 2023-01-14:
Follow exemplo file:

[teste.zip](https://github.com/FreeCAD/FreeCAD/files/10417434/teste.zip)

WandererFan, 2023-01-14:
5 μm?  That smaller than a human hair!

luzpaz, 2023-01-14:
Looks like a PEBKAC error. Closing ticket

closed 2023-01-14 by luzpaz

ANV-2023, 2023-01-14:
Obrigado e desculpem. Não percebi.