FreeCAD/FreeCAD #8188: [Problem] Preferences dialog destroys sketch colors

PLChris, 2023-01-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.0.31513 (Git)
Build type: Release
Branch: master
Hash: b2ab8edba4bfd71681e639f8c3f1105066bed4c7
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * DynamicData 2.46.0
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.4.0
  * fcgear 1.0.0
  * fasteners 0.4.16
  * sheetmetal 0.2.60
  * ExplodedAssembly
  * Curves 0.6.5
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

When opening and confirming the preferences dialog, Sketch colors are set to white.

Steps to reproduce:
- Rename attached file to *.FCStd
- Open the file, it shows a black rectangle and a red line
- Open Preferences dialog, no change is needed to show the issue
- Confirm with Ok

-> Result: All lines are white
[bendTest.zip](https://github.com/FreeCAD/FreeCAD/files/10417116/bendTest.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-14:
@wwmayer can you take a look at this ?

wwmayer, 2023-02-15:
It's because when saving the parameters (it's irrelevant if changed or not) will affect all existing sketches. So, the question is: should these parameters only affect newly created sketches or also existing sketches?

luzpaz, 2023-03-03:
@wwmayer what about presenting the user with both options?

wwmayer, 2023-03-05:
In one of the recently merged PR it has been decided that changing the preferences shouldn't affect existing sketches and IMO that's fine.

PR #9151 (2023-04-01): Sketcher: do not apply line/edge color preference to existing sketches
fixes #8188
commit 58a9af21 (2023-04-01): Sketcher: do not apply line/edge color preference to existing sketches, fixes #8188
closed 2023-04-01 by wwmayer