FreeCAD/FreeCAD #8197: ProDark theme: leaves artifacts on screen

HuibSmeets, 2023-01-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Issue description

Hi,

When switching to the ProDark theme (and the same happens with the ExtrememProDark preference pack) it leaves blacklines in the window after resizing panels (Combo view, Report view etc). Switching back to one of the other themes fixes it. The behavior 'survives' stopping and restarting FreeCAD. As long as this theme is used, the lines showup when resizing panels. Not a big deal but maybe causing memory-leakage. 

### Anything else?

FreeCAD installed on an iMac with MacOS 12.6 (Monterey)

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-15:
@PLChris do you mind testing on macOS latest dev to see if you can reproduce ?

HuibSmeets, 2023-01-15:
I'm not PLChris but downloaded the latest dev build and it exhibits the same issue. Too add, this on an Intel Mac (Late 2015 model)

[image]

HuibSmeets, 2023-01-16:
To add, same artifacts happen with the latest weekly development build for Apple/ARM silicon on MacOS 13.1 (Ventura).

luzpaz, 2023-01-16:
Why was this ticket closed ?

HuibSmeets, 2023-01-16:
mhmm, strange, I edited my last comment, most likely I did

luzpaz, 2023-01-16:
@ipatch can you recreate on macOS ?

ipatch, 2023-01-17:
i don't a ventura VM setup at the moment. kindof trying to juggle AFK things at the moment. 

 - i'm curious if this could be ventura related?
 - ventura only on intel?
 - or ventura on intel and arm?
- strictly related to conda?

if one those guys who is using ventura wants to download one of the builds i uploaded a while back that'd be the first place i check and if same results, i can try and check on a big sur vm at some point.

https://github.com/FreeCAD/homebrew-freecad/releases/tag/0.20.0.release

ipatch, 2023-01-17:
@chennes 

you getting these artifacts on that m1 box?

chennes, 2023-01-18:
@ipatch Yes, I just checked it on the latest weekly on my M1 and it is showing this behavior.

luzpaz, 2023-01-19:
So what's the issue here?  
Qt?  
Coin3d?

chennes, 2023-01-20:
Looks Qt to me, but that's just a guess, not solid information.

JohnOCFII, 2023-08-13:
Issue still occurs with release version of 0.21.  Issue happens with `ProDark`, as reported, and as a different test case, does not happen with the `Behave-Dark` style sheet.

```
OS: macOS 13.5
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```

[image]

luzpaz, 2023-08-13:
Can someone test if this is reproducible on a macOS build using Qt6 ?

berniev, 2023-08-13:
Is there a macOS build using Qt6 ?

luzpaz, 2023-08-13:
@ipatch do you happen to know if there is one ?

ipatch, 2023-08-13:
i have not built freecad with qt6.

luzpaz, 2023-08-16:
Anyone willing to start working on a Qt6 build for macos ?

berniev, 2023-08-16:
Some work was done [https://github.com/FreeCAD/FreeCAD/pull/7682](url) but was inconclusive

luzpaz, 2023-10-13:
@HuibSmeets, i'm going to close this as a duplicate of #9914

closed 2023-10-13 by luzpaz