FreeCAD/FreeCAD #8202: [Issue] Pad with reference direction does not allow to select edges from a feature different than the previous

PLChris, 2023-01-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.0.31598 (Git)
Build type: Release
Branch: master
Hash: f7db685a2821adaf23c85a25855401ecc1eeff2c
Python 3.10.8, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * DynamicData 2.46.0
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.4.0
  * dxf-library
  * fcgear 1.0.0
  * fasteners 0.4.16
  * sheetmetal 0.2.60
  * ExplodedAssembly
  * Curves 0.6.5
[/code]
```

### Subproject(s) affected?

None

### Issue description

To minimize topological issues it is sensible to reference the first occurrence of an edge. For Pads with a reference direction - and probably for pockets too - the selection is restricted to the direct predecessor. Other tools allow to change the visibility while the tool is active and selecting the reference direction should behave the same. An example is Pad's "Up to face" type

The image 
[image]
shows the final result, the selected edge defines the direction of the pad.

If selected from the pad dialog Edge9 defines the direction. If the previous sketch is changed, then Edge9 is in a different place and the last pad fails. The image shows where it is now.
[padReferenceDirection.zip](https://github.com/FreeCAD/FreeCAD/files/10430283/padReferenceDirection.zip)
In the attached FreeCAD file this is fixed by changing the reference to Pad[Edge6] instead of Pocket[Edge9].
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PLChris, 2023-01-17:
Here is the file, it didn't make it in the first post.
[padReferenceDirection.zip](https://github.com/FreeCAD/FreeCAD/files/10430291/padReferenceDirection.zip)

luzpaz, 2023-02-13:
@AjinkyaDahale do you have a moment to look in to this ?

AjinkyaDahale, 2023-02-19:
@luzpaz this is just another toponaming issue, right?

AjinkyaDahale, 2023-02-19:
I feel a bit confused what the point here is: is it to implement a simpler form of toponaming fix?

luzpaz, 2023-02-19:
@AjinkyaDahale yes, indeed that is what it looks like. 

@PLChris is there a forum thread where this is discussed or did you intend for this to be the discussion place?

luzpaz, 2023-11-12:
ping @PLChris

maxwxyz, 2024-06-23:
is this still relevant?
@FlachyJoe @bgbsww FYI

FlachyJoe, 2024-06-23:
Fixed only by showing the Eye icons, else we can't switch features visibility when edge selection mode is activated.

maxwxyz, 2024-06-23:
@NomAnor how the eye for default going?

NomAnor, 2024-06-24:
#14393 is merged

maxwxyz, 2024-06-24:
So this issue could be closed then?
@chrisb-github

maxwxyz, 2024-07-14:
I'm closing this then.

closed 2024-07-14 by maxwxyz