FreeCAD/FreeCAD #8203: [Problem] Tech Draw: Measure Type = True always returns 0

reald, 2023-01-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.1 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31622 (Git)
Build type: Unknown
Branch: master
Hash: 199ef42f625aafdda03be4d5cd8bb311fa98abdc
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
[/code]
```

### Subproject(s) affected?

Techdraw

### Problem description

Steps to reproduce:
1. Add some distance measurements to a tech draw. 

[image]

2. Change "measure type" from "Projected" to "True"

[image]

Result: All changed measurements will have 0 length instead of the true length.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-01-17:
Does the dimension have 3d references?  If not, it can not calculate the True value

reald, 2023-01-17:
Thanks for the tip! 3D reference has to be added as described here:

https://wiki.freecadweb.org/TechDraw_LinkDimension

closed 2023-01-17 by reald

caco3, 2023-04-09:
I have the exactly same issue and can't figure out how to fix this..

[image]

I am using the Linux version (snap, 0.20.2, 64bit)

Example file:
[techdraw.zip](https://github.com/FreeCAD/FreeCAD/files/11186473/techdraw.zip)

WandererFan, 2023-04-10:
How did you get the same dimension object in 2 different views?  That isn't supposed to happen.

True dimensions are only available for dimensions with a reference to the 3d model.  Otherwise 
only the projected dimension is available.

There should have been a message a some point: 
11:31:34  Dimension001 - True dimension has no 3D References

In v0.20 you'll have to use LInkDimension command as mentioned above.  In v0.21 you can create 3d references by point and click.

caco3, 2023-04-10:
Thanks for asking back.

I tried to follow the instructions at https://wiki.freecad.org/TechDraw_LinkDimension but obviously I understrood it wrong.
I now finally found a video which shows how to do it correctly: https://www.youtube.com/watch?v=vOUZB0mMKKA

I think my mistake was on in step 2: I selected the edge in the **TechDraw** tab instead of the **Parts** tab.