FreeCAD/FreeCAD #8206: [Issue] segfault on 0.20.2 + python3.11

andy128k, 2023-01-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Debian GNU/Linux bookworm/sid (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.2.
Build type: Release
Python 3.11.1, Qt 5.15.7, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
[/code]
```

### Subproject(s) affected?

None

### Issue description

FreeCAD segfaults being run with parameters (e.g .desktop file has `/usr/bin/freecad --single-instance`).

```
$ gdb freecad
GNU gdb (Debian 12.1-4+b1) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from freecad...
Reading symbols from /usr/lib/debug/.build-id/97/3c2c777d1a78a5b6602ce47bea03998897d482.debug...
(gdb) r --single-instance
Starting program: /usr/bin/freecad --single-instance
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
unknown option --single-instance
usage: /usr/bin/freecad [option] ... [-c cmd | -m mod | file | -] [arg] ...
Try `python -h' for more information.
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
[New Thread 0x7fffee8b86c0 (LWP 1206751)]
[New Thread 0x7fffee0b76c0 (LWP 1206752)]

Thread 1 "freecad" received signal SIGSEGV, Segmentation fault.
__wcslen_sse4_1 () at ../sysdeps/x86_64/multiarch/strlen-sse2.S:142
142	../sysdeps/x86_64/multiarch/strlen-sse2.S: Нет такого файла или каталога.
(gdb) bt
#0  __wcslen_sse4_1 () at ../sysdeps/x86_64/multiarch/strlen-sse2.S:142
#1  0x00007ffff62d0223 in encode_current_locale (text=text@entry=0x0, str=str@entry=0x7fffffffd9f8, error_pos=error_pos@entry=0x0, reason=reason@entry=0x0, 
    raw_malloc=raw_malloc@entry=0, errors=errors@entry=_Py_ERROR_SURROGATEESCAPE) at ../Python/fileutils.c:673
#2  0x00007ffff62d090e in encode_locale_ex (errors=_Py_ERROR_SURROGATEESCAPE, current_locale=0, raw_malloc=0, reason=0x0, error_pos=0x0, str=0x7fffffffd9f8, 
    text=0x0) at ../Python/fileutils.c:818
#3  encode_locale (current_locale=0, raw_malloc=0, error_pos=0x0, text=0x0) at ../Python/fileutils.c:828
#4  Py_EncodeLocale (text=0x0, error_pos=0x0) at ../Python/fileutils.c:855
#5  0x0000555555559034 in main (argc=<optimized out>, argv=0x7fffffffde08) at ./src/Main/MainGui.cpp:252
(gdb) q
A debugging session is active.

	Inferior 1 [process 1206748] will be killed.

Quit anyway? (y or n) y
```

### Anything else?

I **guess** important part is

```
unknown option --single-instance
usage: /usr/bin/freecad [option] ... [-c cmd | -m mod | file | -] [arg] ...
Try `python -h' for more information.
```

`/usr/bin/freecad` is misleading and this is an output from Python invoked [here](https://github.com/FreeCAD/FreeCAD/blob/master/src/Base/Interpreter.cpp#L568-L571). I think, it throws an exception and [`Py_Initialize`](https://github.com/FreeCAD/FreeCAD/blob/1972dfe5a3a2f307804c166c1619e71fefc27283/src/Base/Interpreter.cpp#L580) is not invoked. Then, an exception is caught [here](https://github.com/FreeCAD/FreeCAD/blob/079bdc64b1e4c69a518f4aa80f8b129163c455bc/src/Main/MainGui.cpp#L232), then `Py_GetPath` returns `null` and then `Py_EncodeLocale` segfaults.

-----

Probably this issue is specific to Python < 3.11. A code for older versions [ignores errors and does not throw an exception](https://github.com/FreeCAD/FreeCAD/blob/1972dfe5a3a2f307804c166c1619e71fefc27283/src/Base/Interpreter.cpp#L552).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-18:
Thanks for the thorough bug report. Can you reproduce this on the development (HEAD) version as well ?

andy128k, 2023-01-18:
It reproduces on current HEAD `982152c332480663c0b1e0ecdf56fd9dc5d69191`

```
[code]
OS: Debian GNU/Linux bookworm/sid (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31623 (Git)
Build type: Debug
Branch: master
Hash: 982152c332480663c0b1e0ecdf56fd9dc5d69191
Python 3.11.1, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
[/code]
```

```
$ gdb ./bin/FreeCAD
GNU gdb (Debian 12.1-4+b1) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./bin/FreeCAD...
(gdb) r --single-instance
Starting program: /home/andy/tmp/FreeCAD/freecad-build/bin/FreeCAD --single-instance
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
unknown option --single-instance
usage: /home/andy/tmp/FreeCAD/freecad-build/bin/FreeCAD [option] ... [-c cmd | -m mod | file | -] [arg] ...
Try `python -h' for more information.
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
[New Thread 0x7fffec4b86c0 (LWP 1264361)]
[New Thread 0x7fffebcb76c0 (LWP 1264362)]

Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
__wcslen_sse4_1 () at ../sysdeps/x86_64/multiarch/strlen-sse2.S:142
142	../sysdeps/x86_64/multiarch/strlen-sse2.S: Нет такого файла или каталога.
(gdb) bt
#0  __wcslen_sse4_1 () at ../sysdeps/x86_64/multiarch/strlen-sse2.S:142
#1  0x00007ffff3ad0223 in encode_current_locale (text=text@entry=0x0, str=str@entry=0x7fffffffd388, error_pos=error_pos@entry=0x0, reason=reason@entry=0x0, 
    raw_malloc=raw_malloc@entry=0, errors=errors@entry=_Py_ERROR_SURROGATEESCAPE) at ../Python/fileutils.c:673
#2  0x00007ffff3ad090e in encode_locale_ex (errors=_Py_ERROR_SURROGATEESCAPE, current_locale=0, raw_malloc=0, reason=0x0, error_pos=0x0, str=0x7fffffffd388, 
    text=0x0) at ../Python/fileutils.c:818
#3  encode_locale (current_locale=0, raw_malloc=0, error_pos=0x0, text=0x0) at ../Python/fileutils.c:828
#4  Py_EncodeLocale (text=0x0, error_pos=0x0) at ../Python/fileutils.c:855
#5  0x0000555555567f46 in main (argc=2, argv=0x7fffffffdda8) at /home/andy/tmp/FreeCAD/src/Main/MainGui.cpp:232
(gdb) q
A debugging session is active.

	Inferior 1 [process 1264348] will be killed.

Quit anyway? (y or n) y
```

luzpaz, 2023-01-18:
@wwmayer If you have a moment, can you take a look at this ?

ohazi, 2023-02-03:
`freecad --dump-config` fails similarly:
```
ohazi@woodstock:~$ freecad --dump-config
unknown option --dump-config
usage: freecad [option] ... [-c cmd | -m mod | file | -] [arg] ...
Try `python -h' for more information.
Program received signal SIGSEGV, Segmentation fault.
```

Seems like maybe all of the arguments are getting routed to python rather than to freecad's argument parser? The error string looks suspiciously identical. 

```
ohazi@woodstock:~$ python3 --foo
unknown option --foo
usage: python3 [option] ... [-c cmd | -m mod | file | -] [arg] ...
Try `python -h' for more information.
```

fwiw, both seem to work when an extra `-` is used, i.e. `freecad - --dump-config` and `freecad - --single-instance`

wwmayer, 2023-02-03:
That's weird. Can you check the file /usr/bin/freecad? Is it still a symlink to the actual FreeCAD executable or is it a Python script?

ohazi, 2023-02-03:
```
ohazi@woodstock:~$ file `which freecad`
/usr/bin/freecad: symbolic link to /etc/alternatives/freecad
ohazi@woodstock:~$ file /etc/alternatives/freecad
/etc/alternatives/freecad: symbolic link to /usr/lib/freecad/bin/freecad-python3
ohazi@woodstock:~$ file /usr/lib/freecad/bin/freecad-python3
/usr/lib/freecad/bin/freecad-python3: ELF 64-bit LSB pie executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=973c2c777d1a78a5b6602ce47bea03998897d482, for GNU/Linux 3.2.0, stripped
```

This is the package in Debian testing, btw

wwmayer, 2023-02-10:
https://alioth-lists.debian.net/pipermail/debian-science-maintainers/2023-February/108142.html

DEL-TRONIK, 2023-02-11:
I had exactly the same problem under Debian Testing (Bookworm), But the problem seems solved with the last update (0.20.2+dfsg1-4).

The launcher (.desktop file in /usr/share/applications) has been changed as this with the update:

> "/usr/bin/freecad - --single-instance %F"

FreeCad is normally execute with the launcher and the single instance works perfectly.

But when I make a freecad --version in terminal, I get the following message:

> Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3bf90) [0x7fc61ac5af90]
#1  /lib/x86_64-linux-gnu/libc.so.6(+0x161cf9) [0x7fc61ad80cf9]
#2  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(+0x2d0223) [0x7fc61c6d0223]
#3  /lib/x86_64-linux-gnu/libpython3.11.so.1.0(Py_EncodeLocale+0x4e) [0x7fc61c6d090e]
#4  freecad(+0x5034) [0x559f3680f034]
#5  /lib/x86_64-linux-gnu/libc.so.6(+0x2718a) [0x7fc61ac4618a]
#6  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x85) [0x7fc61ac46245]
#7  freecad(+0x5ad1) [0x559f3680fad1]

Apart from that, it works perfectly.

andy128k, 2023-02-11:
0.20.2 works in Debian because they applied a workaround patch https://salsa.debian.org/science-team/freecad/-/commit/af3683ae457b6a69b8f3fa08f8272e32f83c3b79

The root cause is not addressed.

wwmayer, 2023-02-11:
Today I downloaded the files
* http://deb.debian.org/debian/pool/main/f/freecad/freecad_0.20.2+dfsg1-4.dsc
* http://deb.debian.org/debian/pool/main/f/freecad/freecad_0.20.2+dfsg1.orig.tar.xz
* http://deb.debian.org/debian/pool/main/f/freecad/freecad_0.20.2+dfsg1-4.debian.tar.xz
executed `dpkg-source -x freecad_0.20.2+dfsg1-4.dsc` and `debuild`

The compilation went fine but when it came to build the .deb files it stopped with this error:
> Can't exec "/home/user/Downloads/freecad/sources/freecad-0.20.2+dfsg1/debian/freecad-python3.install": No such file or directory at /usr/share/perl5/Debian/Debhelper/Dh_Lib.pm line 1561.
dh_install: error: cannot run debian/freecad-python3.install: No such file or directory
make: *** [debian/rules:3: binary] Error 255

Anyway, when I cd to debian/tmp/usr/share/freecad/bin/ then running `./FreeCAD-Python3 --single-instance` doesn't raise the reported error. So, to me it looks like a packaging problem, not an upstream bug.

FlachyJoe, 2023-02-11:
Hi,
I have the same issue with a self compiled from git (debian Bookworm too).

It seems cli options are passed to python which failed to start.
Try -v : it gives FreeCAD version information + verbose python output

@wwmayer can you take a look? It could be related to f4b4aab

wwmayer, 2023-02-11:
Wait! I found it!

The Debian package uses Python3.11 while the default on my system is Python3.10. However, when looking at the code https://github.com/FreeCAD/FreeCAD/blob/master/src/Base/Interpreter.cpp#L517 then the init() function for Py3.10 is different to Py3.11. To this function all arguments are passed that are already checked by Python.

wwmayer, 2023-02-11:
https://github.com/FreeCAD/FreeCAD/pull/8435

closed 2023-02-11 by wwmayer

luzpaz, 2023-02-11:
@DEL-TRONIK just a heads-up please put warnings/errors in markdown code tags or else they may possibly spam other unrelated issues. I've updated [your ](https://github.com/FreeCAD/FreeCAD/issues/8206#issuecomment-1426655273) to unlink non-relevant issues.

commit c7a21ecb (2023-02-11): Fixes #8206: FreeCAD segfaults being run with parameters (e.g .desktop file has /usr/bin/freecad --single-instance)
commit b28821d2 (2023-02-11): Fixes #8206: FreeCAD segfaults being run with parameters (e.g .desktop file has /usr/bin/freecad --single-instance)