FreeCAD/FreeCAD #8218: [Problem] [Scripting] Exporting compressed step stpZ fails

Jaaaky, 2023-01-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: openSUSE Tumbleweed (KDE//usr/share/xsessions/plasma5)
Word size of FreeCAD: 64-bit
Version: 0.20.2.Unknown
Build type: Release
Python 3.10.9, Qt 5.15.7, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.0
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Using Part.export to export step format in python code works; for example
`Part.export(toExport,ExportFile+u".step")`

But if I changed this to stpz or stpZ it fails with `<Exception> Unknown extension`
`Part.export(toExport,ExportFile+u".stpZ")`

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-21:
@Jaaaky does part export support stpz format in the first place ?

luzpaz, 2023-01-21:
Found https://github.com/FreeCAD/FreeCAD/search?q=stpZ&type=commits  
@easyw can you weigh-in on how to pythonically export `stpZ` format ?

superhappychris, 2023-01-31:
I'm able to export to stepz using the following code:
```python
import stepZ
stepZ.export(toExport,ExportFile+u".stpZ")
```

luzpaz, 2023-01-31:
@superhappychris  Is there documentation for this?

superhappychris, 2023-01-31:
@luzpaz [the wiki mentions](https://wiki.freecadweb.org/Import_Export) that you can export stpz files using std export, so I tried that and looked at the output in the Python console window

luzpaz, 2023-01-31:
@superhappychris 
The docs don't mention the need to `import stepZ`  
Can you offer an idea how to inform devs to do this? Should we add something to the API to throw a hint in the error?

superhappychris, 2023-01-31:
@luzpaz I'm completely new to FreeCAD (just downloaded it yesterday lol) so I don't think I know the app or code-base well enough to make any suggestions...

Maybe @Jaaaky has some thoughts?

luzpaz, 2023-03-03:
Ping @easyw

luzpaz, 2023-03-15:
@easyw do you mind checking this out ?

easyw, 2023-04-05:
@luzpaz 
exporting in stepZ format need an additional library/wb which is included in FreeCAD/Mod/Import module.
Simply it can be triggered by the export Gui command or through the code that @superhappychris mentioned...
The python Panel would trigger the used code...
What @Jaaaky mentioned in the wiki is the Standard Import/Export way through the Gui, not the python code to do it by script.
So I would close this because is not an issue nor a wiki error IMO.

easyw, 2023-04-05:
the same panorama with exporting in DXF format:
```
import importDXF
importDXF.export(__objs__, u"/home/mau/Downloads/Cube.dxf")
```

luzpaz, 2023-04-06:
Closing per easyw

closed 2023-04-06 by luzpaz