FreeCAD/FreeCAD #8220: [Problem] Constraints of angles

Almisuifre, 2023-01-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

Hi,
I came to see you because I loaded the version 0.20.2.
I notice a nice bug positioning on the constraints of angles. See for yourself!
They seem unarmed attached to the central axis instead of the usual position. This is rather embarrassing because we no longer know what the constraint applies to.
[image]
[image]
[image]

_The steps that lead to the bug:_
Start FreeCAD 0.20.2
New tab
New body
Select a starting face (construction line)
Switch to sketcher mode
Draw a picture
Place the angle rib. It appears in the wrong place.

This produces a bad experience because we no longer know what the constraint is attached to. It seems to be frozen between the central axes and the angle we want to set. It seems to stand out from the selection.

### Anything else?

https://discord.com/channels/870877411049357352/873944627806339082/1065907714242387978

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-21:
Can you please share an example file?

Almisuifre, 2023-01-21:
Okay, so here’s a video that shows the problem, as well as a file generated during the video.
[Youtube Bug contraint](https://youtu.be/S9CqsQkTgdU)
[Drive>FreeCAD0.20.2](https://drive.google.com/file/d/1qX56aV30jiUkzXNJDTkcUXhGqxfEjbi9/view?usp=share_link)

adrianinsaval, 2023-01-22:
I reattach here the file so we don't depend on external hosting (zip extension is a dummy, just rename)
[TestContraintAngle-001.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10475706/TestContraintAngle-001.FCStd.zip)

adrianinsaval, 2023-01-22:
I can reproduce this on 0.20.2 but not on 0.21
```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31641 (Git)
Build type: Release
Branch: master
Hash: d28d63b87b60161419c6c0c532fbbfaed96926b8
Python 3.10.8, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Spanish/Paraguay (es_PY)
Installed mods:
```

Almisuifre, 2023-01-23:
> I reattach here the file so we don't depend on external hosting (zip extension is a dummy, just rename) [TestContraintAngle-001.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10475706/TestContraintAngle-001.FCStd.zip)
Okay, is that better?
[TestContraintAngle-001.zip](https://github.com/FreeCAD/FreeCAD/files/10476082/TestContraintAngle-001.zip)

> I can reproduce this on 0.20.2 but not on 0.21
>
> ```
> OS: Windows 10 Version 2009
> Word size of FreeCAD: 64-bit
> Version: 0.21.0.31641 (Git)
> Build type: Release
> Branch: master
> Hash: [...]
I haven’t tried with the last build. I seem to have heard that it didn’t have an impact on Linux. But now I’m on Windows!
I will try anyway-we never know. I come back to see to tell you what the situation is.

Almisuifre, 2023-01-23:
Ok, I tested on the 0.21 successfully. So this is a problem with the 0.20.2 version!
Here’s the video evidence.
[Youtube not bug contraint in 0.21](https://youtu.be/YDvC-c9ywHY)
Tested with the same workstation. Namely:
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31641 (Git)
Build type: Release
Branch: master
Hash: d28d63b87b60161419c6c0c532fbbfaed96926b8
Python 3.10.8, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
[/code]
Here is the file generated in a zip.
[TestContraintAngle-002-FC021.zip](https://github.com/FreeCAD/FreeCAD/files/10476137/TestContraintAngle-002-FC021.zip)
What more can I bring?

Almisuifre, 2023-01-27:
Hi there,
I don’t see a comeback. Are the investigations still ongoing? I haven’t seen a new stable version either. What about?

adrianinsaval, 2023-01-27:
Did this work correctly on 0.20.1? Someone would have to do a bisect to find out where the problem originated

Almisuifre, 2023-01-27:
Yes, it worked on version 0.20.1

Almisuifre, 2023-01-29:
Good morning,
For your information, yesterday I helped a beginner in FreeCAD 0.20.2. The behavior is strictly the same as at home. The angle constraint works but its position is not correct. It is also under Windows 10. We can therefore exclude the case of a single installation on a single station.
Thank you so much for your help.

Syres916, 2023-02-06:
@adrianinsaval I really don't think that a Git Bisect is required in this instance, I've proved that this is a Windows LibPack specific issue unless you believe there's some code that's changed and only affected the LibPack build. The following is what I wrote in another report on the forum and I've advised Uwe to please check it on the next LibPack build which version that may be:

> this regression is not a general 0.20.2 regression as possibly first thought, it's specific to the Windows Libpack build. I'd thought to myself "Why have I not seen this on my Conda Linux build?" (I use 0.20.2 on a daily basis) because it doesn't occur. It also doesn't occur on the same system with the same config files using the Conda Windows build, it only occurs in the Libpack build

Syres916, 2023-02-06:
Could the title if this issue please be changed to something like `[Issue] Sketcher angle constraint is not displayed where expected in 0.20.2 Windows LibPack build`

Almisuifre, 2023-02-06:
Hello @Syres916, thank you for returning.

I changed the title, as you suggested.

I found the bug in other windows. Probably we have the same LibPack. In this case, would it not be possible to test on our machine, then switch to the correct adhequat version to regain optimal operation.
I’m talking about a minor change in the code (if LbPack W10, then old version/ Otherwise, take the new version). That would avoid the problem.

Another solution,  providing an equivalent to LibPack can be by switching to Conda including for windows.

Finally, well, first and foremost is to have seen what happens ;)

PeterVKelly, 2023-02-17:
I'm using Windows 10 and have had the same experience. I uninstalled, went to version 20.1, NO problems there. Re-downloaded 20.2, unlucky, problem is permanent. Verified on 2 different machines running different versions of windows 10. One is a laptop (Asus), the other is a generic tower with a Nvidia RTX20.. video board.
Cheers, p.

Almisuifre, 2023-02-17:
Thank you very much for your repport. It is a contribution. We must now count on the developers to find a way out.

luzpaz, 2023-02-17:
@donovaly tagging you temporarily on this if that's OK, since it's libpack related

PeterVKelly, 2023-02-17:
Hi, please pardon my ignorance; I'm a first time commenter. What is a tag? and what does 'libpack related' mean?
Sorry for imposing on your time to answer,
Cheers, p. 

⁣Get BlueMail for Android ​

PeterVKelly, 2023-02-17:
Thanks Alexandre,
Cheers, p. 

⁣Get BlueMail for Android ​

donovaly, 2023-02-18:
> @donovaly tagging you temporarily on this if that's OK, since it's libpack related

it seems it is not Libpack-related: https://forum.freecad.org/viewtopic.php?p=658814#p658814

And if it would be, what should I do? What library is causing this?

donovaly, 2023-02-18:
@abdullahtahiriyo , do you have an idea what could cause this regression bug.
As it seems not to be a problem with the libraries in the Libpack, it might be something else that was changed between FC 0.20.1 and 0.20.2.

PeterVKelly, 2023-02-18:
Hi Abdullah (please forgive my name spelling - I am a pretty dumb newbie).
I have no idea what is causing this bug - it seems 'affected' - flips as the dimensioning angle is dragged across the 'x' axis; also, when the newly established angle 'text' is selected by clicking on it, it 'jumps' a significant distance 'down' - ie vertically down the 'y' axis (when drawing on the 'xy' plane and viewing from the top).
This DEFINITELY an artifact of the 20.2 windows 10 release, it is not in and previously viewed version - from 17 up.
I have uninstalled 20.2 from both my laptop (Asus) and tower (generic 64 bit system with Nvidia RTX20 and 2 large Asus monitors, and reinstalled 20.1 on both, about 3 times now, and the problem stays with 20.2.
At first I thought it was me as I am a very experienced programmer (mainly c++), but I haven't even looked at development code for about 15 years (I am 75 years old) so naturally, I am the first suspect when a bug appears.
I can definitely say that it ain't me or other versions, it was introduced (or reintroduced) in the 20.2 windows 10 release. More I cannot say!
Cheers, p. 
PS Abdullah, can you explain briefly to me what 'tagging' is please - don't go to too much trouble, or be diplomatic - I am  beyond being offended!
Cheers Abdullah, p. 
​

abdullahtahiriyo, 2023-02-18:
Hi there!

I think tagging here means "notifying you". Not to be mixed with git tag, which represents a name for a given commit. Here 0.20.2 is a tag for the commit on which this release build.

There are quite some commits between 0.20.1 and 0.20.2:
https://github.com/FreeCAD/FreeCAD/compare/0.20.1...0.20.2

I have checked those directed to the files responsible for representation of constraints. I see nothing evident.

I have built tag 0.20.2 to see if cherry-picking certain commits would fix the issue. However, I cannot reproduce the issue. I am using Ubuntu though (and I have no windows).

donovaly, 2023-02-18:
It would be strange if it is the LibPack because the only change in it between FC 0.20.1 ans 0.20.2 is the OpenSSL library and Gmsh:
https://github.com/FreeCAD/FreeCAD-LibPack/releases

Gmsh is only used for meshing thus cannot be the reason.

I will now recompile FC 0.20.2 using older LibPacks to find out more.

PeterVKelly, 2023-02-18:
Hey Abdullah, nice try, it's a shame you don't have windows; but one thing seems clear(ish) to me:- you have shown that it is a windows only artifact, so you have progressed the search for the answer; well done sir!
Cheers and thanks, p. 

⁣Get BlueMail for Android ​

PeterVKelly, 2023-02-18:
Good luck. I really admire members of your tenacious nature and research driven character. Thanks for letting me know. 
Cheers, p. 

⁣Get BlueMail for Android ​

Almisuifre, 2023-02-20:
Hi,

If only I could code effectively, I could have checked further. What I noticed is that I do not hear about this artifact only in sketcher and only in windows. It happened on several of my machines, on that of friends too.

I’m admiring the compilation possibilities. Indeed, see between the two versions what may have changed, possibly in a compiled test version that we could also try. What do you think? Would that help?

Health to you and thank you very much for taking the time to help resolve this artifact.

PeterVKelly, 2023-02-20:
Hi Alexandre, if only I could still code (I'm an ancient C++ programmer, too old), yes I found it interesting in that the proximity to the x-axis seems to change the behavior, AND there is. Negative jump in the y-axis when subsequently clicked upon. A weird one! Thanks for getting in touch.
Cheers, p. 

⁣Get BlueMail for Android ​

abdullahtahiriyo, 2023-02-20:
There are two effective ways of fixing this: (a) by providing a way to reproduce that a developer can reproduce (there is the tricky part), or (b) if the issue is not present in current master, by making a new release, which should happen in a couple of months.

Actually there is a third one, but it requires somebody being able to reproduce to download the code, compile the code and debug the code. That third one is the most helpful of all, but I do understand not everybody is in a position to do it.

PeterVKelly, 2023-02-21:
> Indeed, I don’t have the skills to recompile FreeCAD on my machine and
> find the bug. Maybe we should find someone among programmers who can
> test the behaviour. I also showed it in the videos.
>
> Another solution, cross-over, is to provide us with a compiled version
> that we could test and video back what works or not. If we have a lead.
> Unless it wastes more time.
>
> Yours truly.

Dearest Alexandre, please don't think for a moment that I am patronizing you in a negative way; you are so keen and enthusiastic that you remind me of myself many years ago; perhaps you are brighter, certainly energetic and keen. I'm sure that the programming heavyweights into freeCAD will fix this anomaly - their comparative egos will drive them. Watch and wait Alexandre.
Thanks for your comments, 
Cheers, p. 

⁣Get BlueMail for Android ​

PeterVKelly, 2023-02-21:
I agree Abdullah, and I can assure you that the issue exists in the master as I have downloaded and tested it several times. As I said to Alexandre, the big guns will fix this; by the way, I also admire and commend your enthusiasm and intentions.
Thanks and cheers, p. 

⁣Get BlueMail for Android ​

abdullahtahiriyo, 2023-02-21:
By master you do mean downloading the code from GitHub and compiling the source to generate the binary, right?

With the steps at the top of this issue is not possible in a non-windows machine to reproduce. @donovaly  has tried in a Windows machine and could not reproduce either.

PeterVKelly, 2023-02-22:
Hey Abdullah, I'm not surprised; compiling such a huge bank of source code with so many dependencies is no easy job:- you have to have all the libraries and paths set up just right or it fails, believe me. I'm a very experienced c++ programmer (and compiler) and therefore I am VERY familiar with compile failures, and, of course, tracking bugs in source code. I haven't done this for about 20 years and don't want to relearn the tricks of building a wheel. Leave it to the current crop of top guns in the freecad camp, they have probably already fixed this as it seems a simple typo or similar form of bug to me.
Cheers Abdullah (if I'm not spelling your name correctly, please forgive me and tell me without fear of offending, and I will correct my errant ways),
p. 

⁣Get BlueMail for Android ​

donovaly, 2023-02-22:
> @donovaly has tried in a Windows machine and could not reproduce either.

No, I can reproduce the issue, also with master. The only change in the LibPack was OpenSSL and that cannot be the reason, so I assume another issue. To be sure, I will compile using the LibPack we used for FC 0.20.1. I just did not have the time because this costs about 2 hours.

PeterVKelly, 2023-02-22:
I'll send you what I just said to Abdullah, and if you permit, I'll send Abdullah your comment.
Cheers, p.

donovaly, 2023-02-22:
> I'll send you what I just said to Abdullah, and if you permit, I'll send Abdullah your comment. Cheers, p.

Cheers as well but please don't pollute the GitHub comments with useless statements. Please also don't make full cites, one the parts you refer to.
I also don't understand why you advertise BlueMail for Android. This might be a nice app but has nothing to do with the issue. Maybe it is better if you login to GitHub to join the conversation more properly.

abdullahtahiriyo, 2023-02-22:
@donovaly 

We must be doing something different.

The easiest, if you download the file and open it. Do you get the issue in the video?

I get this:
[image]

donovaly, 2023-02-22:
> I get this:

I get this with today's master and also 0.20.2:

[image]

So I can confirm the bug and also that with 0.20.1 everything is fine.
Give me a day to check to compile with the LibPack we used for 0.20.1 onto master. Then i can say for sure if the LibPack is to blame or not.

abdullahtahiriyo, 2023-02-22:
@adrianinsaval 

Adrian, I wonder if you could help me out here. Could you please confirm that in Linux master we do not get this issue? 

If we got it, then it would save Uwe some quite compiling time (which appears to be pointless if the difference is really just a SSL fix) and we could do a git bisect to track the PR.

I mean if you have time, of couse.

abdullahtahiriyo, 2023-02-22:
@adrianinsaval 

Sorry. Do not bother. I have finished the PR I was handling and I am taking care now.

abdullahtahiriyo, 2023-02-22:
@donovaly 

I have just built today's master and I do not get the artifact. 

OS: Ubuntu 20.04.5 LTS (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31995 (Git) AppImage
Build type: Debug
Branch: master
Hash: b2ee277a94d0e07ae6288d3b4ce84111d7720aa1
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2

PeterVKelly, 2023-02-25:
Well done!
p.

GrzegorzKrug, 2023-03-03:
I got same issue. Maybe this could help you solve it.
[image]

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
[/code]

attachment sketch:
[angle.zip](https://github.com/FreeCAD/FreeCAD/files/10885270/angle.zip)

Bebop182, 2023-03-03:
@GrzegorzKrug it is already solved un the latest 0.21 dev build

PeterVKelly, 2023-03-03:
Great news, thanks for letting me know Bebop. I'm a total newbie and don't know the procedural protocols. I have a)ready been chastised for my sins. Cheers, p.

PeterVKelly, 2023-03-04:
Oops, sorry for premature praise, I've downloaded and tested, bug still stuck in relay contacts!
Cheers, p.

Bebop182, 2023-03-04:
I just tested the latest dev installer commit 32110 and the bug is there. But I'm running the previous non installer variant commit 31917 and the bug is not present, So it was either reintroduced or built differently.
Here's last week build if you want to try it out:
https://drive.google.com/file/d/1miMmSk-1zLebMTR10Fnukd44Nky-rtvv/view?usp=share_link

PeterVKelly, 2023-03-04:
Bebop182, man I am impressed, very well done and thank you. I envy your talent. Cheers, p.

PeterVKelly, 2023-03-06:
Thanks Andre

⁣Get BlueMail for Android ​

Almisuifre, 2023-03-09:
Hi,
I just downloaded the weekly version. I installed it and tested it.

Here is my report. Everything is going well. I no longer encounter the bug I had before. So I propose to switch fairly quickly to the branch 0.21.0.

[image]

https://drive.google.com/file/d/1JkqnShrq6Cbw-1MRSQziMloj5yo8r4TJ/view?usp=share_link

Anyway, great work.

donovaly, 2023-03-11:
Wow, after just 4 hours of testing I found the reason and a fix:
- the LibPack is not to blame
- the bug is in Visual Studio since version 17.4 (FC 0.20.1 was compiled using version 17.3.x)
- the "fix" is to disable the compiler optimization for the build target "SketcherGui" (default is /O2 and also setting /O1 fails)

I am preparing a new build for FC 0.21.dev right now and will inform you when it is online.
Then I will also make a new binary for FC 0.20.2, but this needs some hours, this the master build comes first.

donovaly, 2023-03-11:
Important info: Don't update to version 17.5 of Visual studio because it fails completely due to a bug with boost. The Microsoft guys wrote 2 days ago that they found a fix and will release version 17.5.2 in a few days. Then  will also check if the compiler optimization issue we have here is fixed too because in the forum people wrote that the boost issue also only appears with compiler optimization turned on.

abdullahtahiriyo, 2023-03-11:
@donovaly 
That is indeed a good catch. We tend to think compilers are never the problem. Most often they are not. Yet you got your black swan.

donovaly, 2023-03-11:
For those who are interested:
Here is a new Windows build of latest master: https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD-0.21.0dev-32258-Installer.exe

donovaly, 2023-03-11:
Here is a new build for FreeCAD 0.20.2: https://github.com/FreeCAD/FreeCAD/releases/download/0.20.2/FreeCAD-0.20.2-WIN-x64-installer-3.exe

donovaly, 2023-03-21:
The new installer is online since some days now -> closing as fixed.

closed 2023-03-21 by donovaly

PeterVKelly, 2023-03-21:
Thanks for letting me know, I have ben checking periodically but was unaware.
Cheers, p.

PeterVKelly, 2023-03-21:
Cheers and applause; I always trust compilers. So well done to realize the problem so quickly and expertly; again, very well done!!
Cheers, p.

Almisuifre, 2023-03-22:
Hi,

I just saw your messages. Sorry for late mojn. Life is so full of challenges, I can’t be everywhere.

I tested the dev version, it works perfectly. Bug fixed.

I am now taking the official version 0.20.3 to continue my work.

Thank you very much for your patience and also for finding a way out. Thanks to the whole team, it’s great to be able to help and fix what needs to be done.

PeterVKelly, 2023-03-22:
Hi Alexandre, I deserve no credit for anything regarding fixing this odd bug (in VS code), all I did was to clumsily  complain about it. The Fabulous Freecad Fixers did the rest, in good time and in a team oriented approach.
Thanks for your comments, but I think the real fixers deserve the praise.
Cheers, Alexandre, Peter Kelly.