FreeCAD/FreeCAD #8224: [Issue] The Union tool in Mesh Design depends on OpenSCAD, but fails to work even after installing OpenSCAD

frodeseverin, 2023-01-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=75375&p=654315&hilit=openscad#p654315

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Norwegian Bokmal/Norway (nb_NO)
Installed mods: 
  * fasteners 0.4.23
  * Help 1.0.3
  * Pyramids-and-Polyhedrons
[/code]
```

### Subproject(s) affected?

OpenSCAD

### Issue description

On Windows, using the "Union" tool in the Mesh Design Workbench on two meshes gives a dialog box informing user to install OpenSCAD. After installation of OpenSCAD using the "Union" tool in the Mesh Design Workbench on two meshes produces the error
<class 'OpenSCADUtils.OpenSCADError'>: 'OpenSCAD executable unavailable'
in the Report view. Sorry, I did not keep the whole error stack. I have included --write-log as a parameter for my FreeCAD shortcut on my Windows Menu. That should sort things for upcoming crashes and bugs. I also need to remember to usse the "Save as" feature in the Report view.

I fiddeled a bit with different things, and found a workaround:
After I loaded the OpenSCAD Workbench once in "Edit => Preferences => Workbenches" by using the "Load now" button for OpenSCAD, the problem automagically disappeared.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-01-21:
Yeah, probably Mesh Design needs to trigger a load of the OpenSCAD WB if the exe field is not populated.

luzpaz, 2023-02-13:
@chennes Are we referring to the external or internal OpenSCAD wb ?

chennes, 2023-02-13:
It's the internal WB that needs to be loaded. Who is maintaining Mesh Design?

luzpaz, 2023-02-13:
I don't think there is a designated dev for Mesh workbench.  Looks like wmayer wrote a bunch of code for it.

chennes, 2023-02-13:
OK, I'll assign this one to myself, I can probably figure this much out (famous last words, I know...).

closed 2025-01-13 by github-actions[bot]