FreeCAD/FreeCAD #8232: [Issue] IDE does not handle clang formatting correctly

WandererFan, 2023-01-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

n/a

### Version

0.21 (Development)

### Full version info

```shell
n/a
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

As reported here https://github.com/FreeCAD/FreeCAD/commit/2b7de8315088a4806f588b5647d724c93045bd93#comments the use of some clang-format options causes issues in Microsoft Visual Studio - specifically during cut and paste operations.

The cause of these issues should be investigated and resolved or work around developed.

Possible solutions:
- remove problematic options from the project level clang-format file
- change clang formatter version and/or MSVS options to support problematic options

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-22:
Related to #7858 ?

WandererFan, 2023-01-22:
see #7858 and 2b7de83150

WandererFan, 2023-01-23:
Forum user @abdullah reports that KDEvelop also struggles with clang-format.
" It has the option to set the clang formater (and I have configured it so), but it does not appear to work. "

chennes, 2023-01-24:
I believe we mean @abdullahtahiriyo :)

abdullahtahiriyo, 2023-01-25:
Yes, indeed. It is is a pity that Kdevelop is not applying the clang format file.

It apparently is known:
https://stackoverflow.com/questions/59595816/how-to-set-up-code-formatting-in-kdevelop-with-clang-format

I have even tried that hacky approach without success...

WandererFan, 2023-01-26:
I haven't noticed any issues in QtCreator when "format on save" is in effect.

I wonder if we should run clang-format as part of CI instead of asking contributors to format before submission.   That assumes that running clang-format as a stand-alone would avoid the issues with MSVS and KDevelop.

berniev, 2023-01-27:
> I wonder if we should run clang-format as part of CI 

At the risk of repeating myself, that is, and always has been, my stated aim. Right now is a testing phase. Its a bit tricky in that you can't just format all the whole project's files and wait for a reaction as that messes up git history. It needs people to try for themselves.

Once in CI it matters not what individuals do in their local environment.

All that remains is to get concensus on style. The current version is looking pretty close (after adding back braces).

If you can't use clang-format in your IDE, or don't use an IDE it is easy to run from the command line.

WandererFan, 2023-01-28:
It appears that the clang tools included with Visual Studio are behind the state-of-the-art.
https://developercommunity.visualstudio.com/t/update-clang-format-to-latest-version/1291564
https://github.com/MicrosoftDocs/visualstudio-docs/issues/7989

I have VS 17.3 and it complains vigourously about "InsertBraces"

WandererFan, 2023-01-30:
berniev's test sample (txt extension to satisfy github)
[clang-format_test.cpp.txt](https://github.com/FreeCAD/FreeCAD/files/10539402/clang-format_test.cpp.txt)

abdullahtahiriyo, 2023-02-03:
I have tried to format a file from terminal.  The clang-format available in Ubuntu 20.04 (clang-format version 10.0.0-4ubuntu1) does not support at least one entry in the clang-format file:

```
clang-format -style=file Console.cpp
YAML:5:30: error: invalid boolean
AlignConsecutiveAssignments: None
                             ^~~~
Error reading /home/abdullah/github/freecad/.clang-format: Invalid argument
```

abdullahtahiriyo, 2023-02-03:
Thanks. Eventually I will update to Ubuntu 22.04. There v15 is available:
https://packages.ubuntu.com/search?keywords=clang-format

chennes, 2023-02-03:
@abdullahtahiriyo did you mean to close this? I thought it was still an issue for VS.

WandererFan, 2023-02-04:
Applied our format file and a format file selected at random (from QtCreator project) to 3 modules.   There is no significant difference in the number of changes generated between our standard and the random file 
[image]

abdullahtahiriyo, 2023-02-04:
@chennes 

No, I did not mean to. Sorry for that.