FreeCAD/FreeCAD #8239: [Bug] [OpenSCAD] Snap & AppImage launch of OpenSCAD broken

luzpaz, 2023-01-23:
Placeholder issue to continue #7182

luzpaz, 2024-07-13:
Is this something we can make time for to fix before 1.0 ?

luzpaz, 2024-08-30:
@maxwxyz before we close all these tickets as dupes, lets sort this out since it seems like a systemic issue for the whole FreeCAD Linux ecosystem.

adrianinsaval, 2024-09-18:
this might be fixed in the latest appimage, flatpak and snap are expected to have trouble with launching external apps since they are sandboxed. Flatpak is currently shipping with a workaround that let's it launch the flatpak version of openscad by default and it might be possible to do something similar for the snap. Specifically for flatpak I think it might also be possible to add a workaround to launch an external install of openscad

luzpaz, 2024-09-20:
Snap should be a bit more liberal than flatpak and therefore easier to achieve.

adrianinsaval, 2024-09-20:
isn't snap more tight than flatpak? On a short google search I didn't find a way to launch an external application from inside snap, apparently there is a way to launch another snap from a snap as is the case with flatpak, but I don't know how convoluted to use it is. https://snapcraft.io/docs/desktop-launch-interface

luzpaz, 2024-09-23:
Just the opposite from what I've understood.

> When going through forums and [Reddit threads](https://www.reddit.com/r/linuxmasterrace/comments/u4te9z/flatpak_or_snap/) before writing this article, I found some interesting insights from app maintainers. Some of those include:

>    Snaps are easier to maintain/build compared to Flatpak packages.
>    Snap security is a tad bit advanced.
>    With Flathub as Flatpak's store, maintainers get help from various developers to test and improve the Flatpaks.

Source: https://itsfoss.com/flatpak-vs-snap/

closed 2025-01-19 by github-actions[bot]