FreeCAD/FreeCAD #8250: [Issue] Bold text is somewhat distorted

K4LCIFER, 2023-01-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.6, Qt 5.15.7, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * ProDarkThemePreferencePack 1.0.0
[/code]
```

### Subproject(s) affected?

None

### Issue description

Bold text is somewhat distorted/blurry.
[image]
[image]

Could be an issue with how anti-aliasing is being handled on bold text.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-24:
Is this theme related?  (If you switch themes does this issue persist?)

K4LCIFER, 2023-01-26:
Ah it may be a theme issue with ProDark.

Here is no stylesheet:

[image]

Darker Blue theme:

[image]

K4LCIFER, 2023-01-26:
However, the tree view is still somewhat distorted, I am noticing:

Darker Blue theme:

[image]

No style sheet:

[image]

luzpaz, 2023-03-15:
@MisterMakerNL can you have a look when you get a chance ?

MisterMakerNL, 2023-03-15:
ProDark is one of the only one that use bold fonts, looks like that is the issue.
Found similar issue here but no solution:
https://forum.qt.io/topic/114649/why-fonts-in-qt-are-appearing-blurry-or-pixelated

luzpaz, 2023-03-18:
Thanks mistermaker. 

CC @turn211 FYI

luzpaz, 2023-04-15:
Can anyone locate an upstream Qt ticket for this pixelated issue ?

chennes, 2023-04-15:
Does the font being used there have a real bold variant, or is it being faked by the renderer?

adrianinsaval, 2023-04-28:
sounds like a font configuration problem, @luzpaz I don't think we can target this for 0.21 since it's most likely a system/packaging problem not something on freecad's code.

turn211, 2023-04-29:
I took a look at my theme and the bold text is crisp and perfect on Windows. So Linux related.

adrianinsaval, 2023-04-29:
Check your font settings on qt5ct, my first suspicion is that gnome doesn't properly configure Qt applications. What distro are you using?

turn211, 2023-04-29:
I’m only on windows but I did test on Manjaro KDE live iso and fonts displayed properly.

xtemp09, 2023-04-30:
@K4LCIFER, could you run FreeCAD with the following command:
```ruby
QT_SCALE_FACTOR=1.3 FreeCAD
```
varying the scale factor from 1.0 to 2.0?

adrianinsaval, 2023-05-23:
@K4LCIFER please provide feedback, otherwise I think we should close the issue since it's not actionable as is

luzpaz, 2024-07-27:
bump @K4LCIFER

obelisk79, 2024-08-17:
@luzpaz @maxwxyz I am unable to reproduce this text rendering issue on linux. Everything bold is crisp as reported by others. I suggest this issue be closed as it hasn't really be reproducible and the originator has seemingly abandoned this report.

closed 2024-08-18 by maxwxyz