FreeCAD/FreeCAD #8265: FreeCAD stop responding after a user tries to Copy a cell or cells in Spreadsheet WB[Issue]

PAS-DRAFTER, 2023-01-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=8&t=73520&start=10

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31641 (Git)
Build type: Release
Branch: master
Hash: d28d63b87b60161419c6c0c532fbbfaed96926b8
Python 3.10.8, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.4
  * CurvedShapes 1.0.4
  * Curves 0.6.4
  * DynamicData 2.46.0
  * fasteners 0.4.25
  * fcgear 1.0.0
  * sheetmetal 0.2.59
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Issue description

I got a new computer earlier this year and installed the latest version of FreeCAD (at that time) on it. For the most part, it seemed to be working fine. However, as I was working on a project with spreadsheets, I noticed that it would stall out for ~20 to 30 seconds, and the computer fan would kick on when I would select a single cell or multiple cells in a spreadsheet and then try to copy those cell(s).  During this time, I could do nothing in FreeCAD. Eventually, it would come back to normal, and I could continue working again.  I can get this to happen if I open a new file, create a new spreadsheet, and then just try to copy some empty cells.

If I just copy text from one cell to another one, I do not see this problem happen.  I have occasionally seen the problem happen when I try to copy the text in the Alias to use in another cell.  However, this happens only occasionally, and I cannot determine what events cause it to happen.  Trying to copy a cell or cells seems to always cause it.  Occasionally the Paste command causes a slightly shorter delay, but if you paste cells multiple times after the first delay, there is no more delay.  

Initially, I thought this was related to [https://github.com/FreeCAD/FreeCAD/issues/8061](url) and posted my findings there, but as I reread the information posted there, I realized although the symptom was the same, the cause is completely different.  I am not having a problem with the computer stalling when I just open a spreadsheet.  That part seems to work fine in my case.  It is only when I copy a cell or group of cells.  Since this seems like a different issue, I decided to start a new one that describes this problem instead, as I think the other issue has been resolved.

Although I have posted much of this information in the other post, I will reiterate my findings here to keep things simple.

It was suggested that I turn off auto recomputes to see if that helped. I gave that a shot, however, it did not have an effect on the issue.

During my testing to try and find a solution, I ran across the above post and tried to use the version given in the link to solve the issue.  However, it did not solve the copy issue.

I had another computer in my office next to the new one I was trying to get to work.  I went and tried coping and pasting on that one, and it worked just fine.  I looked at the version I had installed on that computer and decided to uninstall everything, including the configuration files, and install it on my new machine since it appeared to work.  However, after I installed that same version on the new machine and tested it, it WOULD NOT copy without stalling out.  I found that interesting as I was using the same version (even the same file) on both machines.

The first thing I realized was that I was running Windows 10 on my working machine and the new machine came with Windows 11 installed.  I also noticed that the "Copy Info" button showed that both computers were running Windows 10, which is not correct.  I learned that this is a known issue.  However, this does mean that people may not know from the system info which OS is actually being used and could be confusing the matter.

The next thing I tried was taking a laptop that had Windows 10 originally installed and that had been upgraded to Windows 11 and installed that same version of FreeCAD.  That computer running Windows 11 DID work properly.  So my simple testing seems to show that this problem happens on clean Windows 11 installs but not on previous Windows versions or Windows versions that were older and eventually upgraded to Windows 11.  I suspect the confusion on finding the solution to this problem is further complicated by FreeCAD not giving the correct OS in the Copy Info.

According to the Forum discussion linked to this report, the first person to report it did so back on November 22, 2022, so it has been around for a while.  I have seen some other posting that have been able to duplicate the problem with the conditions I have documented, so at least I know it is not an isolated case.  Since I do not believe this problem and the stall described in the other ticket are the same issue, I am hoping that by putting this in its own bug report, that it can start to be treated as a separate incident and hopefully someone can find the cause.  I use the Spreadsheet WB a lot and this issue is killing me.  I have a brand new computer and basically can't use it.

If anyone needs more information or wants me to test things, I am more than happy to help out and provide feedback.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

acpopescu, 2023-01-31:
Yeaaah, I can repro this on my git master build. Something's off with UI, it's accessing somehow ALL of the POSSIBLE cells:
[image]

acpopescu, 2023-01-31:
It's basically stuck in this windows component, traversing the entire sheet....

[image]

PAS-DRAFTER, 2023-02-01:
> It's basically stuck in this windows component, traversing the entire sheet....

I guess the easy fix is to make a smaller spreadsheet :-)

That would explain why it seems to be stalling as I'm sure that would take a lot of time to look at each cell.  Was this duplicated using a Clean Windows 11 build or were you able to duplicate it with another version?

Thanks for digging into it deeper and getting closer to a solution.  Obviously, it has been driving me mad since I got my new machine.

acpopescu, 2023-02-01:
This is clean build on Windows 11 - it actually freezes when I'm trying to select something. Once that 30 seconds is gone, it works fine. 

You can try and reduce the max sheet size locally for you in SheetModel.cpp:
```
int SheetModel::rowCount(const QModelIndex &parent) const
{
    Q_UNUSED(parent);
    return 16384;
}

int SheetModel::columnCount(const QModelIndex &parent) const
{
    Q_UNUSED(parent);
    return 26 * 26 + 26;
}
```

acpopescu, 2023-02-01:
For the proper fix, I'm afraid a QT source debug build would be needed.

PAS-DRAFTER, 2023-02-01:
The copy problem seems to be getting worse in the latest weekly build.  I still have the problem every time I try to copy a Cell, but I've noticed that I also see it happens around 1 in 3 times (just a guestimate) when I try to copy the Alias.  I had noticed this before, but it seems to be happening more now.

I do not know if the cell copy and the alias copy are related to the same root issue.  The stall seems to be less for the Alias copy than the cell copy.

acpopescu, 2023-02-01:
Alias copy, as in copy-paste the alias somewhere else?

That could be caused by the document recompute - you can try and disable recomputes while you're working with the Spreadsheet - try https://wiki.freecadweb.org/Spreadsheet_Workbench#Complex_models_and_recomputes

acpopescu, 2023-02-01:
Narrowing down the problem it looks it's related to Windows ACCESIBILITY.

QT implements an accessibility interface - that interface has a windows implementation, and windows will use it. It is possible that an update to windows caused a change in behavior, where QT is traversing every single cell/grid based on windows accessibility requests.

acpopescu, 2023-02-01:
To fix this, we'd need to create our own accessibility interface for the SheetViewModel:
`QAccessible::installFactory(qMySheetViewMode)`

And in there take into account that rowCount and cellCount are virtual to avoid traversing the entire virtual space.

To disable it:
```
class QtSheetModelNoAccesibility: public QAccessibleWidget
{
public:
    QtSheetModelNoAccesibility(QtTestAccessibleWidget *w): QAccessibleWidget(w) {}
    QString text(QAccessible::Text t) const override
    {
        if (t == QAccessible::Help)
            return QString::fromLatin1("Help yourself");
        return QAccessibleWidget::text(t);
    }
    static QAccessibleInterface *ifactory(const QString &key, QObject *o)
    {
        if (key == "SpreadsheetGui::SheetTableView")
            return new QtTestAccessibleWidgetIface(static_cast<QtSheetModelNoAccesibility*>(o));
        return 0;
    }
};

...

QAccessible::installFactory(QtSheetModelNoAccesibility::ifactory);
```

acpopescu, 2023-02-01:
@PAS-DRAFTER  if you are blocked, and can compile it for windows, you can do the changes highlighted in the image below 
[image]

It will cause other errors, but won't lock up
```
16:58:45  Cannot create accessible child interface for object:  SpreadsheetGui::SheetTableView(0x278551caed0, name = "cells")  index:  1407
```
Text for reference:
```
#include "SheetTableView.h"
#include "QtWidgets/qaccessiblewidget.h"

class QtSheetModelNoAccesibility : public QAccessibleWidget
{
public:
    QtSheetModelNoAccesibility(SpreadsheetGui::SheetTableView* w) : QAccessibleWidget(w) {}
    QString text(QAccessible::Text t) const override
    {
        if (t == QAccessible::Help)
            return QString::fromLatin1("Help yourself");
        return QAccessibleWidget::text(t);
    }
    static QAccessibleInterface* ifactory(const QString& key, QObject* o)
    {
        if (key == QString::fromUtf8("SpreadsheetGui::SheetTableView"))
            return new QtSheetModelNoAccesibility(static_cast<SpreadsheetGui::SheetTableView*>(o));
        return 0;
    }
};
...

/* Python entry */
PyMOD_INIT_FUNC(SpreadsheetGui)
{
...

    // instantiating the commands
    CreateSpreadsheetCommands();

    **QAccessible::installFactory(QtSheetModelNoAccesibility::ifactory);**

    SpreadsheetGui::ViewProviderSheet::init();

```

PAS-DRAFTER, 2023-02-02:
> Alias copy, as in copy-paste the alias somewhere else?

Yes, I often select the Alias from the window and copy it to paste it into the equation in another cell or reference box, so I don't have to remember it or type it later.  The delay seems to happen after I hit Control C and am trying to leave the Alias box to select another cell.  It doesn't happen all the time.  In fact, right now it is not near as bad as it was yesterday.  I have no idea what the trigger is as it seems to be random.

If I experience it a lot again, I'll give the "turn off recomputes" option to see if it makes it better.

PAS-DRAFTER, 2023-02-02:
@acpopescu I'm afraid I have never tried to compile anything.  If I were to do so, I'd need a great deal of guidance on how to do.  I greatly appreciate the work you did to get to the bottom of this.  It looks to me like you have committed changes to disable this.  Would that mean it would be included in the next weekly build?  If that is the case, I can wait until Monday and download that to test.

THANKS AGAIN

acpopescu, 2023-02-02:
@PAS-DRAFTER  - I've pushed a PR to the maintainers, but it's up to them to accept it or not. It disables any accessibility features that were on the spreadsheet table itself, and that could be an issue for that. Let's see if it makes it in or not before you trying to compile anything first :)

PAS-DRAFTER, 2023-02-03:
Did something change in:

chennes, 2023-02-07:
Thanks for the follow-up @PAS-DRAFTER -- I've just merged @acpopescu's workaround, which worked well for me. I'm going to close this issue as solved, but if the problem recurs, please feel free to reopen.

closed 2023-02-07 by chennes