FreeCAD/FreeCAD #8272: [Feature Request] Implement Keep Tool Down for MillFace

more10, 2023-01-26:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=75519

### Subproject(s) affected?

None

### Idea description

Planing a surface using "Offset Pattern" : "Offset" could be improved by keeping the tool down. The zigzag pattern keeps the tool down but is not adhering to the Keep Tool Down checkbox. 

The zigzag angle in Pocket is affecting the pattern in MillFace. Consider putting common parameters in a separate box.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

more10, 2023-02-02:
Apparently there is a problem with the corners when the stepover is > 70%. Consider improving the path to also handle the corners. See the forum thread.

luzpaz, 2023-02-05:
@luvtofish would you be interested in looking in to this?

closed 2025-11-28 by chennes