FreeCAD/FreeCAD #8274: [Feature Request] I10N: Ease language change in task panels created from .ui using Python

jbaehr, 2023-01-26:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://github.com/FreeCAD/FreeCAD/pull/8247#issuecomment-1403664088

### Subproject(s) affected?

Other (specify in description)

### Idea description

When creating task panels in C++, one can override `QWidget::changeEvent(QEvent*)` and react to `QEvent::LanguageChange` to retranslate the UI, cf. e.g. [here](
https://github.com/jbaehr/FreeCAD/blob/30ad918c63df49a24d5f8041ba3683922765b1b3/src/Mod/PartDesign/Gui/TaskHelixParameters.cpp#L577)

Trying the same with Python is not that easy (which is probably the reason why I haven't found any Python task panel doing it):
- [`TaskDialogPython`](https://github.com/jbaehr/FreeCAD/blob/30ad918c63df49a24d5f8041ba3683922765b1b3/src/Gui/TaskView/TaskDialogPython.cpp) offers a very convenient way to setup ones task panels. It calls ones methods (if defined) like `accept`, `reject`, `getStandardButtons`, and some more.  It does not offer such a hook for `changeEvent` or some potential higher-level `retranslateUi` to the like. So to react on this, one has to roll out a custom solution. E.g. the Draft-WB seems to [provide a custom base class](https://github.com/jbaehr/FreeCAD/blob/30ad918c63df49a24d5f8041ba3683922765b1b3/src/Mod/Draft/DraftGui.py#L193) emitting a `retranslate` signal, which again makes it harder to just use .ui-files to define the task panel.
- The widget created by `FreeCADGui.PySideUic.loadUi`, to be assigned to `self.form` when setting up ones task panel, does not seem to have a `retranslateUi` method, in contrast to the python code generated by uic. So even if `TaskDialogPython` would offer a retranslation hook, one could only use it with pre-generated or hand-written widgets.

My wish is be to offer the same seamless adaption to language change for task panels written in Python than the ones written in C++. And all that without much boiler-plate for the Python developer. Ideally, `TaskDialogPython` would automatically call the `retranslateUi` method on the generated widgets in `self.form`, if present. In cases where additional translations need to be made, e.g. when using the property docs as tool tips, the hook could be used like this:
```py
class MyTaskPanel:
    def __init__(self, obj, mode):
        self.form = FreeCADGui.PySideUic.loadUi(pathlib.Path(__file__).with_suffix(".ui"))
        self.assignToolTipsFromPropertyDocs()
        ...
    def retranslateUi(self):
        self.form.retranslateUi()
        self.assignToolTipsFromPropertyDocs()
```

### Anything else?

Overall I have to admin that this is only a minor issue (when just not reacting to language changes): After closing and reopening the task panel, the language change is honoured. And, frankly, one does not change the ui language that often. Anyway, I stumbled upon this while working on #8247 and was encouraged to open a ticket for it. That way, it doesn't get lost.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jbaehr, 2023-02-22:
@chennes I do not think this is done. Unless I misses something, but #8247 does definitely not implement this request, quite the opposite: it does not feature instantaneous reaction to a change in language configuration exactly because this request here is not yet available.

chennes, 2023-02-22:
Yes, sorry -- it closed automatically because it had gotten linked. Sorry I didn't notice it sooner, it's reopened now.

luzpaz, 2023-02-25:
My bad, I linked it.

jbaehr, 2023-03-01:
I think @wwmayer implemented both parts of this feature request with #8602.

I was not able to test it myself, though. An OS upgrade broke my compilation environment, yet waiting for me to fix it, and the weekly build is [currently broken](https://forum.freecad.org/viewtopic.php?t=76338), too.

If you want to test and have a recently compiled FreeCAD available, you could give this unverified change of mine a try: https://github.com/jbaehr/FreeCAD/commit/d3117cdadee107a65603d8e37e16e738726ca0d7 and see whether the PD:InvoluteGear's TaskPanel retranslates on language change without having to close/reopen it.

jbaehr, 2023-03-05:
I can now confirm that this feature request is fulfilled with #8602.

I just filed #8746 to make use of this new feature for the PD:InvoluteGear's TaskPanel.

closed 2023-03-05 by jbaehr