FreeCAD/FreeCAD #8278: Navigation Cube - NOT FIXED. Per #8082 nor #8266

bstott, 2023-01-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2004
Word size of FreeCAD: 64-bit
Version: 0.21.0.31695 (Git)
Build type: Release
Branch: master
Hash: 64caea207394bb05587c8ffb29b7a3e8696d24df
Python 3.10.8, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools
  * A2plus 0.4.60g
  * Assembly4 0.12.5
  * boltsfc 2022.11.5
  * CfdOF 1.20.6
  * CurvedShapes 1.0.4
  * Curves 0.6.5
  * Defeaturing 1.2.0
  * Design456 0.0.1
  * DynamicData 2.46.0
  * EM
  * Estimate 0.1.2
  * ExplodedAssembly
  * fasteners 0.4.50
  * fcgear 1.0.0
  * FEMbyGEN 1.2.0
  * FEM_FrontISTR 0.1.0
  * freecad.gears 1.0.0
  * Help 1.0.3
  * InventorLoader 1.3.0
  * lattice2 1.0.0
  * Launcher
  * Manipulator 1.4.9
  * offline-documentation 1.0.0-alpha
  * OpticsWorkbench 1.0.9
  * parts_library
  * pcb
  * PieMenu 1.2.4
  * POV-Ray-Rendering
  * QuickMeasure 2022.10.28
  * Silk 1.0.0
  * slic3r-tools
  * SlopedPlanesMacro
  * symbols_library
  * ThreadProfile 1.84.0
  * toSketch 1.0.1
[/code]
```

### Subproject(s) affected?

None

### Issue description

The Navigation Cube issue is worse. The same but, more of the cube is non-functioning. 

Now, the old problem. Plus, the up and right ARROWS do not work, either.

I am using the latest weekly bundle downloaded Jan 26, 2023 @ 9:20pm Central Time.

The entire problem still exists and is worse.   Also, not addressing the navigation problem there seems to be a related issue with menu drop downs of File, Edit, and Save posting in another issue. Seem to be same actions. So, same issues?  Sections of screen/window not responsive to mouse over and clicks.

Sorry ---- you are attempting. 

The new cube shape is odd. The corners look are quite extreme. But, I guess - ok. More pointed corners with larger selection arias would be more esthetically pleasing.   But --- having it simply work - would be best.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-01-27:
That build is before 588f2d50e4e108299a754e5c229cdc396342d2dd was committed. To get that commit you'll have to compile yourself, or wait for the weekly to be regenerated.

bstott, 2023-01-27:
So, closed issues are still before beta.  My bad. 
Thanks.

closed 2023-01-27 by luzpaz