FreeCAD/FreeCAD #8279: [Problem] Is the installer of "FreeCAD-0.20.2-WIN-x64-installer-2" broken?

softempire, 2023-01-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
I always got the error message
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

donovaly, 2023-01-27:
For this we have the checksum. Download the checksum file and compare it with the checksum of what you have downloaded. Here is my result:
[image]

By deleting the downloaded file and re-downloading your problem should be solved.

donovaly, 2023-01-27:
Moment, from where have you downloaded the file?
Your screenshot shows as if the installer requires admin permissions. That is not the case. So in your icon shouldn't be the small shield icon.

closed 2023-01-27 by donovaly