FreeCAD/FreeCAD #8285: [Issue] Angle constraint is not displayed where expected

usarskyy, 2023-01-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 1909
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Sweden (en_SE)
Installed mods: 
  * Assembly4 0.12.5
[/code]
```

### Subproject(s) affected?

PartDesign

### Issue description

After I updated from v20.1 to v20.2 I see the following weird behavior: angle constraint is displayed between an axis line and an invisible line instead of two lines where it was initially added. It is visible when those two lines are above the axis line ("zero point"), if one of them is on the axis line then you can't see that problem.

[image]

### Anything else?

I could reproduce it multiple times with different sketches. But if you need a sample created specifically on my machine, let me know.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-02-06:
Thanks for the report but there's already an existing issue with detailed information recorded see https://github.com/FreeCAD/FreeCAD/issues/8220 Could you therefore please close this as a duplicate.

usarskyy, 2023-02-06:
> Thanks for the report but there's already an existing issue with detailed information recorded see #8220 Could you therefore please close this as a duplicate.

Sorry, somehow I missed that issue.
I will close mine right now.

closed 2023-02-06 by usarskyy