FreeCAD/FreeCAD #8288: [Feature Request] Sketcher - resize constraint icons independent from the font size

Intuos5, 2023-01-28:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?style=2&f=10&t=49972&start=70

### Subproject(s) affected?

Sketcher

### Idea description

I would like to resize the constraint icons independent from the font in the Sketcher workbench. The reason being, the icons are half the size of the font for me. I want to increase the constraint icons to improve readability, but I don't want very large dimension texts, because that deteriorates readability. 
Please add a dedicated constraint icon size parameter in the preferences of the Sketcher workbench. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-28:
You can attach images to the ticket. Before/After images are welcome.

Intuos5, 2023-01-29:
Don't mind the silly sketch, it's for illustration purposes. Also note that this is on a 27 inch 4k screen.
Text size is correct, constraints are too small (text size is 20): 
[image]

Constraints are readable, but text is too large (text size is 40): 
[image]

This is how I would like it to be (text size is 20, icon size is 40):
[image]

luzpaz, 2023-02-06:
Not directly related but in the same vein to #7030

LancashireDave, 2023-07-28:
I use a high res 43" monitor and the icons are so small I can hardly see them. The ability to change the icon size would be a great improvement

luzpaz, 2023-09-29:
@Syres916 offered a possible solution in https://github.com/FreeCAD/FreeCAD/compare/master...Syres916:FreeCAD:patch-64

luzpaz, 2023-09-29:
@PaddleStroke can you review proposed change ?

PaddleStroke, 2023-09-29:
@Syres916 why keeping the dependency to the font size if we introduce a parameter for the size of icons ?

`    Client.drawingParameters.constraintIconSize = std::lround(sketchericonSizeRatio * 0.8 * sketcherfontSize);`

I think if we introduce a setting, we might as well remove the sketcherfontSize from the equation and just use a size preference for the icons.

adrianinsaval, 2023-10-09:
I don't agree with adding a separate preference for this, instead it looks to me like there's a problem with how the text and the symbols are scaling at different rates for this person and it should be fixed to be more sensible. Here's how it scales for me, at 40px:
[image]
at 20 px:
[image]

Intuos5, 2023-10-10:
If possible, I would prefer independent scaling of text and icons, because I prefer to set them at scales that make it most comfortable for me. I may want smaller text and larger icons for legibility, others may want the same size, etc.

adrianinsaval, 2023-10-10:
if implemented I would suggest it be through a parameter only, let's not keep polluting preference pages with obscure options to satisfy individual wishes.

luzpaz, 2024-03-09:
Is this on the 1.0 roadmap ?

ChrisAbra, 2024-06-06:
> keep polluting preference pages with obscure options to satisfy individual wishes.

Seems strange that this obvious accessibilty feature would be the line to draw.... 

It also makes sense to tie it somewhat to font-size as the default behaviour would want them scaling together but the ability to override is important too as clearly demonstrated above.

adrianinsaval, 2024-07-01:
> Seems strange that this obvious accessibility feature would be the line to draw....

the ability to scale is an accessibility feature, this already exists. In what situation do you need to scale one but not the other?

ChrisAbra, 2024-07-02:
I think its reasonable to want to have legibly sized constraints icons without having giant text as demonstrated in the images above. There are multiple users who say this will benefit them and one saying they dont like it for "clutter" or "works on my machine" reasons.

"Youre doing it wrong" cant keep being the FreeCAD approach

adrianinsaval, 2024-07-02:
see my previous comment: https://github.com/FreeCAD/FreeCAD/issues/8288#issuecomment-1753607495