FreeCAD/FreeCAD #8289: [Bug] Input boxes on latest builds dont "capture" inputs

KenwoodFox, 2023-01-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31695 (Git) AppImage
Build type: Release
Branch: master
Hash: 64caea207394bb05587c8ffb29b7a3e8696d24df
Python 3.10.8, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.2.59
  * kicadStepUpMod 10.16.5
  * Assembly4 0.12.4
  * A2plus 0.4.60f
  * fcgear 1.0.0
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Issue description

Any and every input box, does not "capture" the input, if you press f2 to edit the name of a body, you can only input keys witch have no keybindings.

[image]

Same with numbers witch makes this really hard to workaround, inputting numbers 1, 2, 3, 4 don't type the number but instead rotate the camera. Even with the box selected.
[image]

A workaround i found is for boxes that support it to use the formula editor
[image]

### Anything else?

Thanks for reading!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

KenwoodFox, 2023-01-28:
I don't see a lot of other activity or issues like this, its repeatable for me on my machine and a VM, ill try and see if i can put together a "steps to reproduce" but, for me any of the latest builds seem to have this issue.

Any advice on how/where to capture more detailed logs would be appreciated! Normally i expect once I'm typing in a box, that hotkeys will not be interpreted. Instead any hotkey is just lost and performs its action, even when in an edit box.

chennes, 2023-01-28:
You've got a couple of options for logging: the easiest (I think) is to run FreeCAD with the `--log-file /path/to/file` option, which will cause it to create a log file in the given path.

KenwoodFox, 2023-02-01:
[freecadLog.log](https://github.com/FreeCAD/FreeCAD/files/10552118/freecadLog.log)

Heres what i captured! I went ahead, made a 1x1 cube, extruded it, and pressed `1` then `"` to input that i wanted a 1 inch thick pad.

Instead of actually typing `1` in the box though, the camera rotated to face the front of the part, as if i had pressed 1 without the text box being selected.

I was able to make the 1x1 part by using the function editor, inputting `1"` and then using that.

KenwoodFox, 2023-02-02:
[image]

This is what i see, in the gif i try to pad the sketch, input `1.00` into the box, but it just rotates the camera around.

Same when i try and rename the body, i cant type upercase `B` and space just hides it, like you would expect if the text box was **not** selected. Does anybody else experience this or is it only me?

KenwoodFox, 2023-02-04:
Im now on
```
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31709 (Git) AppImage
Build type: Release
Branch: master
Hash: e188802ca6997d2564e7570ab648462e6a059f87
Python 3.10.8, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.2.59
  * kicadStepUpMod 10.16.5
  * Assembly4 0.12.4
  * A2plus 0.4.60f
  * fcgear 1.0.0
[/code]
```

But after some more testing, I dont have it using this version:
```
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.2.59
  * kicadStepUpMod 10.16.5
  * Assembly4 0.12.4
  * A2plus 0.4.60f
  * fcgear 1.0.0
[/code]
```

SO whatever is broken must be something between 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8 and e188802ca6997d2564e7570ab648462e6a059f87 ?

https://github.com/FreeCAD/FreeCAD/compare/930dd9a76203a3260b1e6256c70c1c3cad8c5cb8...e188802ca6997d2564e7570ab648462e6a059f87 Thats a lot of diff :<

KenwoodFox, 2023-02-06:
If there's some more debugging i could do with the new version (https://github.com/FreeCAD/FreeCAD/commit/e188802ca6997d2564e7570ab648462e6a059f87) please let me know! Id be happy to get some more verbose logs!

KenwoodFox, 2023-02-11:
Let me know if theres more logging or debugging i can do on my end!

KenwoodFox, 2023-03-03:
Still experiencing this issue on `0.20_31917`

chennes, 2023-03-03:
I believe this is the same issue described in #8346 -- I'm going to close this ticket and centralize the discussion there. This should be fixed when https://github.com/conda-forge/qt-feedstock/pull/239 gets merged.

closed 2023-03-03 by chennes