FreeCAD/FreeCAD #8290: [Feature Request] Pop-up command search (like Blender's)

Intuos5, 2023-01-28:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?t=61175

### Subproject(s) affected?

Other (specify in description)

### Idea description

I would like to have a pop-up command search. Ideally, it would work the same way as Blender's:
[image]
You set up a hotkey which opens the search menu. It uses a fuzzy search, so you can still make typos and get the desired search results. On the right hand side, it displays hotkeys mapped to the command. 

### Anything else?

I know that Triplus' Launcher addon is able to do command searches. You can even search by pressing a hotkey. But the downsides are that it does not show hotkeys and you cannot open Launcher by calling a hotkey. This means you need to dock the window. Moreover, it doesn't have a search icon in the text field, which would make it more recognisable. Another downside to Launcher is that it only highlights results that are a very close match, instead of always highlighting the first search result. The latter would enable you to hit enter and execute any command. But now you need to explicitly activate a specific command by either clicking it or pressing the arrow keys before hitting enter. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-28:
See https://github.com/triplus/Launcher and its successor https://github.com/SuzanneSoy/SearchBar

luzpaz, 2023-02-04:
@SuzanneSoy would you be open to adding this type of functionality in to FreeCAD core ?

luzpaz, 2023-03-03:
@SuzanneSoy ping

SuzanneSoy, 2023-04-12:
Compatibility issue fixed, let me know if there's anything I can do to help migrate this into the core, if we wish do do so.

luzpaz, 2023-04-12:
CC @chennes @sliptonic  
Should we starts a discussion about the possibility of merging https://github.com/SuzanneSoy/SearchBar in to core ? Maybe we can also create a search for Preferences dialog as well ?

chennes, 2023-04-12:
I certainly think that it would be a popular addition, it's one of the Addons I see recommended the most often on the Forums. I don't know that we really have a "process" for integrating an external addon, maybe it's time we made one.

luzpaz, 2023-04-12:
Firstly, we should talk about Licenses. Looks like SearchBar is [CC0 1.0 Universal](https://github.com/SuzanneSoy/SearchBar/blob/main/LICENSE). We need to figure out if it's compatible with LGPLv2 ?

chennes, 2023-04-12:
CC0 is a public domain dedication: it's compatible with everything 😄

SuzanneSoy, 2023-04-13:
> CC0 is a public domain dedication: it's compatible with everything smile
Indeed, that's the whole point of using it, you don't need to ask any questions or think :-p

First these two critical bugs need to be fixed/mitigated:
- [ ] https://github.com/SuzanneSoy/SearchBar/issues/12 (segfault on right click→copy in the "info" pane, due to GC of the widget before the context menu disappears)
- [ ] https://github.com/SuzanneSoy/SearchBar/issues/18 (global keybinding RETURN stolen, no clue why, should be some silly bug)

There is an issue with the FreeCAD core itself which causes *lots* of instability with the addon, and needs to be fixed by core developpers (I've done as much circumventing as possible):
- [ ] https://github.com/SuzanneSoy/SearchBar/issues/25 (FreeCAD uses a modified `QuarterWidget` library and doesn't re-export Python bindings, so the only way to create a 3D preview is to create a tab at the same time with `FreeCADGui.createViewer`, and the tab *really* wants to be garbage-collected and destroy the `QuarterWidget` before I'm done with it).

xtemp09, 2023-04-13:
Most programs put search bar at the top-right corner of the window or implement it as Blender-style pop-up. 

Why is 3D preview of Quarter widget required for command search?

Intuos5, 2023-04-13:
In my opinion there are two good options to implement this. 
1. Make the pop-up appear at a fixed location on screen, like Krita (so you know where you should look)
2. Make the pop-up appear next to the cursor (which is where you are always looking)

The third possibility is what Triplus' Launcher does, which is add the search box to a panel. This is also what Qgis (Qt based FOSS) does. So perhaps it's possible to look at the Qt code for this. But, showing it in a fixed location always consumes UI space even if you are not using the command search. Which is why I think it is the lesser option of the three. You can indeed click on the search bar, but with option 1 or 2, you can still invoke search with a hotkey.
For inspiration/ reference:

https://user-images.githubusercontent.com/63149655/231756796-323a323d-0389-43f8-81dc-aa364536a6da.mp4

yorikvanhavre, 2023-04-13:
This is a cool add-on. I imagine in the future when we have a properly integrated documentation system that the search bar could also be extended to search through the documentation, etc..  And it could be a huge help for new users struggling with FreeCAD.

Technically, adding an add-on as a "core" component of FreeCAD requires nothing more than copying it all inside a folder in src/Mod.

The main question is that it changes "management". @SuzanneSoy you would need to go through the whole PR process if you wish to bring changes, which is more annoying. If you plan to develop more features, you might prefer to keep it where you have more control and can tweak it more easily...

Otherwise, I'd be in favor of including it too!

luzpaz, 2023-04-13:
This is a QOL type feature, especially for a complex software like FreeCAD. I think we should add to core even if it may make the maintainer need to jump through hoops to add functionality.

chennes, 2023-04-13:
Again, I'm for it, if @SuzanneSoy is OK with the added step for the maintenance of the feature. Maybe @wwmayer can comment on the issue you are facing with garbage collection in https://github.com/SuzanneSoy/SearchBar/issues/25.

SuzanneSoy, 2023-04-15:
> Most programs put search bar at the top-right corner of the window or implement it as Blender-style pop-up.

Yes, I would've put it there, but FreeCAD's main menu bar (the rectangle containing `File Edit View …`) is a very basic Qt widget which allows only textual menus, so I can't add anything like a search box there.

> Why is 3D preview of Quarter widget required for command search?

The search bar shows results for  tools, objects and preferences. The info pane for objects shows a 3D preview of the object (e.g. selecting the result `Cube001` will show a preview of that object alone, clicking the result will select that object in the tree view).

xtemp09, 2023-04-15:
Cool. Since it is going to be added to the core, I think, it is very possible to add it to the top-right corner.

SuzanneSoy, 2023-04-15:
> In my opinion there are two good options to implement this.
>
> 1. Make the pop-up appear at a fixed location on screen, like Krita (so you know where you should look)
> 2. Make the pop-up appear next to [...]

These four (somewhat overlapping) issues track those possible improvements:

https://github.com/SuzanneSoy/SearchBar/issues/22
https://github.com/SuzanneSoy/SearchBar/issues/28
https://github.com/SuzanneSoy/SearchBar/issues/24
https://github.com/SuzanneSoy/SearchBar/issues/17

Combined together, you should be able to decide to have a hotkey to pop-up the search bar under your cursor, or add it as a search field to a toolbar, and optionally collapse that search field to a single 🔎 icon (clicking it would pop up the search field to the right/left depending on whether it's too close to the right edge of the screen), or disable all of these and use your own 5 lines of Python to have the personalized behaviour that suits you.

SuzanneSoy, 2023-04-15:
> Cool. Since it is going to be added to the core, I think, it is very possible to add it to the top-right corner.

That was initially where I tried to put it :grin: but wasn't enthusiastic about replacing the main menu bar with a dirty hack :-p

xtemp09, 2023-04-15:
The workbench selector can be placed to the top-right corner. 🙄

SuzanneSoy, 2023-04-15:
> The workbench selector can be placed to the top-right corner. roll_eyes

@xtemp09 How do you do that? With FreeCAD 0.20.2, if I drag&drop the workbench selector toolbar, I can't drop it to the right of `File Edit View Tools Macro Windows Help Accessories`, it just becomes a floating toolbar.

xtemp09, 2023-04-15:
@SuzanneSoy

<details><summary>Screenshot 1</summary>

[image]

</details>
<details><summary>Screenshot 2</summary>

[image]

</details>

SuzanneSoy, 2023-04-15:
Oh. This was implemented in https://github.com/FreeCAD/FreeCAD/pull/7679 and is part of the current dev version / future V1 version, it's not in 0.20.2. Couldn't find an AppImage for that if there is one.

[image]

luzpaz, 2023-04-15:
> If all that is done is copying this to `src/Mod`, then I assume that with a tiny bit of code cleanup, I should be able to have it both installed as part of FreeCAD core `src/Mod` and in my `~/.FreeCAD/Mod` (would have two search bars visible). That would allow me to tinker with enhancements and PR them to the core when they're stable :) . If we start moving files around and scattering changes in the FreeCAD repo, it'll make improvements/upgrades much more difficult, so a simple copy to `src/Mod` is a :heart:  :+1: from me.

Well, it's a good point. Maybe some of the devs can take a look and respond ?

chennes, 2023-04-16:
I don't think you will have two bars visible, the one in ~/.FreeCAD/Mod should just take precedence, but I haven't looked at your source to see for sure. At any rate, I'm sure we can make something like that work. Do you want to get your code to a good stopping place and maybe add a git tag? Then I can copy it in from a well-defined point.

SuzanneSoy, 2023-04-17:
Sounds good to me, I'm finishing off a small side project for a couple more days before I get back on to fixing the few remaining critical issues. I'm not in a stable situation and can't make promises, unfortunately.

yorikvanhavre, 2023-04-19:
Or, make a PR yourself @SuzanneSoy so you keep the "git authorship" of the files... Wouldn't feel correct to not see you anywhere in the git logs...
and indeed if you have a copy in your own .FreeCAD/Mod it should override the src one

pierreporte, 2023-11-13:
Any news on this?

luzpaz, 2023-11-13:
LIbreArts posted a neat UI/UX feature:

https://twitter.com/lgworld/status/1722251804690481304
https://twitter.com/lgworld/status/1722731447293730994

pierreporte, 2023-11-13:
@luzpaz I saw this today and it made me wondering if an issue was already open.

luzpaz, 2023-11-13:
@SuzanneSoy any spare time to make a PR ?

pierreporte, 2024-04-22:
@maxwxyz for the UI/UX project.

GAZ082, 2025-11-29:
The seachbar add on is unstable (RC1) and has not been developed in the last couple of years. Would be awesome to have something ala Blender in the core system.

luzpaz, 2026-09-05:
Note: a recent Addon approved in the Addon Manager, ToolSeek https://github.com/robdevtech/ToolSeek is also another option
[image]