FreeCAD/FreeCAD #8294: [Issue] [Path] Adaptive - Use Outline fails for STEP import

dresco, 2023-01-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=15&t=75605

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Fedora Linux 35 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31641 (Git)
Build type: Release
Branch: master
Hash: d28d63b87b60161419c6c0c532fbbfaed96926b8
Python 3.10.8, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Assembly4 0.12.4
  * FeedsAndSpeeds 0.5.0
  * kicadStepUpMod 10.16.5
[/code]
```

### Subproject(s) affected?

Path

### Issue description

The Path_Adaptive operation attempts to find the outline of a face using face.Wires[0]

https://github.com/FreeCAD/FreeCAD/blob/e188802ca6997d2564e7570ab648462e6a059f87/src/Mod/Path/Path/Op/Adaptive.py#L838-L843

For the particular STEP import referenced in forum threads [here](https://forum.freecadweb.org/viewtopic.php?f=15&t=75605) and [here](https://forum.freecadweb.org/viewtopic.php?f=15&t=75475), this fails - as Wires[0] is a hole in the face, rather than the outline.

This can be resolved by using `face.OuterWire` instead of `face.Wires[0]`, but am unsure whether this is the correct approach, or whether the model is just broken somehow? (Worth noting that Pocket also fails to find the outline, which I believe uses different logic).

Happy to submit a PR for this one line change, if this approach is useful / acceptable..

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-31:
@mlampert can you weigh-in here too ? TIA!

luzpaz, 2023-02-13:
@sliptonic would you be in favor of the proposed change suggested above?

sliptonic, 2023-02-13:
Looks good to me.  Let's try it.  Go ahead and submit the PR.

luzpaz, 2023-02-13:
Thanks sliptonic!  
@dresco please acknowledge

dresco, 2023-02-13:
Thanks, will do this week!

dresco, 2023-02-20:
@sliptonic @luzpaz Have raised PR #8563, thanks.

closed 2023-02-23 by sliptonic