FreeCAD/FreeCAD #8296: [Issue]

eliasmagn, 2023-01-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?t=49294

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Debian GNU/Linux bookworm/sid (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
[/code]
```

### Subproject(s) affected?

Techdraw

### Issue description

Scale parameter of view in Tech Draw is greyed out!

### Anything else?

Reproducible following this video:
https://www.youtube.com/watch?v=w3IUOA5Xd7I&list=PLw48L7HmCgML1XsR7LSaXlUZkLHyrvVcR&index=12

[image]

I only changed file extension of the FCStd.

[Grundkurs FreeCAD 0.19 - Übung #01 - Winkelprofil.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/10530928/Grundkurs.FreeCAD.0.19.-.Ubung.01.-.Winkelprofil.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-01-30:
You have ScaleType set to "Page".  If you want the View to use a different scale than that set for the Page, you need to set ScaleType to "Custom"

eliasmagn, 2023-01-30:
Haha, thank you i was blinded by the light!

closed 2023-01-30 by eliasmagn