FreeCAD/FreeCAD #8297: [Feature Request] Radial/ elliptical gradient as display background colour

Intuos5, 2023-01-30:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?t=72672&start=60

### Subproject(s) affected?

Other (specify in description)

### Idea description

Currently, it is possible to choose between a flat background colour and a linear gradient:
[image]

It would be nice to have a radial/ elliptical gradient as option:
[image]

The reason is that this gives a more balanced colour distribution across the screen. That way, there is less of a strong highlight in the centre and more a halo-like effect, which helps improve visibility of the model and sketches.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-03:
@Intuos5 just wondering, can this be set via stylesheet?

MisterMakerNL, 2023-03-03:
Been digging into this:
Came out on the 3DInventoryViewer.
IS this the viewer that the Quarter widget uses to overlay the Coin3D engine?
https://freecad.github.io/SourceDoc/d5/d29/classGui_1_1View3DInventorViewer.html#a80066cb7b376b603218006587af684a1

luzpaz, 2023-03-03:
@realthunder would you happen to know ?

MisterMakerNL, 2023-03-03:
Did some more digging but the deeper I go the further I am out of my league.
3DInventoryViewer has an [GLOverlayWidget()](https://freecad.github.io/SourceDoc/df/d70/classGui_1_1GLOverlayWidget.html#af98a4ccb67c263d0e0bc85c74ad22215) that use [Qpainter](https://doc.qt.io/qt-6/qpainter.html), that also has a background and an image option.
My goal is an image btw not a gradient, images are way more versatile and current systems won't mind the extra memory usage. (also you can just go back to single color if you are worried about that.)
An SVG file option would be ideal, since it would be small sized and versatile. Although having png as an option would be cool too!

Intuos5, 2023-03-04:
@MisterMakerNL indeed an image is more versatile. Inventor also uses images, noticed that when I gave it a spin. Maybe it's possible to ship an inkscape template (SVG) file so people can easily adjust it. Completely agree with you.

Intuos5, 2023-03-04:
> @Intuos5 just wondering, can this be set via stylesheet?

It's not in the default templates IIRC, haven't read it, but didn't use the search function in VSCode either.

MisterMakerNL, 2023-03-04:
Currently we change the gradient colors with the theme cfg file. There is a background image/watermark in the stylesheet, you see that when you close all documents.
If we can set the background of the 3D window to transparent. 😆

Intuos5, 2023-03-04:
Ah yes the FreeCAD watermark.Don't add hidden easter eggs 😂

MisterMakerNL, 2023-03-05:
More diggin:
https://www.openinventor.com/reference-manuals/NewRefMan9912/RefManJava/com/openinventor/inventor/vrmlnodes/SoVRMLTextureBackground.html#backTexture
and
https://www.openinventor.com/reference-manuals/NewRefMan9912/RefManJava/com/openinventor/inventor/vrmlnodes/SoVRMLBackground.html#backUrl
Now if only I knew how to test those command :)

luzpaz, 2023-03-08:
@chennes do you mind weighing-in ?

PR #9149 (2023-04-02): [GUI] Radial gradient implementation
This commit implements radial gradient as background and adds the option to settings. It also renames "Color gradient" as "Linear gradient", keeping Linear gradient default. Internally, it remains unchanged for compatibility.

Radio gradient is more suitable for CAD, since it gives a more balanced color distribution across the screen, improving visibility of model and sketches with a halo-like effect.

This pull request closes #8297. I advise to set radial gradient default in the upcoming version of FreeCAD.

<details><summary>Options</summary>

[image]

</ [... cut]
closed 2023-04-02 by chennes