FreeCAD/FreeCAD #8298: [Bug] [Part] Referencing a Placement with an expression in any compatible property field of most Part_* prevents exactly one dimensional property from updating

PeterSHollander, 2023-01-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
OS: Windows 10
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
```

### Subproject(s) affected?

Part

### Issue description

I am working on placing multiple components on a circuit board, and defining each component's position/rotation via a `Placement` in a spreadsheet for easy and condensed updating as the positions will change via a configuration table.

I have distilled the issue down to these reproducible instructions:

1. In a new document, enter the Spreadsheet workbench and create a new spreadsheet.
2. Create a placement entry:
```= create(<<placement>>; create(<<vector>>; 1; 2; 3); create(<<rotation>>; 90; 0; 0))```
Give this placement an Alias `TestPlacement`.
3. Create a vector entry:
```= create(<<vector>>; 1; 2; 3)```
Give this vector an Alias `TestVector`.
4. Switch to the Part workbench.
5. Add a new `Std_Part`
6. Toggle the visibility of the new Part's `Origin` for coordinate reference
7. Add a `Part_Box`.
8. In the Cube's properties, right click on `Length`, and select `Expression...`
9. Enter an element from the created vector `Spreadsheet.TestVector.x`
10. Do the same for the `Width` property using `Spreadsheet.TestVector.y`
11. Do the same for the `Height` property using `Spreadsheet.TestVector.z`
12. Change the x/y/z values of the vector in the spreadsheet.
13. Notice how the dimensions of the Cube have been updated as expected.
14. Next, in the Cube's properties, right click on `Placement`, and select `Expression...`.
15. Enter the created placement `Spreadsheet.TestPlacement`
16. Change the position and rotation values of the placement in the spreadsheet.
17. Notice how the placement of the Cube has been updated as expected.
18. Change the x/y/z values of the vector in the spreadsheet again.
19. Notice *now* how the `Length` and `Height` properties of the Cube have updated as expected, **but the `Width` property is not updating!**
20. In the Cube's properties, right click on `Placement`, select `Expression...`, and select `Clear`.
21. Notice how the `Width` property of the Cube is updating once again now that the Placement is no longer being referenced.

I have confirmed that this bug affects `Part_Box` (Width), `Part_Cylinder` (Radius), `Part_Sphere` (Radius), `Part_Cone` (Radius2), `Part_Torus` (Radius2).  **It does not appear to affect `Part_Tube`**, so hopefully that piece of information is helpful!

It is also worth noting that it appears this bug can be created by referencing a `Placement` in any property field, including `Attachment` and a custom property; it does not have to be the Base `Placement`.  The placement can also be sourced from referencing another part; it doesn't have to be a spreadsheet.  The result is still the same with being unable to edit exactly one of the dimensional properties.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-01:
Can you zip the example freecad file and attach it to this ticket?  TIA!

acpopescu, 2023-02-02:
This may have been fixed by https://github.com/FreeCAD/FreeCAD/pull/8311

PeterSHollander, 2023-02-03:
Hey, that's promising to hear!  I was also noticing some unexpectedly similar Sketcher update failures when referencing vectors and placements, so hearing this is very serendipitous 🙂  I'm not yet familiar with building FreeCAD from source and I couldn't locate any nightly builds after a quick search, so I will plan to confirm this behaviour is fixed with the next release.  Thank you, @acpopescu!

adrianinsaval, 2023-05-11:
@PeterSHollander you can find development builds here: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

can you please test if this is fixed?

luzpaz, 2023-05-15:
@PeterSHollander ping

Roy-043, 2023-07-15:
I confirm that this has been fixed. Closing this issue...

closed 2023-07-15 by Roy-043