FreeCAD/FreeCAD #8302: <class 'AttributeError'>: 'NoneType' object has no attribute 'setOrigin'

maxcel2, 2023-01-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Linux Mint 20.3 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

Starting new project in Part Design workbench: 
File->New
Tasks -> Create body
In the Report window:

16:52:46  Traceback (most recent call last):
  File "/tmp/.mount_FreeCAJpnyr3/usr/Mod/Path/PathScripts/PathJobGui.py", line 1622, in addSelection
    self.updateSelection()
  File "/tmp/.mount_FreeCAJpnyr3/usr/Mod/Path/PathScripts/PathJobGui.py", line 1368, in updateSelection
    self.form.setOrigin.setEnabled(False)
<class 'AttributeError'>: 'NoneType' object has no attribute 'setOrigin'

Anything done after that brings up the same or similar 'AttributeError' messages

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-01-30:
Thanks for the report -- I am not sure why anything in Path is even executing when you create a single Part! Maybe @sliptonic or @mlampert can shed some light on it. Have you tried posting in the FreeCAD forum to get help with troubleshooting steps? Also, if memory serves, that version of FreeCAD isn't quite current, so you could try downloading 0.20.2 and see if it is a known bug that has been fixed already.

mlampert, 2023-01-31:
I can't reproduce the issue. `Part` has no dependency on `Path` it is a mystery to me why the Job's gui should be involved.

maxcel2, 2023-01-31:
Thanks to @chennes, I've installed version 0.20.2, and I'm not getting any error messages. @mlampert, it was Part Design workbench, not Part, I don't know if that is significant. Wonder if this is related to Python versions, it looks similar to problems I was getting when trying to run PyCAM, where I had to change uppercase in scripts to lowercase, because of different Python versions.

chennes, 2023-01-31:
OK, sounds like this was resolved by the upgrade. If there's still an issue with 0.20.2 please feel free to re-open.

closed 2023-01-31 by chennes

drogenlied, 2023-07-02:
I hit this issue in 0.20.2 packaged for Debian unstable (0.20.2+dfsg1-6) :

    File "/usr/share/freecad/Mod/Path/PathScripts/PathJobGui.py", line 1632, in clearSelection 
        self.updateSelection()
    File "/usr/share/freecad/Mod/Path/PathScripts/PathJobGui.py", line 1369, in updateSelection 
        self.form.setOrigin.setEnabled(False)
     <class 'AttributeError'>: 'NoneType' object has no attribute 'setOrigin'

### Steps:
 * New file
 * Create Sketch, add some lines and circles forming a surface with holes
 * Part Workbench, Part_Extrude
 * Path Workbench, new Path_Job, add the Sketch from step 2 to Job
 * Python exception every time some part of the Job is selected in the model treeview

After removing the extruded part and the path job, saving, restarting and loading, I could successfully create a Path_Job and generate GCode from that.

michaelosthege, 2023-11-01:
I'm running into this error a lot (v0.21.1) when working on files that contain multiple Path jobs.

I don't know how FreeCAD works on the inside, but to me it looks like the method is getting called when the form has already been disposed?
Maybe the issue is more prominent when things are slow? That's certainly the case with my file 😅

A simple
```python
if self.form is None:
    return
```
would prevent the `AttributeError`, but not the root cause, of course.

Maybe `FreeCADGui.PySideUic.loadUi()` returns `None` in some cases?

luzpaz, 2023-11-01:
Full [about](https://wiki.freecad.org/About) info please