FreeCAD/FreeCAD #8305: [Issue]

Strahlensauger, 2023-01-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&t=75591

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31709 (Git) AppImage
Build type: Release
Branch: master
Hash: e188802ca6997d2564e7570ab648462e6a059f87
Python 3.10.8, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly4.backup1662658912.4353526
  * Behave-Dark-Colors 0.1.1
  * Behave-Dark-Colors.backup1662668082.023044
  * Assembly4 0.12.4
  * FeedsAndSpeeds 0.5.0
  * freecad.gears 1.0.0
  * A2plus 0.4.60e
  * A2plus.backup1662291454.6363213
[/code]
```

### Subproject(s) affected?

None

### Issue description

Hello,
this error is OS specific, it appears only in Arch OS. (I didn't check arch based distros like manjaro...)
When typing in the python console or when typing in the macro editor the letters 
w e t 
don't react, letter z shows up only  after typing z twice.
The Error 
'failed to create compose table'
appear in the console and in the log window.
In opensuse leap only the error appears, no keys are suppressed, in debian everything works fine
Since version 31641 this error is persisting, before everything worked fine.

Thanks for programming such a great software!!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-30:
Please add a succinct title to this bug

chennes, 2023-01-30:
Does this happen anywhere else (e.g. when entering an expression in the expression window, etc.) or is it literally just those two places?

chennes, 2023-01-30:
For anyone looking into this: that error text appears in two Qt source files (this is Qt 6.4 but it's probably similar in 5.15):
```
qtbase/src/plugins/platforminputcontexts/compose/qcomposeplatforminputcontext.cpp:55:        
    qCWarning(lcXkbCompose, "failed to create compose table");
qtwayland/src/client/qwaylandinputcontext.cpp:235:        
    qCWarning(qLcQpaInputMethods, "failed to create compose table");
```

Strahlensauger, 2023-02-06:
the error occured with Qt5.15.6, where Qt 5.15.4 seems to work fine

mikijov, 2023-02-22:
I have the same problem. In addition I am unable to use these keys in the spreadsheet either, although keys are different in the spreadsheet. Indeed it seems to be that shortcuts supersede data entry.

chennes, 2023-03-03:
This is the same issue as #8346 (at least I think so) -- I'm going to close this one in favor of that one (no particular reason, I've just pointed the other reports to that one already). If it turns out this is a different issue, please re-open.

closed 2023-03-03 by chennes