FreeCAD/FreeCAD #8309: Path Profile cut direction reversed with tool compensation off and extra offset <> 0

rmu75, 2023-01-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
OS: KDE neon 5.26 (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Austria (de_AT)
```

### Subproject(s) affected?

Path

### Problem description

Cut direction of Path Profile generated tool path is reversed in case "compensation" is off and an extra offset is specified. Cases "compensation on" and "compensation off, extra offset = 0" seem to be correct.

Work around (changing the cut direction) doesn't work e.g. when using the LeadIn/Out dressup, which generates odd and dangerous lead in / out paths as it interprets the cut direction by looking at the "direction" property and doesn't calculate it from the path it is operating on.

### Anything else?

Forum discussion https://forum.freecadweb.org/viewtopic.php?f=15&t=75476
[profilecutdirection.zip](https://github.com/FreeCAD/FreeCAD/files/10545064/profilecutdirection.zip)
[source](https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Path/Path/Op/Profile.py#L355)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-01-31:
@rmu75 You've linked to the current HEAD source code  so the problem should also theoretically  exist on current development version.  Any interest in creating a PR to fix this bug ?

rmu75, 2023-01-31:
Indeed I also tested it with some semi-current appimage (same results), the code in question hasn't been changed recently apart from Path reorganisation.

What bugs me: for some reason, with compensation off and extra offset == 0.0, orientation has to be set opposite to what it has to be set to if offset <> 0 or if compensation is on. That seems strange.

The "obvious" fix 

```
        if not obj.UseComp and obj.OffsetExtra.Value == 0.0:
             if direction == "CCW":
                params["orientation"] = 1
            else:
                params["orientation"] = 0
```

is probably wrong somehow. Need to dig a bit deeper it seems. Would create a PR of course.

luzpaz, 2023-02-05:
@mlampert if you don't mind, would you weigh-in?

rmu75, 2023-02-06:
It seems libarea doesn't always determine direction of path by sorting wires but assumes a particular direction in some cases. Will have to further debug this, I'm not too familiar with the code base. It seems PathProfile doesn't have a test case, so I will have to add a new Profile-Test also.

sliptonic, 2023-02-22:
I believe libarea follows the wire direction directly. and then reverses it for nested wires.  So if the outer wire is CW, the first level nested will be CCW, etc.

rmu75, 2023-06-26:
So i again found some time to look into it.

Profile is using some kind of polygon offset functionality in libarea. In case nothing is to offset, i.e. tool compensation is off and no extra offset is defined or extra offset = - tool_diameter/2, there is nothing to offset, and that is the problematic case, convention from libarea and of FreeCAD seem to differ. It looks like the FreeCAD code checks for this with case with "if not obj.UseComp:", but this misses extra offset without compensation or radius  + extra offset == 0, both these cases result in wrong orientation.

I really don't want to dig any further into libarea. IMO profile (and pocket) should use code like in adaptive clearing, without calling into libarea, but I'm not sure if that would have some drawbacks.

So I propose to change https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Path/Path/Op/Profile.py#L355 to
```py
        if (obj.UseComp and self.radius or 0.0) + obj.OffsetExtra.Value == 0.0:
            if direction == "CCW":
                params["orientation"] = 1
            else:
                params["orientation"] = 0
```
which inverts direction iff case no effective offset is applied to the path.

J-Dunn, 2023-06-26:
`self.radius or 0.0`
??

Yeah, I'm not sure if patching one-off errors is the best way but at least it's a hack around an existing issue. 

I've been looking into the relationship of libarea, Area.cpp etc. with the idea to try updating clipperlib ( from which Dans Heeks' clone was later pulled into FreeCAD's libarea.  It seems not much was changed from Angus Johnson's original , so updating may be feasible. 

There's quite a lot of improvements and bug fixes in clipper since 2017 and hopefully that may fix some oddities in FC. 

 Area.cpp is quite complex and I'm still ploughing through how much it still uses libarea. How familiar are you with it?

rmu75, 2023-06-26:
```py
(obj.UseComp and self.radius or 0.0)
```
if `obj.UseComp` evaluates to `True` this yields `self.radius`, else it yields `0.0`.

I'm not really familiar with the code, and I find the multiple layers of glue code quite hard to follow.

My suspicion is that convention how to orient a wire with offset "0" differs between libarea and what is expected in Path, and the inversion of orientation depending on UseComp was a "hack" that didn't catch all cases. I'm not comfortable to change anything in libarea and/or the glue code at this point.

I will try to come up with automated tests and then a pull request.

J-Dunn, 2023-06-26:
> if obj.UseComp evaluates to True this yields self.radius, else it yields 0.0

So you can do bitwise booleans on real values ?  python is truly a silly language. 

Thanks.

rmu75, 2023-06-27:
While looking for useful templates for a unit-test I checked the deburr operation which seems to suffer from the same problem. If parameters are set so the path is exactly on the edge, i.e. offset is zero, the direction of path is reversed.

Deburr uses offsetWire from Path/Op/Util.py, which doesn't use libarea AFAIS.

J-Dunn, 2023-06-27:
Thanks,  I reported this bug is deburr at least a year ago.   IIRC it got resounding silence.

rmu75, 2023-06-27:
Am Montag, dem 26.06.2023 um 12:36 -0700 schrieb J-Dunn:
> > if obj.UseComp evaluates to True this yields self.radius, else it
> > yields 0.0
> So you can do bitwise booleans on real values ? python is truly a
> silly language.

This is kinda like the C ?: operator(s). This idiom "(v and
result_if_v_true or result_if_v_false)" is quite useful in python and a
consequence of short circuit evaluation of "and" and "or" operators.
The only thing that is evaluated in a boolean context is "v".

obj.UseComp is a boolean value, so no harm done there.

J-Dunn, 2023-06-27:
"v and result_if_v_true"  .  That is a boolean operation on a f.p value. 

> This is kinda like the C ?: operator(s).
No, that is just a shorthand for the usual if..then..else syntax. 

if True and 0.7  in C does an implicit type cast on 0.7 to make it bool, it does not attempt to return a real ! 

Python is truly silly.

rmu75, 2023-06-27:
we should probably take that somewhere else because it is quite off
topic.

C:

double test(int b) { return (b) ? 1.0 : 2.0; }

is perfectly valid C, nothing is type-cast, num != 0 returns 1.0, num
== 0 return 2.0. ?: is not a short-hand for if-then-else. 

äquivalent python:

def test(b): return b and 1.0 or 2.0

python "and" operator looks at first argument, if it evaluates to
False, first argument is returned, else the second. python "or" is the
opposite, returns first argument if it evaluates to True, else the
second argument is returned.

I don't know if it is silly or not, it is like it is, and IMO it is
quite logical to do it like that. see 6.11 in python docs
https://docs.python.org/3/reference/expressions.html?highlight=operator#boolean-operations

The difference to "if ... then ... else" in C and in python is that
those are statements whereas ?: resp (... and ... or ...) are
expressions, you can use expressions e.g. as arguments to function
calls, you can't use statements there.

python also knows expression-if syntax, that would look something like

def test(b): return 1.0 if b else 2.0

to the original problem:

```py
if (obj.UseComp and self.radius or 0.0) + obj.OffsetExtra.Value == 0.0:
```

I doubt this equivalent is much more readable:

```py
if (obj.UseComp and self.radius + obj.OffsetExtra.Value == 0.0) or \
   (not obj.UseComp and obj.OffsetExtra.Value == 0.0):
```

J-Dunn, 2023-06-27:
You are correct, that is not any clearer. 

The following costs a tmp variable but is a lot more readable and maintainable. 
```
if (obj.UseComp) : 
    eff_offset= self.radius + obj.OffsetExtra.Value
else:
    eff_offset= obj.OffsetExtra.Value

if (eff_offset==0) :   ........
```

That was what I meant by the "write-only code" quip. 

Try not make anyone else reading the code scratch their to unwind some obscure syntax unless provides a useful performance gain and it's in  a critical code path.

rmu75, 2023-06-27:
Agree. See PR.

J-Dunn, 2023-06-27:
Thanks for the PR . One niggly oddity less.  :)

rmu75, 2023-06-28:
J-Dunn: can you point me to your deburr bug report, thanks.

closed 2023-06-30 by sliptonic