FreeCAD/FreeCAD #8313: [Issue] addon manager

manu-nie, 2023-02-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecadweb.org/viewtopic.php?f=3&p=657493#p657493

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: makepkg
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.0.0
[/code]
```

### Subproject(s) affected?

None

### Issue description

Unable to connect to git hub gets following error when opening ADD manager with No proxy setting
06:47:39 Failed to connect to GitHub. Check your connection and proxy settings.
06:47:53 Unable to retrieve a description from the wiki for macro documentation
06:48:02 Unable to fetch the code of this macro. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-01:
@manu-nie open the python console and paste this:
```python
import urllib.request
r = urllib.request.urlopen("https://api.github.com/zen")
r.read()
```

manu-nie, 2023-02-01:
tried but still problem exist

manu-nie, 2023-02-01:
> @manu-nie open the python console and paste this:
> 
> ```python
> import urllib.request
> r = urllib.request.urlopen("https://api.github.com/zen")
> r.read()
> ```

manu-nie, 2023-02-01:
tried this

luzpaz, 2023-02-01:
What's the output from the command ?

manu-nie, 2023-02-01:
> What's the output from the command ?

>>> import urllib.request
>>> r = urllib.request.urlopen("https://api.github.com/zen")
>>> r.read()
b'Keep it logically awesome.'
>>>

luzpaz, 2023-02-01:
@chennes have you encountered this issue before ?

chennes, 2023-02-01:
Only in the case of messed up proxy settings: I'd double check Preferences to make sure that didn't get screwed up somehow.

luzpaz, 2023-02-01:
@manu-nie please check

manu-nie, 2023-02-02:
what should be settings thn becoz i dont use proxy

luzpaz, 2023-02-02:
What do you see in your preferences ? https://wiki.freecadweb.org/Preferences_Editor#Addon_Manager  
Can you take a screenshot and paste to this ticket ?

manu-nie, 2023-02-03:
this is the preference settting
[image]

luzpaz, 2023-02-05:
@chennes i guess the last suggestion is to flash their user config files, right ?

chennes, 2023-02-05:
Probably a good thing to try: possibly just rename the whole FreeCAD directory so it doesn't get loaded, that will eliminate any Addons as the source of the error as well.

luzpaz, 2023-02-05:
@manu-nie would you try renaming your config file directory to see if this temporarily fixes the issue'? 

An aside: this would be another usecase for #6106

WaterSums, 2023-02-25:
I have just started using FreeCAD.  On Mac OS I had the same problem.  Could not load anything in the Addon Manager.  ping wouldn't work either.  Found this: https://www.debugpoint.com/failed-connect-raw-githubusercontent-com-port-443/.
The first option (adding IP to hosts file) fixed my problem.  Now the Addon Manager works.

chennes, 2023-02-25:
Thanks for the tip, @WaterSums -- so for you it was just `githubusercontent` giving the issue, not GitHub.com itself?

WaterSums, 2023-02-26:
It seemed to be a problem locating raw.githubusercontent.com.  Initially ping was showing (note the IP address)...
PING raw.githubusercontent.com (49.44.79.236):
and there was no response (timeout).

After adding the extra line to /etc/host (linking it to a different IP address), ping displayed...
PING raw.githubusercontent.com (185.199.108.133):
Ping began to work immediately, and so did the Addon Manager.

luzpaz, 2023-03-08:
@chennes any idea how to proceed ?

chennes, 2023-03-08:
Since we haven't heard back from @manu-nie I think we'll have to close the issue. If they re-appear we can reopen if it looks like there is some action we can take.

closed 2023-03-08 by chennes