FreeCAD/FreeCAD #8316: [Issue]

fernom, 2023-02-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Portuguese/Brazil (pt_BR)
Installed mods: 
  * BIM 2021.12.0
  * dodo 1.0.0
  * FreeCAD-NativeIFC 0.2.59
  * Glass
  * InventorLoader 1.3.0
  * ModernUI (Disabled)
  * sheetmetal 0.2.59
[/code]
```

### Subproject(s) affected?

Arch

### Issue description

#### Error trying to export a file to IFC format

- Original assembly made in Autodesk Inventor 2023
- Exported to STEP file
- Saved in Freecad, trying to export to IFC
- Error **"list index out of range"**

File is big (198407 KB)

Complete report view message below

### Anything else?

`09:25:41  Traceback (most recent call last):
  File "<string>", line 4, in <module>
  File "C:\Users\*****\Desktop\FreeCAD-0.20.2-WIN-x64-portable-2\FreeCAD-portable\Mod\Arch\exportIFC.py", line 528, in export
    representation,placement,shapetype = getRepresentation(
  File "C:\Users\*****\Desktop\FreeCAD-0.20.2-WIN-x64-portable-2\FreeCAD-portable\Mod\Arch\exportIFC.py", line 2241, in getRepresentation
    v2 = verts[1].sub(c)
<class 'IndexError'>: list index out of range`

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-01:
Can you zip the file or add it to a 3rd party download service? Also it's customary to write a succinct subject for the title of the ticket. Please do so, tia!

fernom, 2023-02-01:
> Can you zip the file or add it to a 3rd party download service? Also it's customary to write a succinct subject for the title of the ticket. Please do so, tia!

Sorry, is my first issue report.

About the example file, is subject of IP policy. I'll try to reproduce the problem in another file.

luzpaz, 2023-02-05:
@fernom understood. Waiting for the file. Thanks!

luzpaz, 2023-02-12:
@fernom bump

luzpaz, 2023-02-13:
@yorikvanhavre have you seen this issue before ?

luzpaz, 2023-03-03:
@berndhahnebach have you seen this issue before ?

yorikvanhavre, 2023-03-08:
IFC export seems to be working normally on my side... Indeed we need a test file to be able to reproduce the problem, otherwise there is no way to see what's happening

luzpaz, 2023-03-08:
@fernom ping

luzpaz, 2023-04-13:
@fernom bump

luzpaz, 2023-09-23:
Closing due to unresponsive @fernom and no ability to recreate on latest stable.

closed 2023-09-23 by luzpaz