FreeCAD/FreeCAD #8317: [Issue] Color selector papercut - cannot input/ paste values without # symbol

Intuos5, 2023-02-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31391 (Git)
Build type: Release
Branch: master
Hash: 0ab2608a429f3a5ee9a63da9b66d13557655d7e4
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BIM 2021.12.0
  * Launcher
  * persistenttoolbars
  * SelectorToolbar
[/code]
```

### Subproject(s) affected?

Core

### Issue description

The color selector requires the HEX symbol # when pasting a HEX code. If you accidentally didn't provide the symbol when copying the color code externally or try to type in the box, it won't accept any inputs. 
I would expect to be able to input 6 character codes and have the HEX symbol added by FreeCAD itself. 
[image]

### Anything else?

With copy-pasting it can be annoying that you cannot just paste and add the symbol. I often copy the code from the source again only to realize that I am missing the # symbol. Removing the restriction and adding a check for the symbol would make things easier for users.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-03-03:
Where is this dialog and correlate code in the source? any idea ?

Intuos5, 2023-03-04:
In the software, you can find it in the preferences when you change sketcher colours or the viewport background.

luzpaz, 2023-03-04:
```shell
$ grep -Ri "Basic colors"  
grep: src/Gui/Language/qtbase_uk.qm: binary file matches
...
```

My search hits are pulling up Qt `.qm` files so I think this is a Qt-related or generated.  @0penBrain can you lend a hand here ?

0penBrain, 2023-03-04:
Indeed, we use the Qt-provided `QColorDialog`. I can confirm that `#` char is mandatory (not only to paste, but to enter a value in general).
See the regex validator used here : https://code.qt.io/cgit/qt/qtbase.git/tree/src/widgets/dialogs/qcolordialog.cpp?h=5.15#n1314
However to me it makes sense as a so-called "HTML color" always starts with `#`. But just my opinion. ;)

luzpaz, 2023-03-04:
Thanks 0penBrain!  
@Intuos5 would you like to make an upstream Qt bug, in that case ?

closed 2023-03-04 by luzpaz

Intuos5, 2023-03-04:
Yeah, I'll file it for Qt tomorrow and post the link here

Intuos5, 2023-03-05:
Link to the bugreport: https://bugreports.qt.io/browse/QTBUG-111742

Intuos5, 2023-03-17:
The issue appears to have been fixed for Qt 6.5:
https://bugreports.qt.io/browse/QTBUG-111742

luzpaz, 2023-03-18:
Thanks for following-through with this effort and also reporting back. Cheers!