FreeCAD/FreeCAD #8329: [Issue] Changes to draft workbench In-command shortcuts require software reboot

Intuos5, 2023-02-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.31391 (Git)
Build type: Release
Branch: master
Hash: 0ab2608a429f3a5ee9a63da9b66d13557655d7e4
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * BIM 2021.12.0
  * Launcher
  * persistenttoolbars
  * SelectorToolbar
[/code]
```

### Subproject(s) affected?

Draft

### Issue description

When I make changes to the In-command shortcuts for the Draft workbench, I need to reboot FreeCAD before I see the changes take effect. Ideally this would happen instantly or I would be notified that this indeed requires me to reboot FreeCAD.
 
[b]Steps to reproduce[/b]
1. Go to the Draft workbench
2. Go to Edit > Preferences > Draft Workbench
3. Go to the User interface tab
4. Change the shortcut for Close from O to C.
5. Start the Polyline command
6. Draw two lines
7. Notice how the task panel still shows the O shortcut

[image]

8. Hit the C shortcut and notice that it does nothing. 
9. Hit the O shortcut and notice how it is still used to close a shape

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-02-09:
Note that many Draft preferences require a FreeCAD restart. This is documented in the Wiki: https://wiki.freecad.org/Draft_Preferences.

Intuos5, 2023-02-10:
Yes, I understand now because of the way it is implemented in the user.cfg.

There's plenty of space in the UI to add the text:
"Changes require a reboot of FreeCAD"

Roy-043, 2023-02-10:
> Yes, I understand now because of the way it is implemented in the user.cfg.

No, that is unrelated. In DraftGui.py the shortcuts are read once, and then stored in a variable which is never updated during a FreeCAD session. Which is a sensible solution from the perspective of program speed.
https://github.com/FreeCAD/FreeCAD/blob/3f29b469a445447ee7e7c002038a03cfb66794d0/src/Mod/Draft/DraftGui.py#L54

BTW: You should not use the word 'reboot' in this context. You reboot a computer, not a program.

luzpaz, 2023-11-15:
@Roy-043 what's the verdict? How to indicate to the user a restart is needed ?

yorikvanhavre, 2023-11-16:
Note that there are now preference observers. We might try to get rid of needs to restart entirely...

Roy-043, 2023-11-17:
I'll have to look into these observers. We will probably need them for grid related prefs. The in-command shortcuts can probably be fixed without an observer.

PR #11428 (2023-11-18): Draft: make in-command shortcuts non-global
Fixes #8329.
commit 03e8ac4a (2023-11-18): Draft: make in-command shortcuts non-global

Fixes #8329.
closed 2023-11-18 by Roy-043