FreeCAD/FreeCAD #8334: Cannot export code

cgerrist, 2023-02-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
Freecad 0.20.2.29603
```

### Subproject(s) affected?

Path

### Problem description

I try to export the gcode the export fail with a clear error message.  

10:52:07  Running the Python command 'Path_Post' failed:
Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Path/PathScripts/PathPost.py", line 437, in Activated
    (fail, rc, filename) = self.exportObjectsWith(slist, job)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Path/PathScripts/PathPost.py", line 216, in exportObjectsWith
    gcode = processor.export(objs, filename, postArgs)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Path/PathScripts/PathPostProcessor.py", line 105, in export
    return self.script.export(obj, filename, args)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Path/PathScripts/post/KineticNCBeamicon2_post.py", line 227, in export
    job = PathUtils.findParentJob(obj)
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Path/PathScripts/PathUtils.py", line 405, in findParentJob
    for i in obj.InList:

'_TempObject' object has no attribute 'InList'

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-02-05:
@cgerrist would you mind providing an `.FCStd` file in zipped format and attach to this ticket?

cgerrist, 2023-02-05:
Hi, I created a new simple Cnc task en this is working. Strange. You can close the incident.
ps, where do I need to fill-in the About information?

luzpaz, 2023-02-05:
[image]

luzpaz, 2023-02-05:
So is there a way for us to recreate this bug?

sliptonic, 2023-02-14:
closing issue per OP request

closed 2023-02-14 by sliptonic